SmingHub/Sming #1159: TcpClient reconnects after calling .close()

riegaz, 2017-06-13:
I'm creating a TcpClient like this:

`globClient = new TcpClient(nmOnCompleted, nmOnReadyToSend, nmOnReceive);`

At some point I want to restart and before I want to close the TcpClient to do some calculation. However, after calling
`globClient->close();`
the TcpClient reconnects to the tcp server.

Maybe I do not understand the purpose of the close() function?

ADiea, 2017-06-14:
Hi. Offtopic: Does this mean that your other issue with compiling in Windows is solved now? If so please close the other issue thank you:)

slaff, 2017-06-14:
> the TcpClient reconnects to the tcp server.

That should not be the case. But in order to understand the issue better we need a sample application. And the log from the calls.

riegaz, 2017-06-14:
@ADiea actually no. I moved to Linux for now since I did not manage to make it work under Windows.

@slaff I think I found the issue. I did not clear my onComplete callback.

closed 2017-06-14 by riegaz