SmingHub/Sming #1165: Prebuilt esp-open-sdk binaries from bintray

zgoda, 2017-06-21:
Currently prebuilt esp-open-sdk binaries are hosted on Dropbox. Would it be possible to host them on [Bintray](https://bintray.com/)? This service has a free plan for open source projects that seems more friendly and more transparent than Dropbox.

slaff, 2017-06-21:
@zgoda Feel free to do this. It would be great if we have pre-build SDKs for 2.0.0 and 2.1.0 so that we can extend our CI system.

zgoda, 2017-06-21:
I'm not a member of Sming project so the repository at Bintray will be unofficial for this moment. I will switch ownership to Sming once ready.

zgoda, 2017-06-21:
How are these binaries built? My 2.0.0 SDK is ~40MB compressed vs ~160MB for 1.5.4 on Dropbox, archive content seems to be the same:

```
$ tar tfz esp-open-sdk-2.0.0-linux-x86_64.tar.gz --exclude '*/*/*'
esp-open-sdk/
esp-open-sdk/xtensa-lx106-elf/
esp-open-sdk/ESP8266_NONOS_SDK_V2.0.0_16_08_10/
esp-open-sdk/sdk
```

slaff, 2017-06-22:
> 1.5.4 on Dropbox

Which link do you mean actually? It would be great if you can point out the places where we download or suggest to download the toolchain from Dropbox?

> How are these binaries built?

I can suggest the following.

Step 1) Build the compiler only.
```
make STANDALONE=n
```

Then you should archive it to tar.xz format (as nodemcu team is doing: https://github.com/nodemcu/nodemcu-firmware/raw/master/tools/esp-open-sdk.tar.xz).

Step 2) We can download the needed Espressif SDK directly from their download link and add it on top of the compiler. We are actually doing this in our [travis build system](https://github.com/SmingHub/Sming/blob/develop/.travis.yml)

zgoda, 2017-06-22:
https://github.com/SmingHub/Sming/wiki/Linux-Quickstart has links.