SmingHub/Sming #1167: rBoot ssl error

icodk, 2017-06-22:
In docker, I compiled sming with ENABLE_SSL=1 and got a confirmation 
`+ SSL support is enabled`
Then I comiled Basic_rBoot, and used the following urls in the code:
`#define ROM_0_URL  "https://192.168.1.100:443/static/fw/rom0.bin"`
`#define ROM_1_URL  "https://192.168.1.100:443/static/fw/rom1.bin"`
`#define SPIFFS_URL "https://192.168.1.100:443/static/fw/spiff_rom.bin"`
However entering ota to start ota I get a warning:
`WARNING: SSL is not compiled. Make sure to compile Sming with 'make ENABLE_SSL=1' `
and the ota filed. Entring  https://192.168.1.100:443/static/fw/rom0.bin in a browser downloads the file
Here is the debug information(see the warning about half way down):
```
Updating...
Download file:
    (0) https://192.168.1.100:443/static/fw/rom0.bin -> 202000
Creating new httpConnection
HttpConnection::connect: TCP state: -1, isStarted: 0, isActive: 0
HttpConnection::connecting ...
+TCP connection
connect to: 192.168.1.100
TcpConnection::connect result:, 0
Download file:
    (0) https://192.168.1.100:443/static/fw/spiff_rom.bin -> 300000
HttpConnection::connect: TCP state: 2, isStarted: 1, isActive: 1
OnConnected
WARNING: SSL is not compiled. Make sure to compile Sming with 'make ENABLE_SSL=1' 
HttpConnection::onConnected: waitingQueue.count: 2
Storing 55 bytes in stream
realloc 55 -> 203
Storing 19 bytes in stream
Storing 2 bytes in stream
Storing 60 bytes in stream
Storing 19 bytes in stream
Storing 2 bytes in stream
Nothing in the waiting queue
TCP connected
onReadyToSendData: 0
Written: 157, Available: 157, isFinished: 1, PushCount: 1
TcpClient request completed
TCP sent: 157
onReadyToSendData: 2
The headers are complete
rboot_write_flash: item.size: 11
staticOnMessageComplete: Execution queue: 1, https://192.168.1.100:443/static/fw/rom0.bin

Firmware download failed..
In callback...
Firmware update failed!
HTTP parser error: HPE_CB_message_complete
TCP received: (null)
TCP connection error: -9
```

slaff, 2017-06-22:
> Then I comiled Basic_rBoot, and used the following urls in the code:

You have to instruct `Basic_rBoot` to use the SSL enabled Sming library. 
First make sure to start with `clean` code.

```
cd $SMING_HOME/.../samples/Basic_rBoot/
make clean
```

Then either compile the project with the following command
```
cd $SMING_HOME/.../samples/Basic_rBoot/
make ENABLE_SSL=1 
```

or add
```
ENABLE_SSL=1
```

to the Makefile-user.mk in the `samples/Basic_rBoot` and rebuild

```
cd $SMING_HOME/.../samples/Basic_rBoot/
make
```

I would recommend you to use the second option.

icodk, 2017-06-22:
@slaff 
I tried both Not working.Give the same error:
I included the first option results but the second option gives the same error (oops I already wrote it:-)
```
4efafb1668e6:/workspace/Sming/samples/Basic_rBoot# make clean
4efafb1668e6:/workspace/Sming/samples/Basic_rBoot# make ENABLE_SSL=1
Generating unique certificate and key. This may take some time
/bin/sh: /workspace/Sming/Sming/third-party/axtls-8266/tools/make_certs.sh: Permission denied
/workspace/Sming/Sming/Makefile-rboot.mk:482: recipe for target 'include/ssl/private_key.h' failed
make: *** [include/ssl/private_key.h] Error 126
4efafb1668e6:/workspace/Sming/samples/Basic_rBoot# 
```

slaff, 2017-06-22:
> /bin/sh: /workspace/Sming/Sming/third-party/axtls-8266/tools/make_certs.sh: Permission denied

Sigh.... Is anyone reading the error messages?

```
chmod a+x /workspace/Sming/Sming/third-party/axtls-8266/tools/make_certs.sh
```

icodk, 2017-06-22:
@slaff 
Changing permissions did the trick.
Should also be done in the base image ?
I have other issues that might be attributed to the Http server I use (Rocket)
I get the following error:
```
Updating...
Download file:
    (0) https://192.168.1.100:443/static/fw/rom0.bin -> 202000
Creating new httpConnection
HttpConnection::connect: TCP state: -1, isStarted: 0, isActive: 0
HttpConnection::connecting ...
+TCP connection
connect to: 192.168.1.100
TcpConnection::connect result:, 0
Download file:
    (0) https://192.168.1.100:443/static/fw/spiff_rom.bin -> 300000
HttpConnection::connect: TCP state: 2, isStarted: 1, isActive: 1
OnConnected
staticOnConnected: useSSL: 1, Error: 0
SSL: Starting connection...
SSL: Switching to 160 MHz
SSL: handshake start (19284 ms)
SSL: Session Id Length: 0
please start sntp first !
SSL: handshake is in progress...
TCP sent: 100
please start sntp first !
please start sntp first !
SSL: Got error: -517
SSL: closing ...
done

TCP connection closing
-TCP connection

```
Will try with nginx and let you know if still an issue
Thanks

icodk, 2017-06-22:
@slaff 
Still an issue. Works fine with http but get the same error (see above)  with https on port 443 against an nginx server.
By the way, the switch to 160 Mhz  and back should be placed in user app. IMHO.  What if I run my app in 160Hz ? I have to return to 160MHz each and every transaction. But let's solve the first issue first:-)
Thanks

slaff, 2017-06-22:
> SSL: Got error: -517

The certificate validation was not successful. Because... well, long story short, your ESP device does not have the same time as the server and/or does not trust the root CA.  You need to first use sntp to get the exact time and then make your SSL request. Then you need to have also the root CAs that you trust on your ESP 

OR 

you can disable the full SSL verification (ex: https://github.com/SmingHub/Sming/blob/develop/samples/Basic_Ssl/app/application.cpp#L106) and just check the fingerprints ( ex: https://github.com/SmingHub/Sming/blob/develop/samples/Basic_Ssl/app/application.cpp#L124 ).

icodk, 2017-06-22:
how can I tell otaUpdate to setSslOptions(SSL_SERVER_VERIFY_LATER);  short of change  in the  library ? 
The TCPClient  is buried inside private hierarchy
Thanks

slaff, 2017-06-23:
Create a request that will be used as a base for the rBoot Http request(s). Set all needed SSL settings on that base request and then use rBootHttpUpdate::setBaseRequest https://github.com/SmingHub/Sming/blob/develop/Sming/SmingCore/Network/rBootHttpUpdate.h#L63.

Something like this:
```
HttpRequest* baseRequest = new HttpRequest(URL());

#ifdef ENABLE_SSL

baseRequest->setSslOptions(SSL_SERVER_VERIFY_LATER);
//.. etc... 

#endif

//...

rBootHttpUpdate* otaUpdater = new rBootHttpUpdate();
otaUpdater->setBaseRequest(baseRequest);

```

icodk, 2017-06-23:
@slaff 
Will try it out 
Thanks a lot

icodk, 2017-06-24:
@slaff 
I did some tests and get watchdog timeout: I use your sugestion:
```
// SSL	with No  verification
#ifdef ENABLE_SSL

		HttpRequest* baseRequest = new HttpRequest(URL());
		baseRequest->setSslOptions(SSL_SERVER_VERIFY_LATER);
		rBootHttpUpdate* otaUpdater = new rBootHttpUpdate();
		otaUpdater->setBaseRequest(baseRequest);
#endif
```

here is the full message (minus memory dump) may be you can see whats wrong:

```Updating...
Download file:
    (0) https://mysite.com:443/static/fw/0fe00a/rom0.bin -> 202000
Creating new httpConnection
HttpConnection::connect: TCP state: -1, isStarted: 0, isActive: 0
HttpConnection::connecting ...
+TCP connection
connect to: mysite.com
Download file:
    (0) https://mysite.com:443/static/fw/0fe00a/spiff_rom.bin -> 300000
HttpConnection::connect: TCP state: 0, isStarted: 1, isActive: 0
DNS record found: mysite.com = 216.58.211.142
TcpConnection::connect result:, 0
OnConnected
staticOnConnected: useSSL: 1, Error: 0
SSL: Starting connection...
SSL: Switching to 160 MHz
SSL: handshake start (7059 ms)
SSL: Session Id Length: 0
please start sntp first !
SSL: handshake is in progress...
TCP sent: 98
please start sntp first !
TCP sent: 358
SSL: Handshake done (7301 ms).
SSL: Switching back to 80 MHz
HttpConnection::onConnected: waitingQueue.count: 2
Storing 60 bytes in stream
realloc 60 -> 208
Storing 19 bytes in stream
Storing 2 bytes in stream
Storing 65 bytes in stream
Storing 19 bytes in stream
Storing 2 bytes in stream
Nothing in the waiting queue
TCP connected
onReadyToSendData: 0
Written: 167, Available: 167, isFinished: 1, PushCount: 1
TcpClient request completed
TCP sent: 229
onReadyToSendData: 2
ssl->need_bytes=16448 > 6859

E:M 16496

***** Software Watchdog Reset

================================================================
3ffffbd0:  3fff3159 3fff5435 3ffffc00 00004000  
3ffffbe0:  0000048b 0000048b 7fffffff 3fff30c2  
3ffffbf0:  3fff3090 00004000 3fff5435 40230432  
```

slaff, 2017-06-24:
That is something that we have discussed before. ESP has limited RAM and you cannot expect it to work as desktop computer. 

The issue is happening due to not enough memory

```
onReadyToSendData: 2
ssl->need_bytes=16448 > 6859

E:M 16496
```

The remote server uses 16K for fragment size. If this is your own server you can configure it to send 4K fragments instead of 16K.  I am copy-pasting answers from [previous threads](https://github.com/SmingHub/Sming/pull/596):

-----
... some web servers support setting the SSL packet size. The right config directive for NGINX, for example, is `ssl_buffer_size  <size>` and you can use this something like this:

```
server {

    listen        8443;
    ssl on;
    server_name <ssl-name>;

    ssl_certificate      /<path-to>/server.crt;
    ssl_certificate_key  /<path-to>/server.key;
    ssl_client_certificate /<path-to>/ca.crt;
    ssl_verify_client optional;

   #     . -- that setting below sets the SSL packet size. 4k is a reasonable value for ESP. 
   #     |
   #     V
    ssl_buffer_size 4k;

    location / {
        root           /var/www/html;
       # ....
    }
}
```
---

... the http client can ask "politely" the server to send smaller SSL packages. This is done using https://tools.ietf.org/html/rfc6066#page-8. Unfortunately most of the servers do not support this. They still give you back 16K. Now if your client receives request for 16K it tries to allocate 16K in order to store the coming data in it.  In the general case the client cannot say " I will store only 4k and try to decrypt it"  because it needs all 16 K in order to decrypt the data.  That is why you need also control over the server to pass you back 4K instead of 16K.

----

icodk, 2017-06-24:
@slaff 
It works! Amazing ! Greatly appreciate your help and patient.
Hopefully a cosmetics matter. after successful  ota  the esp rests with Fatal exception:
```

Firmware download finished!
 - item: 0, addr: 2000, len: 367968 bytes
 - item: 1, addr: 100000, len: 65536 bytes
In callback...
Firmware updated, rebooting to rom 0...
state: 5 -> 0 (0)
rm 0
pm close 7
TCP connection error: -8
del if0
usl
Firmware updated.

***** Fatal exception 28
pc=0x40106bd7 sp=0x3ffff9e0 excvaddr=0x0000007c
ps=0x00000030 sar=0x00000010 vpri=0x40100e60
r00: 0x40106bd2=1074818002 r01: 0x3ffff9e0=1073740256 r02: 0x00000000=         0 
r03: 0x00000000=         0 r04: 0x60000000=1610612736 r05: 0x0000007e=       126 
r06: 0x6000001c=1610612764 r07: 0x00000004=         4 r08: 0x3fff1cd8=1073683672 
r09: 0x00000130=       304 r10: 0x00000000=         0 r11: 0x3fff0578=1073677688 
r12: 0x3fff2b10=1073687312 r13: 0x00000000=         0 r14: 0x3fff2bf0=1073687536 
r15: 0x3ffffae0=1073740512 

Stack dump:
To decode the stack dump call from command line:
   python $SMING_HOME/../tools/decode-stacktrace.py out/build/app.out
and copy & paste the text enclosed in '===='.
================================================================
3ffff9e0:  40106bd7 00000030 00000010 40100e60  
3ffff9f0:  40106bd2 00000000 00000000 60000000  
3ffffa00:  0000007e 6000001c 00000004 3fff1cd8  
3ffffa10:  00000130 00000000 3fff0578 3fff2b10  
3ffffa20:  00000000 3fff2bf0 3ffffae0 0000001c  
3ffffa30:  3ffffa50 3ffffab6 3ffffab0 40106434  
```

slaff, 2017-06-26:
> Hopefully a cosmetics matter. after successful ota the esp rests with Fatal exception:

If you can reproduce the issue make sure to post the decoded stack trace so that we have some clues what is happening. Just for my understanding - is the OLD rom image resetting ESP? Or the newly updated one?

icodk, 2017-06-26:
The issue is persistent, It is reproduced each time. As far as I can see it is in the current running (OLD) ROM 
The OtaUpdate_CallBack()  is not called. The line above saying: Firmware updated. is not part of my code.
i Tried to analize the Stack dump (on Windows) but get an error from decode-stacktrace.py
```
C:\Dev\shared_with_docker\Basic_rBoot> python decode-stacktrace.py C:\Dev\shared_with_docker\Basic_rBoot\out\build\app_0.out
Traceback (most recent call last):
  File "decode-stacktrace.py", line 34, in <module>
    pipe = subprocess.Popen(shlex.split(command), bufsize=1, stdin=subprocess.PIPE)
  File "C:\Python27\lib\subprocess.py", line 390, in __init__
    errread, errwrite)
  File "C:\Python27\lib\subprocess.py", line 640, in _execute_child
    startupinfo)
WindowsError: [Error 2] The system cannot find the file specified
```

slaff, 2017-06-26:
Most probably the python script cannot find the path to the external xtensa-lx106-elf-addr2line. Make sure that you have it in your PATH.

slaff, 2017-06-26:
> The OtaUpdate_CallBack() is not called. The line above saying: Firmware updated. is not part of my code.

I tend to disagree. In the log above you have

```
 - item: 1, addr: 100000, len: 65536 bytes
In callback...
Firmware updated, rebooting to rom 0...
``` 

Take a look at [that code](https://github.com/SmingHub/Sming/blob/develop/samples/Basic_rBoot/app/application.cpp#L19). It is calling `OtaUpdate_CallBack`.

icodk, 2017-06-26:
Managed to run it (with your suggestion)
Hope you can make sens of it. There are two chanks: 
```
OnHeaderValue(http_parser*, char const*, unsigned int) at /workspace/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:115
0x40100e60: pvPortRealloc at ??:?
0x40106bd2: HttpConnection::staticOnHeaderValue(http_parser*, char const*, unsigned int) at /workspace/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:
115
0x40106434: String::operator=(String const&) at /workspace/Sming/Sming/Wiring/WString.cpp:729
0x40240a72: TcpClient::onError(signed char) at /workspace/Sming/Sming/SmingCore/Network/TcpClient.cpp:208
0x40240ab6: TcpConnection::write(char const*, int, unsigned char) at /workspace/Sming/Sming/SmingCore/Network/TcpConnection.cpp:24
0x40207e10: tsen_meas at ??:?
0x40106524: String::compareTo(String const&) const at /workspace/Sming/Sming/Wiring/WString.cpp:729
0x40240a38: TcpClient::onSent(unsigned short) at /workspace/Sming/Sming/SmingCore/Network/TcpClient.cpp:200
0x4024aa1f: ?? ??:0
0x4024022c: pinMode(unsigned short, unsigned char) at /workspace/Sming/Sming/SmingCore/Digital.cpp:168
0x40207e10: tsen_meas at ??:?
0x40251909: ?? ??:0
0x40106bd2: HttpConnection::staticOnHeaderValue(http_parser*, char const*, unsigned int) at /workspace/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:
115
0x4024004c: _GLOBAL__sub_I__Z6millisv at Clock.cpp:?
0x40251500: ?? ??:0
0x4022dd58: __divsf3 at ??:?
0x40106aad: HttpConnection::staticOnMessageComplete(http_parser*) at /workspace/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:115
0x4024ded6: ?? ??:0
0x402401c0: pinMode(unsigned short, unsigned char) at /workspace/Sming/Sming/SmingCore/Digital.cpp:168
0x4024b474: ?? ??:0
0x40250c26: ?? ??:0
0x40104212: pp_post at ??:?
0x40104212: pp_post at ??:?
0x40103ec1: ppEnqueueRxq at ??:?
0x40214197: rc_set_rate_limit_id at ??:?
0x402141e6: rc_set_rate_limit_id at ??:?
0x40221f28: tcp_listen_with_backlog at ??:?
0x40215b9e: ieee80211_crypto_encap at ??:?
0x40207e24: tsen_meas at ??:?
0x40203079: system_pp_recycle_rx_pkt at ??:?
0x4024022c: pinMode(unsigned short, unsigned char) at /workspace/Sming/Sming/SmingCore/Digital.cpp:168
0x40104212: pp_post at ??:?
0x4023d997: http_parser_execute at /workspace/Sming/Sming/third-party/http-parser/http_parser.c:2169
0x40250d63: ?? ??:0
0x40225d38: handle_dhcp at dhcpserver.c:?
0x4010525c: wDev_ProcessFiq at ??:?
0x40225694: etharp_query at ??:?
0x402280dc: dup_binstr at ??:?
0x402256c8: etharp_query at ??:?
0x40221f28: tcp_listen_with_backlog at ??:?
0x40225708: etharp_output at ??:?
0x402235b1: tcp_rexmit_rto at ??:?
0x40223662: tcp_rexmit at ??:?
0x40000f58: ?? ??:0
0x4022802c: wpa_config_parse_string at ??:?
0x40221f59: tcp_listen_with_backlog at ??:?
0x40225411: etharp_request at ??:?
0x40225dd5: handle_dhcp at dhcpserver.c:?
0x401039ca: lmacTxFrame at ??:?
0x40103de5: ppEnqueueRxq at ??:?
0x40103b7b: ppProcessTxQ at ??:?
0x40103bb6: ppProcessTxQ at ??:?
0x402071f3: ets_snprintf at ??:?
0x40000f49: ?? ??:0
0x40000f49: ?? ??:0

```
And the second:
```
n:3fffffa0:  40000f49 3fffdab0 ffffff01 40000f49  :staticOnHeaderValue(http_parser*, char const*, unsigned int) at /workspace/Sming/Sming/SmingCore/Network/H
ttp/HttpConnection.cpp:115
0x40100e60: pvPortRealloc at ??:?
0x40106bd2: HttpConnection::staticOnHeaderValue(http_parser*, char const*, unsigned int) at /workspace/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:
115
0x40106434: String::operator=(String const&) at /workspace/Sming/Sming/Wiring/WString.cpp:729
0x40240a72: TcpClient::onError(signed char) at /workspace/Sming/Sming/SmingCore/Network/TcpClient.cpp:208
0x40240ab6: TcpConnection::write(char const*, int, unsigned char) at /workspace/Sming/Sming/SmingCore/Network/TcpConnection.cpp:24
0x40207e10: tsen_meas at ??:?
0x40106524: String::compareTo(String const&) const at /workspace/Sming/Sming/Wiring/WString.cpp:729
0x40240a38: TcpClient::onSent(unsigned short) at /workspace/Sming/Sming/SmingCore/Network/TcpClient.cpp:200
0x4024aa1f: ?? ??:0
0x4024022c: pinMode(unsigned short, unsigned char) at /workspace/Sming/Sming/SmingCore/Digital.cpp:168
0x40207e10: tsen_meas at ??:?
0x40251909: ?? ??:0
0x40106bd2: HttpConnection::staticOnHeaderValue(http_parser*, char const*, unsigned int) at /workspace/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:
115
0x4024004c: _GLOBAL__sub_I__Z6millisv at Clock.cpp:?
0x40251500: ?? ??:0
0x4022dd58: __divsf3 at ??:?
0x40106aad: HttpConnection::staticOnMessageComplete(http_parser*) at /workspace/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:115
0x4024ded6: ?? ??:0
0x402401c0: pinMode(unsigned short, unsigned char) at /workspace/Sming/Sming/SmingCore/Digital.cpp:168
0x4024b474: ?? ??:0
0x40250c26: ?? ??:0
0x40104212: pp_post at ??:?
0x40104212: pp_post at ??:?
0x40103ec1: ppEnqueueRxq at ??:?
0x40214197: rc_set_rate_limit_id at ??:?
0x402141e6: rc_set_rate_limit_id at ??:?
0x40221f28: tcp_listen_with_backlog at ??:?
0x40215b9e: ieee80211_crypto_encap at ??:?
0x40207e24: tsen_meas at ??:?
0x40203079: system_pp_recycle_rx_pkt at ??:?
0x4024022c: pinMode(unsigned short, unsigned char) at /workspace/Sming/Sming/SmingCore/Digital.cpp:168
0x40104212: pp_post at ??:?
0x4023d997: http_parser_execute at /workspace/Sming/Sming/third-party/http-parser/http_parser.c:2169
0x40250d63: ?? ??:0
0x40225d38: handle_dhcp at dhcpserver.c:?
0x4010525c: wDev_ProcessFiq at ??:?
0x40225694: etharp_query at ??:?
0x402280dc: dup_binstr at ??:?
0x402256c8: etharp_query at ??:?
0x40221f28: tcp_listen_with_backlog at ??:?
0x40225708: etharp_output at ??:?
0x402235b1: tcp_rexmit_rto at ??:?
0x40223662: tcp_rexmit at ??:?
0x40000f58: ?? ??:0
0x4022802c: wpa_config_parse_string at ??:?
0x40221f59: tcp_listen_with_backlog at ??:?
0x40225411: etharp_request at ??:?
0x40225dd5: handle_dhcp at dhcpserver.c:?
0x401039ca: lmacTxFrame at ??:?
0x40103de5: ppEnqueueRxq at ??:?
0x40103b7b: ppProcessTxQ at ??:?
0x40103bb6: ppProcessTxQ at ??:?
0x402071f3: ets_snprintf at ??:?

```

icodk, 2017-06-26:
You are right, It does call the call back function.
The next two calls are:
rboot_set_current_rom(slot);
System.restart();
before exiting the  call back

closed 2018-05-03 by slaff