SmingHub/Sming #1168: Build system issues

Informatic, 2017-06-24:
- [ ] `make clean` does not properly clean additional modules set in `Makefile-user.mk` variable `MODULES` (only removes `out/build/app*` files)
- [ ] Current build system does not use dependency files (`.d`), so that changes in headers do not cause rebuilds of source files

slaff, 2017-06-26:
@Informatic Please, submit PR with the proposed changes.

Informatic, 2017-06-26:
Yes, I (hopefully) will, in following week/weeks... :)

Informatic, 2017-09-30:
Ok, It seems like the first one is caused by me using relative path (`../base`) in `MODULES`, so that objects instead of being stored in `out/build/MODULENAME`, wind up in `out/build/../MODULENAME`. Sadly, I don't have any idea how to solve that without moving build directories over to module directories...

Informatic, 2017-09-30:
Ok, well, using `$(realpath ../base)` in `MODULES`, as used in rboot appcode, seems to have done the trick

closed 2017-10-12 by slaff