SmingHub/Sming #1169: Supporting C++11 chrono library

simonsilvalauinger, 2017-06-29:
C+11 introduced an awesome [chrono library](http://en.cppreference.com/w/cpp/chrono) which contains semantic time utils like [duration](http://en.cppreference.com/w/cpp/chrono/duration) or [timepoint](http://en.cppreference.com/w/cpp/chrono/time_point). The basis of them are clocks. Three clocks, [steady_clock](http://en.cppreference.com/w/cpp/chrono/system_clock), [system_clock](http://en.cppreference.com/w/cpp/chrono/system_clock) and [high_resolution_clock](http://en.cppreference.com/w/cpp/chrono/high_resolution_clock) are defined.
To improve some time based functions in Sming like `Timer.initializeMs(...)`, one could amend semantic, chrono library based, functions like `Timer.initialize(std::chrono::duration, ...)`, which would be called like `timer.initialize(std::chrono::hours(3), ...)` .
In addition the steady_timer could be implemented (I already did this for privat tests) to move away from custom functions like `micros()` or `millis()`, which often are used for timing purposes, and using the C++ standard library instead.
What do you guys think about it?

slaff, 2017-10-12:
> What do you guys think about it?

Can you send us a PR with the suggested improvements? Once you are ready we can review it.

slaff, 2018-01-30:
Feel free to repon the issue if there is some further interest in the topic.

closed 2018-01-30 by slaff