SmingHub/Sming #1170: Problem with web configuration

sweetpants, 2017-07-01:
Hi, Im trying to write an application based on the MeteoControl example. Got the webserver working, but something strange happens when I configure the password in de WebGUI. I have added a JSON print statement to explain what happens.

1) Started the ESP, went to configuration screen and entered just a SSID,. JSON ouput is:
{
  "network": {
    "ssid": "WLHS-SECURE",
    "password": ""
  }
}
This is ok, screen also looks ok

2) Added password "2Little4Me" and readback the config
{
  "network": {
    "ssid": "WLHS-SECUR",
    "password": "2Little4Me"
  }
}
The last character of the SSID has dissapeared, screen shows wrong SSID

3) Same situation, different password "123456789"
{
  "network": {
    "ssid": "WLHS-SECURE",
    "password": "123456789"
  }
}
Looks ok, screen ok

4) Same situation, different password "1234567890"
{
  "network": {
    "ssid": "WLHS-SECUR",
    "password": "1234567890"
  }
}

The last character of the SSID has dissapeared again
It looks like it happens when the password reaches 10 characters or more, But sometimes the last character is also the last character of the SSID

{
  "network": {
    "ssid": "WLHS-SECUR7",
    "password": "test1234567"
  }
}

I have no idea what is going on here.
Anyone have similar issues?

harryd100, 2017-07-09:
Seems like request.getPostParameter("name") returns bogus data sometimes

slaff, 2017-11-13:
Feel free to reopen the issue if the problem is still present in the latest `develop` version.

closed 2017-11-13 by slaff