SmingHub/Sming #1174: Fatal Exception 20: Simultaneously SoftAP and Station Mode

rickyitexpert, 2017-07-06:
ESP8266 enters into fatal exception 20 in any of the following condition.

1. Enabling SoftAP First and then Station Mode
2. Enabling Station Mode First and then SoftAP

Whether enabling both mode during init or later it always result in Fatal Exception.

Sequence of events during Fatal Exception:
1. Station Mode working good-->AP Mode Enabled-->Some Wifi Client tried to associated with ESP8266--> Enters FATAL Exception;
2. SoftAP Mode working good-->Station Mode Enabled-->No problem-->Station connect function called--> (FATAL Exception occurs form now onwards whenever any wifi client tries to associate to reassociated with ESP SoftAP)-->ESP Association started to AP -->Association Successful-->Dhcpclient got the ip configs.

sweetpants, 2017-07-09:
Since latest update I also have issues running the Webserver either in STA or AP mode. As soon as I try to access index.html I get:

The headers are complete
attached file: index.html (1606 bytes)
TCP received: 140 bytes
READ Template (0)
read errno -10014

***** Fatal exception 3
pc=0x4022cfb6 sp=0x3ffff280 excvaddr=0x40000960
ps=0x00000030 sar=0x00000000 vpri=0xcc407dc8
r00: 0x40234400=1076052992 r01: 0x3ffff280=1073738368 r02: 0x40000960=1073744224 
r03: 0x0000007b=       123 r04: 0x4000cdc2=1073794498 r05: 0x7a13006e=2048065646 
r06: 0x80808080=-2139062144 r07: 0x0100ff01=  16842497 r08: 0x7911ff6d=2031222637 
r09: 0xffffffff=        -1 r10: 0xfefefeff= -16843009 r11: 0x7b7b7b7b=2071690107 
r12: 0x3fff2c68=1073687656 r13: 0x3ffff4e0=1073738976 r14: 0x0000d8e2=     55522 
r15: 0x3fff2c68=1073687656

sweetpants, 2017-07-09:
Hmmm, when I remove rboot, things start to work again

slaff, 2017-07-12:
@rickyitexpert When you use rboot `spiffs_mount` will not work as you might expect (reported in #1009). If you want to use spiffs and rboot then you should try to mount the SpifFS using call like `spiffs_mount_manual(0x100000, 65536);`

slaff, 2017-07-12:
@rickyitexpert We will be able to help you further with this issue if you can provide us a sample code that reproduces the error.

slaff, 2017-11-13:
Feel free to reopen if the issue persists and you have more information related to it.

closed 2017-11-13 by slaff