SmingHub/Sming #1176: HttpRequest.getPostParameter delivers corrupted data

jssilva25, 2017-07-09:
I'm building an application which uses 3 user-modifiable string names. These correspond to the name of three thermometers which have initial names of "Temperature 1", "..2" and "..3".

The problem is when I modify, for instance, "Temperature 2", **all the fields are corrupted** and set to "Temperature 3".

I'm almost 100% sure the problem lies within the library and I would happily submit the solution if I could debug it.

**But this is not possible because as soon as I build Sming with ENABLE_GDB=1 all the IRam is gone.**

Just to take every distraction out of the way, I built an example using HttpServer_ConfigNetwork modifyng only a few lines of code. The most relevant, of course, is the callback function, which goes like this:
```
void onTest(HttpRequest &request, HttpResponse &response)
{
	if (request.method == HTTP_POST)
	{
		// Only "Temperature 2" has been modified
		Serial.print("=======> AppSettings: ");
		Serial.println(AppSettings.TestT1Nome);  // Prints "Temperature 1"
		Serial.print("=======> PostParameter: ");
		Serial.println(request.getPostParameter("TestT1Nome"));  // Prints "Temperature 3"
		AppSettings.TestT1Nome = request.getPostParameter("TestT1Name");
		AppSettings.TestT2Nome = request.getPostParameter("TestT2Name");
		AppSettings.TestT3Nome = request.getPostParameter("TestT3Name");
		AppSettings.save();
	}

	TemplateFileStream *tmpl = new TemplateFileStream("test.html");
	auto &vars = tmpl->variables();
	vars["TestT1Name"] = AppSettings.TestT1Name;
	vars["TestT2Name"] = AppSettings.TestT2Name;
	vars["TestT3Name"] = AppSettings.TestT3Name;

	response.sendTemplate(tmpl);
}

```

slaff, 2017-07-10:
@jssilva25 Get the latest develop version, recompile and check.
```
cd $SMING_HOME
git checkout develop
git pull
make dist-clean
make
```
And then go to your app and recompile it to use the latest sming library

```
cd <your-app>
make clean
make
```

Once you are ready, please, write back to tell us if the issue is still present. If yes - you should provide a way to reproduce it: a sample `curl` request and code that shows the issues that you are experiencing would be very helpful.

jssilva25, 2017-07-10:
The issue is gone, many thanks.

Is there a way to ENABLE_GDB=1 on Sming, even partially, without overloading IRAM, and so being able to debug with GDB?

I think that I, and others, could contribute more if a debug tool is available.

Thanks for your work.

slaff, 2017-07-11:
> Is there a way to ENABLE_GDB=1 on Sming, even partially, without overloading IRAM, and so being able to debug with GDB?

Hm, that is a very good question... I can take a look at it but I doubt it because most of the GDB functions must be in IRAM, otherwise the debugging will crash often.

closed 2017-07-11 by slaff