SmingHub/Sming #1179: TwoWire::begin has reversed parameter order than on ESP8266/Arduino

max-weller, 2017-07-12:
I just spent a few hours debugging a small program I ported from ESP8266/Arduino to Sming. The problem was that I called `Wire.begin(scl, sda);` as it works on the Arduino implementation. On Sming, the order of params is reversed: `Wire.begin(sda, scl);

I'm not sure if this should be changed here or there, but I'd at least like to put this here as a warning for other people searching for problems with I2C on Sming.

zgoda, 2017-07-12:
You have problem elsewhere in code or you use some ancient version of Sming. In 3.2.0 the order matches Arduino core, see https://github.com/SmingHub/Sming/blob/5c37d69f45963dcc25e2fc63ce389494f337f568/Sming/SmingCore/Wire.h#L51

zgoda, 2017-07-13:
I remember we discussed this before. @avr39-ripe @slaff any opinion on this?

avr39-ripe, 2017-07-13:
I think we should make Wire.begin the same as ESP8266/Arduino. Update Wire implementation and update examples app.

slaff, 2017-07-13:
> I think we should make Wire.begin the same as ESP8266/Arduino. Update Wire implementation and update examples app.

@avr39-ripe Agreed. Can you create a new PR?

slaff, 2017-07-27:
Should be fixed with PR #1193 which is merged now in the `develop` branch.

closed 2017-07-27 by slaff