SmingHub/Sming #1180: Exhausting heap with websocket, inconsistent behaviour

zgoda, 2017-07-12:
I have code that just broadcasts what it receives from websocket. The complete messages (with headers) are ~190 bytes and arrive in 5-6 sec intervals, each in new connection. It takes about 1 hour of operation to exhaust available heap on ESP-12 but the problem is that the behaviour is inconsistent. Sometimes it just drops connection (I see `CONNECTION DROPPED` in terminal), and continues after some memory is reclaimed but sometimes WDT fires and resets ESP.

Would it be possible to make this behaviour consistent, eg. always reset ESP when connection is about to be dropped?

slaff, 2017-07-12:
I would say that the best will be to see where the memory leak is coming from and try to fix it. Can you paste a sample code that reproduces the issue?

zgoda, 2017-07-13:
Nothing special, just simplified HttpServer_WebSockets sample:

```c++
#include <SmingCore/SmingCore.h>
#include <SmingCore/Network/Http/Websocket/WebsocketResource.h>

#include "user_config.h"

static const char * WIFI_SSID = "esp_test";
static const char * WIFI_PASSWD = "esp_test";

HttpServer server;

void onIndex(HttpRequest& req, HttpResponse& resp)
{
	resp.sendFile("index.html");
}

void onFile(HttpRequest& req, HttpResponse& resp)
{
	char buf[SPIFFS_OBJ_NAME_LEN + 1];
	char path[SPIFFS_OBJ_NAME_LEN + 1];
	strcpy(buf, req.getPath().c_str());
	if (buf[0] == '/') {
		strncpy(path, buf + 1, sizeof(buf) - 1);
	} else {
		strcpy(path, buf);
	}
	path[strlen(path)] = 0;
	if (path[0] == '.') {
		resp.forbidden();
	} else {
		resp.setCache(86400, true);
		resp.sendFile(path);
	}
}

void onWsConnect(WebSocketConnection& conn)
{
	Serial.println("client connected");
	String msg = "connected";
	conn.broadcast(msg.c_str(), msg.length());
}

void onWsDisconnect(WebSocketConnection& conn)
{
	Serial.println("client disconnected");
	String msg = "disconnected";
	conn.broadcast(msg.c_str(), msg.length());	
}

void onWsMessage(WebSocketConnection& conn, const String& message)
{
	Serial.print("got message: ");
	Serial.print(message);
	Serial.print(", free heap: ");
	Serial.println(system_get_free_heap_size());
	String msg = "echo: " + message;
	conn.send(msg.c_str(), msg.length());
}

void startServer()
{
	server.listen(80);
	server.addPath("/", onIndex);
	server.setDefaultHandler(onFile);
	WebsocketResource * res = new WebsocketResource();
	res->setConnectionHandler(onWsConnect);
	res->setMessageHandler(onWsMessage);
	res->setDisconnectionHandler(onWsDisconnect);
	server.addPath("/ws", res);
}

void STAGotIP(IPAddress ip, IPAddress mask, IPAddress gateway)
{
	startServer();
}

void init()
{
	spiffs_mount();
	Serial.begin(SERIAL_BAUD_RATE);
	Serial.systemDebugOutput(true);
	WifiAccessPoint.enable(false, true);
	WifiStation.config(WIFI_SSID, WIFI_PASSWD);
	WifiStation.enable(true, false);
	WifiEvents.onStationGotIP(STAGotIP);
}
```

The memory usage is steady if I don't create new connection for every message sent by client. As soon as I begin sending messages in separate connections, free heap drops with every connection. Client code (Python 3.5+, messages sent in single connection):

```python
import asyncio
import signal
import random

from autobahn.asyncio.websocket import WebSocketClientFactory, WebSocketClientProtocol

WS_HOST = '10.42.0.54'
WS_PORT = 80
WS_URL = 'ws://{0}:{1}/ws'.format(WS_HOST, WS_PORT)

class ClientProtocol(WebSocketClientProtocol):

    def __init__(self, *args, **kwargs):
        super(ClientProtocol, self).__init__(*args, **kwargs)
        self.counter = 0

    def _task(self):
        msg = 'message no {0}'.format(self.counter)
        self.sendMessage(msg.encode())
        interval = random.randint(1, 10)
        self.counter = self.counter + 1
        self.factory.loop.call_later(interval, self._task)

    def onOpen(self):
        self.sendMessage(b"Hello server, I'm here")
        interval = random.randint(1, 10)
        self.counter = self.counter + 1
        self.factory.loop.call_later(interval, self._task)

def main():
    loop = asyncio.get_event_loop()
    for signame in ('SIGINT', 'SIGTERM'):
        sig = getattr(signal, signame)
        loop.add_signal_handler(sig, loop.stop)
    factory = WebSocketClientFactory(WS_URL, loop=loop)
    factory.protocol = ClientProtocol
    coro = loop.create_connection(factory, WS_HOST, WS_PORT)
    try:
        loop.run_until_complete(coro)
        loop.run_forever()
    finally:
        loop.close()

if __name__ == '__main__':
    main()
```

avr39-ripe, 2017-07-13:
@zgoda If I understand you right, you try to make NEW websocket connection every 5..6 sec? If so, I think it is "normal" that memory "leaks" because each connection, even unused need some time to free resources, it is kind of race and new connection maker wins and eats all esp memory. You should either use one connection or limit new connection count or increase time between new connections.

zgoda, 2017-07-13:
Yes, I know that. The problem is not heap exhaustion but inconsistent behaviour when there's not enough memory to accept incoming connection. Sometimes firmware drops connection and continues after some memory is reclaimed but sometimes WDT resets chip.

slaff, 2017-07-13:
When you create the HttpServer you can [specify settings](https://github.com/SmingHub/Sming/blob/develop/Sming/SmingCore/Network/HttpServer.cpp#L24) for it that differ from the default ones. If you want to have consistent behaviour you can set `settings.minHeapSize` to 0. This way the HttpServer will consume all available memory and ESP will get, most probably, restarted due to not-enough memory.

zgoda, 2017-07-13:
Thanks, that is the thing i'm looking for.

closed 2017-07-13 by zgoda