SmingHub/Sming #1181: How to properly control servo from HTTP Server

yeralin, 2017-07-12:
Hello,

I'm trying to create a HTTP Endpoint that will rotate servo on GET Request:

```
void dispenseFood(HttpRequest &request, HttpResponse &response)
{
	int rotateTimes = 5;
	bool clockwise = true;
	servoCntrl->setValue(-90);
	while (rotateTimes != 0 ) {
		if (clockwise) {
			clockwise = false;
			servoCntrl->setValue(90);	
		} else {
			clockwise = true;
			servoCntrl->setValue(-90);
		}
		
		delay(500);
		rotateTimes--;	
	}
	response.setContentType(ContentType::TEXT);
        response.sendString("Done");

}
```
This fails with: `***** Software Watchdog Reset`

I found similar issue: https://github.com/SmingHub/Sming/issues/497
But it doesn't really cover my case :(

slaff, 2017-07-13:
> This fails with: ***** Software Watchdog Reset

@yeralin Can you send decoded stack trace (when the device crashes it should print a lot of information in the console ) ? 

Are you using the latest `develop` version? 
Can you add after the `delay()` another line which resets the software watchdog `system_soft_wdt_feed ();` ?

yeralin, 2017-07-14:
@slaff Sorry, should have provided more debug info.

Here is my stacktrace:
```
path=/dispenseFood
Name: host
host === 192.168.1.21
parsed
Request: GET, nodata
TCP received: 428 bytes
onReadyToSendData: 1
 Times: 2

***** Software Watchdog Reset

================================================================
3ffff790:  3fff1070 000d000d 00000002 402367e9  
3ffff7a0:  3fff1088 00090009 3fff0320 00010001  
3ffff7b0:  3fff0c30 3ffef944 3ffff7e0 3ffef97c  
3ffff7c0:  3fff0c30 3ffef944 3fff0bf8 40236bea  
3ffff7d0:  3fff0c30 3ffef944 3fff0bf8 4023c8e7  
3ffff7e0:  3fff0a30 000d000d 00000000 402377ec  
3ffff7f0:  74617453 206e6f69 73205049 3ffef944  
3ffff800:  3ffffaa8 3ffe8370 00000000 402377ec  
3ffff810:  69642f27 6e657073 6f466573 2027646f  
3ffff820:  69676572 72657473 000a6465 0a000a64  
3ffff830:  76726500 74207265 6f656d69 75207475  
3ffff840:  00000006 00000000 00000010 00000000  
3ffff850:  00000006 00000000 00000010 00000000  
3ffff860:  00000000 00000000 0000ff00 0003c033  
3ffff870:  00000000 80000000 00000000 0000001b  
3ffff880:  00003000 3ffef97c 00003000 00008000  
3ffff890:  3fff0488 00000000 00003000 40100de4  
3ffff8a0:  00008000 40100dfd 3fff0320 4023816a  
3ffff8b0:  0000000e 4020ec5c 40001e65 3ffff902  
3ffff8c0:  3ffff980 00000006 00000000 4000050c  
3ffff8d0:  3ffff910 3ffff900 00000004 00000022  
3ffff8e0:  3fff0448 000d000d 00000001 4023784c  
3ffff8f0:  00000001 3ffff900 00000004 00000000  
3ffff900:  00000000 00000001 402361ee 00000000  
3ffff910:  3fff0bf8 3fff0c30 3fff0be0 4023e9ac  
3ffff920:  3fff0bf8 3fff0088 00000001 4023a1c0  
3ffff930:  65526e6f 54796461 6e65536f 74614464  
3ffff940:  25203a61 00000a64 3ffffad0 00000008  
3ffff950:  3ffffb60 3ffffb30 00000004 402376b4  
3ffff960:  00383234 3fff0a3b 00000000 00000000  
3ffff970:  00000000 00303900 4021b04f ffffffff  
3ffff980:  000001ac 3ffe8370 3ffff994 3ffff9a4  
3ffff990:  3ffff9a0 00000018 32393120 3836312e  
3ffff9a0:  3ffffa08 00000100 3ffffb30 3ffffb20  
3ffff9b0:  00000004 3ffffb20 00000000 402376b4  
3ffff9c0:  3f003039 00000100 3ffffb50 3ffffb40  
3ffff9d0:  3fff0e60 0000ff00 00000000 00000000  
3ffff9e0:  3fff0bf8 3ffe8370 00000000 402377ec  
3ffff9f0:  20504354 65636572 64657669 3234203a  
3ffffa00:  79622038 0a736574 00000000 40100de4  
3ffffa10:  3fff1040 00000001 00000004 40100de4  
3ffffa20:  3fff0440 0000ff00 0000000f 00000000  
3ffffa30:  3fff0bf8 3fff0088 3fff0be0 4023ee1b  
3ffffa40:  00000010 4020ec5c 3ffffc08 0000000f  
3ffffa50:  3fff0318 0000ff00 00000002 40100de4  
3ffffa60:  00000000 00000002 0000ff00 40105af0  
3ffffa70:  00000003 4020ec5c 3ffffb00 00000002  
3ffffa80:  3ffe9632 000001aa 00000002 402381a2  
3ffffa90:  3ffffad0 3ffffac0 3ffffaa4 3ffffab4  
3ffffaa0:  3ffffab0 00000015 3fff0bf8 4023784c  
3ffffab0:  3ffffb15 00000100 3ffffc40 3ffffc30  
3ffffac0:  00000004 3ffffc30 00000000 3ffe9470  
3ffffad0:  00000000 3ffe946f 3fff0bf8 ffffffff  
3ffffae0:  000001ac 3fff0c0c 3fff0bf8 00000000  
[... 72 lines cut ...]
3fffff70:  3ffeb918 3fffdcc0 3ffea470 3ffea470  
3fffff80:  4021a946 3ffede50 00000000 3fff08e8  
3fffff90:  3fffdc80 00000000 3fff0088 4020e86b  
3fffffa0:  40000f49 3fffdab0 ffffff01 55aa55aa  
================================================================

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 26092, room 16 
tail 12
chksum 0xc5
ho 0 tail 12 room 4
load 0x3ffe8000, len 908, room 12 
tail 0
chksum 0xd5
load 0x3ffe8390, len 5416, room 8 
tail 0
chksum 0x75
csum 0x75
��{�l�$␛�␄␌�l`␂��r�l�$2�␌␌␄l`␂��r�l␄��␂�␌␌$l`␂{lO�timeout updating: 70 -> 65535
Server timeout updating: 40 -> 90
```

According to `/opt/Sming/Sming/SmingCore/SmingCore.h`:
#define SMING_VERSION "3.2.0" // Major Minor Sub

yeralin, 2017-07-14:
With `system_soft_wdt_feed ();` 
It was slightly better. It did couple rotations, but then crashed again :D

```
path=/dispenseFood
Name: host
host === 192.168.1.21
parsed
Request: GET, nodata
TCP received: 402 bytes
onReadyToSendData: 1
 Times: 5Rotated clickwise
Rotated counter-clickwise
Rotated clickwise

***** Software Watchdog Reset

================================================================
3ffff780:  3ffff7e0 3fff0ee0 00000003 402367f3  
3ffff790:  3fff1250 00090009 3fff0340 00010001  
3ffff7a0:  00000000 000d000d 3ffff7e0 402381b2  
3ffff7b0:  3fff0ee0 3ffef964 3ffff7e0 3ffef99c  
3ffff7c0:  3fff0ee0 3ffef964 3fff0ea8 40236bfa  
3ffff7d0:  3fff0ee0 3ffef964 3fff0ea8 4023c8f7  
3ffff7e0:  3fff0d00 000d000d 00000000 402377fc  
3ffff7f0:  74617453 206e6f69 73205049 3ffef964  
3ffff800:  3ffffaa8 3ffe8370 00000000 402377fc  
3ffff810:  69642f27 6e657073 6f466573 2027646f  
3ffff820:  69676572 72657473 000a6465 0a000a64  
3ffff830:  00000000 00000000 0000001f 40100991  
3ffff840:  4000050c 00000000 00000010 00000000  
3ffff850:  4000dcec 00000030 0000001e ffffffff  
3ffff860:  40237a94 00000004 0000000a 0000001c  
3ffff870:  00000002 00000004 0000000e 3ffff972  
3ffff880:  00000020 00000044 00000000 3ffff9c0  
3ffff890:  00000004 00000000 00000022 00000030  
3ffff8a0:  00008000 40100dfd 3fff0340 4023817a  
3ffff8b0:  0000000e 4020ec5c 40001e65 3ffff902  
3ffff8c0:  3ffff980 00000006 0000000d 40001e4b  
3ffff8d0:  3ffff910 3ffff900 00000004 00000000  
3ffff8e0:  3fff0468 000d000d 00000001 4023785c  
3ffff8f0:  00000001 3ffff900 00000004 00000000  
3ffff900:  00000000 00000001 402361ee 00000000  
3ffff910:  3fff0ea8 3fff0ee0 3fff0e90 4023e9bc  
3ffff920:  3fff0ea8 3fff00a8 00000001 4023a1d0  
3ffff930:  65526e6f 54796461 6e65536f 74614464  
3ffff940:  25203a61 00000a64 3ffffad0 00000008  
3ffff950:  3ffffb60 3ffffb30 00000004 402376c4  
3ffff960:  00323034 3fff0d0b 00000000 00000000  
3ffff970:  00000000 00303900 4021b04f ffffffff  
3ffff980:  00000192 3ffe8370 3ffff994 3ffff9a4  
3ffff990:  3ffff9a0 00000018 32393120 3836312e  
3ffff9a0:  3ffffa08 00000100 3ffffb30 3ffffb20  
3ffff9b0:  00000004 3ffffb20 00000000 40105e24  
3ffff9c0:  3f003039 401063f8 3fffc230 40106406  
3ffff9d0:  3fff1018 0000ff00 00000000 00000000  
3ffff9e0:  3fff0ea8 3ffe8370 00000000 402377fc  
3ffff9f0:  20504354 65636572 64657669 3034203a  
3ffffa00:  79622032 0a736574 00000000 40100de4  
3ffffa10:  3fff1080 00000001 00000004 40100de4  
3ffffa20:  3fff0460 0000ff00 0000000f 00000000  
3ffffa30:  3fff0ea8 3fff00a8 3fff0e90 4023ee2b  
3ffffa40:  00000010 4020ec5c 3ffffc08 0000000f  
3ffffa50:  3fff0338 0000ff00 00000002 40100de4  
3ffffa60:  00000000 00000002 0000ff00 40105af0  
3ffffa70:  00000003 4020ec5c 3ffffb00 00000002  
3ffffa80:  3ffe965a 00000190 00000002 402381b2  
3ffffa90:  3ffffad0 3ffffac0 3ffffaa4 3ffffab4  
3ffffaa0:  3ffffab0 00000015 3fff0ea8 4023785c  
3ffffab0:  3ffffb15 00000100 3ffffc40 3ffffc30  
[... 73 lines cut ...]
3fffff50:  3ffeacbc 000001f0 00000000 0000001d  
3fffff60:  00000002 0000001a 4021a9d3 3ffede70  
3fffff70:  3ffeacb0 3fffdcc0 3ffea4d0 3ffea4d0  
3fffff80:  4021a946 3ffede70 00000000 3fff08e8  
3fffff90:  3fffdc80 00000000 3fff00a8 4020e86b  
3fffffa0:  40000f49 3fffdab0 ffffff01 55aa55aa  
================================================================

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 26092, room 16 
tail 12
chksum 0x7c
ho 0 tail 12 room 4
load 0x3ffe8000, len 904, room 12 
tail 12
chksum 0xc8
ho 0 tail 12 room 4
load 0x3ffe8390, len 5456, room 12 
tail 4
chksum 0x36
csum 0x36
��{�l�l␒�␟␀␌�l ␃��{�l�l␓�␌␌␌d`␃��r�$␌��␃�␌␄l$`␂sl��timeout updating: 70 -> 65535
Server timeout updating: 40 -> 90
````

yeralin, 2017-07-15:
@slaff Also, I tested this small snippet:
```
/**
 * Basic Servo test application
 *
*/

#include <user_config.h>
#include <SmingCore/SmingCore.h>
#include <Libraries/Servo/ServoChannel.h>

#define SERVO_PIN 5 // GPIO2

ServoChannel *channel;

void init()
{
	Serial.println("Init Basic Servo Sample");
	System.setCpuFrequency(eCF_80MHz);
	WDT.enable(false);
	system_soft_wdt_feed ();
	channel = new(ServoChannel);
	channel->attach(SERVO_PIN);

	while(true) {
		channel->setDegree(-90);
		delay(2000);
		channel->setDegree(90);
		delay(2000);
	}

}
```
Still fails for the same reason:

``` ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 24980, room 16 
tail 4
chksum 0x60
load 0x3ffe8000, len 900, room 4 
tail 0
chksum 0xdf
load 0x3ffe8390, len 2668, room 8 
tail 4
chksum 0x9f
csum 0x9f
��r�l��

***** Software Watchdog Reset

================================================================
3ffffac0:  0000001d 3ffedb84 3ffeee7c 40235d41  
3ffffad0:  4020a278 3ffedb84 40234080 40235f3f  
3ffffae0:  4020a8b6 4020a8b1 40234080 4020a8b9  
3ffffaf0:  40100734 0000007d 00000099 60000200  
3ffffb00:  00000022 001640c8 74405004 601144c0  
3ffffb10:  32524556 00000000 ffff0001 0000000f  
3ffffb20:  53657649 596e6565 614e756f 0064656b  
3ffffb30:  00000000 00000000 00000000 00000000  
3ffffb40:  00030503 31000001 35373935 36353433  
3ffffb50:  00643232 00000000 00000000 00000000  
3ffffb60:  00000000 00000000 00000000 00000000  
3ffffb70:  00000000 00000000 00000000 00000000  
3ffffb80:  00000000 00000000 b9f91371 05ad12f0  
3ffffb90:  58656620 f0de327e a8cd6a35 2cb779b3  
3ffffba0:  3ce5f562 72aae400 00000000 00000000  
3ffffbb0:  ffffffff ffffffff ffff00ff ffffffff  
3ffffbc0:  0000000a 5f505345 35463242 00003736  
3ffffbd0:  00000000 00000000 00000000 00000000  
[... 151 lines cut ...]
3fffff30:  ffffffff ffffffff ffffffff ffffffff  
3fffff40:  ffffffff ffffffff ffffffff ffffffff  
3fffff50:  ffffffff ffffffff ffffffff ffffffff  
3fffff60:  ffffffff ffffffff ffffffff ffffffff  
3fffff70:  ffffffff ffffffff ffffffff ffffffff  
3fffff80:  ffffffff ffffffff ffffffff ffffffff  
3fffff90:  ffffffff 00000000 00000000 9f000000  
3fffffa0:  000203e9 40004928 ffffff01 55aa55aa  
================================================================

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 24980, room 16 
tail 4
chksum 0x60
load 0x3ffe8000, len 900, room 4 
tail 0
chksum 0xdf
load 0x3ffe8390, len 2668, room 8 
tail 4
chksum 0x9f
csum 0x9f

--- exit ---
```

slaff, 2017-07-18:
@yeralin Soooo... Since the WIFI and tcp stack need to be able to take control on a regular basis the ESP device has to be sure that no user program is occupying the processor for more than X seconds. Therefore there are two watchdog timers in ESP. The watchdog timers are like countdown timers. If they are not reset then they will take control and do their work. One is the software watchdog, that gets activated after around 1.5 - 3 seconds, that can be reset (fed), and the other one is the hardware watchdog which gets activated after 8 seconds.

This means that if you have loops you should take care to feed the software watchdog as long as you have operations that take more than 1.2 seconds. What I can recommend you is to  get yourself the [Kolban's ESP8266 book](https://leanpub.com/ESP8266_ESP32) and read the chapter titled **"Watchdog timer"**.

The other thing that you can do to help yourself is to decode that stack trace. We cannot do it because we do not have your binary program. But you have it. What you can do is save the stack trace into a file called crash.log. Then do the following:

```
python $SMING_HOME/../tools/decode-stacktrace.py out/build/app.out crash.log
```

Where `out/build/app.out` is the location of the binary file. Also make sure that the xtensa bin directory is in your PATH.  For me that is 

```
export PATH=$PATH:$ESP_HOME/xtensa-lx106-elf/bin/
```

I will close the ticket for now because it is not directly Sming related. Feel free to re-open it if you need something more.

closed 2017-07-18 by slaff

yeralin, 2017-07-20:
Ok I found a way. Thanks @slaff  for Kolban's book, very useful.

```
void rotateServo(os_event_t *arg) {
	Serial.println("Got here " + String(arg->par));
	servoCntrl->setDegree(-90);
	delay(730);
	servoCntrl->setDegree(90);
	delay(730);
	arg->par--;
	if (arg-> par > 0) {
		system_os_post(USER_TASK_PRIO_0, 0, arg->par);
	}
}

void dispenseFood(HttpRequest &request, HttpResponse &response)
{	
	int rotateTimes = 5;
	serialQueue = (os_event_t *) malloc(sizeof(os_event_t) * SERIAL_QUEUE_LEN);
	system_os_task(rotateServo, USER_TASK_PRIO_0, serialQueue, SERIAL_QUEUE_LEN);
	system_os_post(USER_TASK_PRIO_0, 0, rotateTimes);
	response.setContentType(ContentType::TEXT);
        response.sendString("Done");

}
```
Basically, here we are scheduling task `rotateServo`, but since we can't do that continuously 5 times (watchdog will fire after 2 seconds), we rotate servo, and scheduling another execution of `rotateServo` by using `system_os_post`. That way, we give control back to OS, and watchdog doesn't think that program stuck.