SmingHub/Sming #1186: Errors compiling MeteoControl

teicors, 2017-07-13:
Hi,
testing the new install from scratch of Sming and esp-open-sdk I get two errors:
1. during the build of spiffs: 
`Unable to write to spiffs file 'jquery.js', errno -10001.`
2. during the flash : 
`Running Cesanta flasher stub...
Flash params set to 0x0000
Writing 45056 @ 0x0... 45056 (100 %)
Wrote 45056 bytes at 0x0 in 3.9 seconds (92.1 kbit/s)...
Writing 299008 @ 0xa000... 299008 (100 %)
Wrote 299008 bytes at 0xa000 in 25.9 seconds (92.3 kbit/s)...
WARNING: Unlikely to work as data goes beyond end of flash. Hint: Use --flash_size
Writing 196608 @ 0x53000... 196608 (100 %)
Wrote 196608 bytes at 0x53000 in 17.0 seconds (92.3 kbit/s)...
Leaving...
python -m serial.tools.miniterm /dev/ttyUSB0 115200 
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
��������b␂*�م����0x000000b5, depc=0x00000000
Fatal exception (28): 
epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000`

The installation seems ok because the Basic_Blink works.

slaff, 2017-07-27:
> WARNING: Unlikely to work as data goes beyond end of flash.

@teicors What is the flash size of the ESP8266 device that you have?

teicors, 2017-07-28:
@slaff the chip is 4MB.
I fixed my error changing into Makefile-project.mk the size of the eprom from 512K to 4M. I've updated also the version esptool from 1.2 to 2.1 beta and when I try to flash I catch this error

`esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0xa000 for file: out/firmware/0x0a000.bin`

in the old version of esptool (1.2) I have no errors at all.

Comparing the Memory allocation from 512K and 4M i cannot find any difference, excepting the memory size of spiffs: is it correct ? 

`angelo@angelo-U36SG:/tmp/Sming/samples/MeteoControl$ export SPI_SIZE=512K
angelo@angelo-U36SG:/tmp/Sming/samples/MeteoControl$ make clean
angelo@angelo-U36SG:/tmp/Sming/samples/MeteoControl$ make
C+ app/webserver.cpp
C+ app/configuration.cpp
C+ app/application.cpp
AR out/build/app_app.a
LD out/build/app.out

#Memory / Section info:
------------------------------------------------------------------------------
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE83CD|     973
    rodata|           ReadOnly Data (RAM)|    3FFE83D0|    3FFEB348|   12152
       bss|      Uninitialized Data (RAM)|    3FFEB348|    3FFF2330|   28648
      text|            Cached Code (IRAM)|    40100000|    40107104|   28932
irom0_text|           Uncached Code (SPI)|    4020A000|    4025305C|  299100
Total Used RAM : 41773
Free RAM : 40147
Free IRam : 3854
------------------------------------------------------------------------------
# Generating image...
Generate firmware images successully in folder out/firmware.
Done
# Generating spiffs_bin
Checking for spiffs files
web directory exists. Creating out/firmware/spiff_rom.bin
Creating rom 'out/firmware/spiff_rom.bin' of size 0x30000 (196608) bytes.
Adding files in directory 'web'.
Added 'MeteoConfig.html' to spiffs (4748 bytes).
Added 'MeteoControl.html' to spiffs (1860 bytes).
Added 'MeteoAPI.html' to spiffs (2074 bytes).
Added 'bootstrap.css' to spiffs (108535 bytes).
Unable to write to spiffs file 'jquery.js', errno -10001.
out/firmware/spiff_rom.bin---------->0x54000
angelo@angelo-U36SG:/tmp/Sming/samples/MeteoControl$ export SPI_SIZE=4M
angelo@angelo-U36SG:/tmp/Sming/samples/MeteoControl$ make clean
angelo@angelo-U36SG:/tmp/Sming/samples/MeteoControl$ make
C+ app/webserver.cpp
C+ app/configuration.cpp
C+ app/application.cpp
AR out/build/app_app.a
LD out/build/app.out

#Memory / Section info:
------------------------------------------------------------------------------
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE83CD|     973
    rodata|           ReadOnly Data (RAM)|    3FFE83D0|    3FFEB348|   12152
       bss|      Uninitialized Data (RAM)|    3FFEB348|    3FFF2330|   28648
      text|            Cached Code (IRAM)|    40100000|    40107104|   28932
irom0_text|           Uncached Code (SPI)|    4020A000|    4025305C|  299100
Total Used RAM : 41773
Free RAM : 40147
Free IRam : 3854
------------------------------------------------------------------------------
# Generating image...
Generate firmware images successully in folder out/firmware.
Done
# Generating spiffs_bin
Checking for spiffs files
web directory exists. Creating out/firmware/spiff_rom.bin
Creating rom 'out/firmware/spiff_rom.bin' of size 0x80000 (524288) bytes.
Adding files in directory 'web'.
Added 'MeteoConfig.html' to spiffs (4748 bytes).
Added 'MeteoControl.html' to spiffs (1860 bytes).
Added 'MeteoAPI.html' to spiffs (2074 bytes).
Added 'bootstrap.css' to spiffs (108535 bytes).
Added 'jquery.js' to spiffs (86314 bytes).
out/firmware/spiff_rom.bin---------->0x54000`

slaff, 2017-07-28:
> Detected overlap at address: 

@teicors That is the problem. The second rom, starting at position 0xa000 is overlapping the first one. At the moment we don't have automatic adjustment of rom starting positions. But there is a workaround. Read the second last comment from #972. It contains instructions with things to do.

slaff, 2017-07-28:
Since issues with more or less the same problem started to pile up I have created a PR that has automatic adjustment built-in. Can you test if the solution is working for you?

```
cd %SMING_HOME%
git fetch origin pull/1208/head:test/pr
git checkout test/pr
```

And then compile your sample and report.

teicors, 2017-07-28:
fetched a new version, checkout test/pr as you wrote, compiled the Metecontrol and here's the result:

` angelo@angelo-U36SG  /tmp/Sming/samples/MeteoControl   test/pr ●  make flash
make[1]: ingresso nella directory "/tmp/Sming/Sming"
make[1]: uscita dalla directory "/tmp/Sming/Sming"
C+ app/webserver.cpp
C+ app/configuration.cpp
C+ app/application.cpp
AR out/build/app_app.a
LD out/build/app.out

#Memory / Section info:
------------------------------------------------------------------------------
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE83CD|     973
    rodata|           ReadOnly Data (RAM)|    3FFE83D0|    3FFEB348|   12152
       bss|      Uninitialized Data (RAM)|    3FFEB348|    3FFF2330|   28648
      text|            Cached Code (IRAM)|    40100000|    40107104|   28932
irom0_text|           Uncached Code (SPI)|    4020B000|    4025405C|  299100
Total Used RAM : 41773
Free RAM : 40147
Free IRam : 3854
------------------------------------------------------------------------------
# Generating image...
Generate firmware images successully in folder out/firmware.
Done
# Generating spiffs_bin
Checking for spiffs files
web directory exists. Creating out/firmware/spiff_rom.bin
Creating rom 'out/firmware/spiff_rom.bin' of size 0x80000 (524288) bytes.
Adding files in directory 'web'.
Added 'MeteoConfig.html' to spiffs (4748 bytes).
Added 'MeteoControl.html' to spiffs (1860 bytes).
Added 'MeteoAPI.html' to spiffs (2074 bytes).
Added 'bootstrap.css' to spiffs (108535 bytes).
Added 'jquery.js' to spiffs (86314 bytes).
out/firmware/spiff_rom.bin---------->0x55000
Killing Terminal to free /dev/ttyUSB0
/tmp/Sming/Sming/Makefile-project.mk:479: set di istruzioni per l'obiettivo "flash" non riuscito
make: [flash] Ucciso (ignorato)
/opt/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 1000000 write_flash -ff 40m -fm qio -fs 32m 0x00000 out/firmware/0x00000.bin 0xb000 out/firmware/0xb000.bin 0x55000 out/firmware/spiff_rom.bin
esptool.py v1.2
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0040
Writing 45056 @ 0x0... 45056 (100 %)
Wrote 45056 bytes at 0x0 in 0.9 seconds (412.8 kbit/s)...
Writing 303104 @ 0xb000... 303104 (100 %)
Wrote 303104 bytes at 0xb000 in 4.1 seconds (586.0 kbit/s)...
Writing 524288 @ 0x55000... 524288 (100 %)
Wrote 524288 bytes at 0x55000 in 6.9 seconds (609.2 kbit/s)...
Leaving...
python -m serial.tools.miniterm /dev/ttyUSB0 115200 
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
��r�$�d␓�␄␌�l`␃��s�l�l␛�␌␄␌l`␃��{�d␌ǟ␀�␄␌ld`␃;l�timeout updating: 70 -> 65535
Server timeout updating: 40 -> 0
fs.start: size:3740 Kb, offset:0x55000

mount res: 0`

Great ! Now I can compile and flash like a charm, but the webserver don't start (I don't know if it was required or not, I'll check the source): before to compile the soft I must set SPI_SIZE=4M.

closed 2017-07-29 by slaff

teicors, 2017-07-29:
I read the Sming 3.3.3 nonos and I downloaded, I can compile after changing the SPI_SIZE into Makefile.project, but Basic_Blink seems orkink only with esptool 1.2: using esptool 2.1 nothing happens. 

`angelo@angelo-U36SG:/opt/Sming/samples/Basic_Blink$ make clean
angelo@angelo-U36SG:/opt/Sming/samples/Basic_Blink$ make flash
C+ app/application.cpp
AR out/build/app_app.a
LD out/build/app.out

#Memory / Section info:
------------------------------------------------------------------------------
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE837E|     894
    rodata|           ReadOnly Data (RAM)|    3FFE8380|    3FFE8950|    1488
       bss|      Uninitialized Data (RAM)|    3FFE8950|    3FFEEB08|   25016
      text|            Cached Code (IRAM)|    40100000|    40105E33|   24115
irom0_text|           Uncached Code (SPI)|    40207000|    40233B3C|  183100
Total Used RAM : 27398
Free RAM : 54522
Free IRam : 8671
------------------------------------------------------------------------------
# Generating image...
Generate firmware images successully in folder out/firmware.
Done
(!) Spiffs support disabled. Remove 'DISABLE_SPIFFS' make argument to enable spiffs.
Killing Terminal to free /dev/ttyUSB0
/opt/Sming/Sming/Makefile-project.mk:479: set di istruzioni per l'obiettivo "flash" non riuscito
make: [flash] Ucciso (ignorato)
/opt/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 1000000 write_flash -ff 40m -fm qio -fs 32m 0x00000 out/firmware/0x00000.bin 0x7000 out/firmware/0x7000.bin
esptool.py v1.2
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0040
Writing 28672 @ 0x0... 28672 (100 %)
Wrote 28672 bytes at 0x0 in 0.6 seconds (416.5 kbit/s)...
Writing 184320 @ 0x7000... 184320 (100 %)
Wrote 184320 bytes at 0x7000 in 2.7 seconds (543.1 kbit/s)...
Leaving...
python -m serial.tools.miniterm /dev/ttyUSB0 115200 
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
��{�d�l␒�␄␌�d`␃��r�l�$␓�␄␌␄l ␃��{�l␄��␀�␌␄ld`␃sd�mode : softAP(1a:fe:34:d2:da:42)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100

--- exit ---
angelo@angelo-U36SG:/opt/Sming/samples/Basic_Blink$ vi /opt/Sming/Sming/Makefile-linux.mk angelo@angelo-U36SG:/opt/Sming/samples/Basic_Blink$ make clean
angelo@angelo-U36SG:/opt/Sming/samples/Basic_Blink$ make flash
C+ app/application.cpp
AR out/build/app_app.a
LD out/build/app.out

#Memory / Section info:
------------------------------------------------------------------------------
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE837E|     894
    rodata|           ReadOnly Data (RAM)|    3FFE8380|    3FFE8950|    1488
       bss|      Uninitialized Data (RAM)|    3FFE8950|    3FFEEB08|   25016
      text|            Cached Code (IRAM)|    40100000|    40105E33|   24115
irom0_text|           Uncached Code (SPI)|    40207000|    40233B3C|  183100
Total Used RAM : 27398
Free RAM : 54522
Free IRam : 8671
------------------------------------------------------------------------------
# Generating image...
Generate firmware images successully in folder out/firmware.
Done
(!) Spiffs support disabled. Remove 'DISABLE_SPIFFS' make argument to enable spiffs.
Killing Terminal to free /dev/ttyUSB0
/opt/Sming/Sming/Makefile-project.mk:479: set di istruzioni per l'obiettivo "flash" non riuscito
make: [flash] Ucciso (ignorato)
/opt/esp-open-sdk/esptool_2.1/esptool.py -p /dev/ttyUSB0 -b 1000000 write_flash -ff 40m -fm qio -fs 32m 0x00000 out/firmware/0x00000.bin 0x7000 out/firmware/0x7000.bin
WARNING: Flash size arguments in megabits like '32m' are deprecated.
Please use the equivalent size '4MB'.
Megabit arguments may be removed in a future release.
esptool.py v2.1-beta1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 1000000
Changed.
Configuring flash size...
Flash params set to 0x0040
Compressed 26544 bytes to 19492...
Wrote 26544 bytes (19492 compressed) at 0x00000000 in 0.2 seconds (effective 873.7 kbit/s)...
Hash of data verified.
Compressed 183100 bytes to 135109...
Wrote 183100 bytes (135109 compressed) at 0x00007000 in 2.2 seconds (effective 680.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
python -m serial.tools.miniterm /dev/ttyUSB0 115200 
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

--- exit ---
`

PR #1208 (2017-07-29): Added dynamic recalculation of image offsets to prevent rom overlapping.
Fixes: #1186, 
Fixes: #1158.