SmingHub/Sming #1188: Decode the stack dump

fernandomorse, 2017-07-13:
After error in running my program.

I tried using pytohn to debug, below 

> Stack dump:
> To decode the stack dump call from command line:
>    python $SMING_HOME/../tools/decode-stacktrace.py out/build/app.out
> and copy & paste the text enclosed in '===='.

I am using windows platform, while trying to run the command

python  C:\tools\sming\tools\decode-stacktrace.py out/build/app.out

Error is displayed 

> File "C:\tools\sming\tools\decode-stacktrace.py", line 15
>     print "Usage: \n\t%s <file.elf> [<error-stack.log>]" % sys.argv[0]
>                                                        ^
> SyntaxError: Missing parentheses in call to 'print'
> 
What am I doing wrong?

slaff, 2017-07-13:
Which is the python version that you have?

fernandomorse, 2017-07-13:
@slaff  Python 3.6.0

slaff, 2017-07-14:
@fernandomorse Ok, try the following:

```
cd $SMING_HOME
git fetch origin pull/1190/head:p3
git checkout p3
```

And tell me if the `decode-stacktrace.py` tool is working for you.

fernandomorse, 2017-07-14:
@slaff command executed , in the gitshell
git fetch origin pull/1190/head:p3

Error: fatal: Not a git repository (or any of the parent directories): .git

.. My plataform is Windows 10.

slaff, 2017-07-14:
> Error: fatal: Not a git repository (or any of the parent directories): .git

The git commands are valid in a directory that is under git version control. That means that you have to `cd` to such a directory before executing the command above. For Windows you can do

```
cd %SMING_HOME%
git fetch origin pull/1190/head:p3
git checkout p3
```

If it is not working for you ask a friend nearby to help you working with `git`.

slaff, 2017-07-14:
Or [download this file](https://raw.githubusercontent.com/slaff/Sming/0849f210219629d32e6b88cd76d81addd4ce643b/tools/decode-stacktrace.py) and replace with it the local `decode-stacktrace.py`

fernandomorse, 2017-07-14:
@slaff thanks, worked.

closed 2017-07-14 by fernandomorse