SmingHub/Sming #1195: Preparation for Release 3.4.0

slaff, 2017-07-21:
This issue is created to collect all the tasks we need to do before the 3.4.0.
This release will be targeting better Arduino compatibility.

### Source Code
- [ ] Update Sming/SmingCore/SmingCore.h (#1107)
- [ ] Update Readme.md (#1107)
- [ ] Merge develop into master
- [ ] Tag the source code in the master branch

### Refactoring 
- [ ] Fix the Wire param order to match Arduino (#1193)
- [ ] Add pgmspace.h (#1192)
- [ ] Use the ESP8266 directive and adjust the code respectively (#1192)
- [ ] Fix wrong GPIO reg type ( Update PR #398 and see if it can be merged )

Vity01, 2017-09-22:
@slaff The last check on refactoring topic can be marked as done - see https://github.com/SmingHub/Sming/pull/398

kwis2, 2017-09-24:
According to #1230 and #1239 it would be good idea to be able to disable stack dump in next release

slaff, 2017-10-10:
@kwis2 That is already possible. When you compile Sming and the application using Sming make sure to add the SMING_RELEASE=1 directive. For example:

```
cd $SMING_HOME
make dist-clean
make SMING_RELEASE=1
```

will compile library for you without debug statements and stack dumps.

See: https://github.com/SmingHub/Sming/blob/develop/Sming/Makefile#L185

closed 2017-11-07 by slaff