SmingHub/Sming #1207: Sample rboot now seems to need Wifistation.connect()

robertkendrick, 2017-07-27:
I'm running Sming-master-3.2.0 and SDK NonOS 2.0.0.
The sample Basic_rboot does not connect to the network when I issue the application's "connect" command. When I issue the application's "ip" command it is always 0.0.0.0. All other commands show relevant output.
If I edit the application source and add WifiStation.connect() to the code in the command handler it all works fine. I can "ota" and "switch" roms etc. See below

`		if (!strcmp(str, "connect")) {   // connect to wifi

			WifiStation.config(WIFI_SSID, WIFI_PWD);
			WifiStation.enable(true);
			WifiStation.connect(); 	// added by me
		} 
`

Is this a change in the WifiStation API behaviour ?

slaff, 2017-07-28:
@robertkendrick May be that is the case with new(er) SDKs. Having `WifiStation.connect()` does not hurt therefore can you make a PR with your change?

robertkendrick, 2017-08-02:
PR #1215 done.

closed 2017-08-03 by slaff