SmingHub/Sming #1214: Serial callback problem

kwis2, 2017-08-01:
When I send garbage data to serial with serial module work unstable. I have tested sample project and there is the same behaviour. For instance with Basic_rBoot example it is enought to send `qwertyuiopasdfghjklzxcvbnm` string to fool module. It looks like there is problem with flushing data.

below is log with 
```
Serial.println(str);
stream.flush();
```
added into serialCallBack.

> ->qwertyuiopasdfghjklzxcvbnm
> qwertyuiopas
> unknown command
> ->help
> dfghjk[10]@ Q[10]@
> unknown command
> ->help
> lzxcvb[10]@ Q[10]@
> unknown command
> ->help
> nm
> unknown command
> ->help
> lp
> unknown command
> ->help
> lp
> unknown command
> ->help
> lp
> unknown command
> ->help
> lp
> unknown command
> ->help
> lp
> unknown command
> ->help
> lp
> unknown command
> ->help
> lp
> unknown command

slaff, 2017-11-22:
@kwis2 Can you check if that PR #1290 fixes the reported problem?

kwis2, 2017-11-25:
@slaff unfortunatelly I am not able to install current version of sming #1241

PR #1290 (2017-12-04): Added Stream::indexOf(char c) that finds a character in a stream
without advancing the internal stream pointers.

Fixes: #1214.
closed 2017-12-04 by slaff