SmingHub/Sming #1216: Docker after make flash - Serial Output Garbage

igormorse, 2017-08-05:
Hello folks, first of all I going to put here several Screenshots.

I'm currently using the [attachix container](https://hub.docker.com/r/attachix/c9-esp8266-sdk/) on Docker to setup my development environment.

Everything is working fine except for this:

First one its my main problem that is the Serial Output after running make flash.

[image]

I can't exit this Python Script too, I need to unplug my device to provoke an Exception in the Script.

How I can prevent all this output gargabe ? I'm sure this is some missconfiguration. 

There is an better way of read the serial output on this Container from my ESP8266 ?

Thanks,
Igor Morse.

slaff, 2017-08-05:
@igormorse Yes, you should [initialize the device](https://github.com/SmingHub/Sming/wiki/Application-prints-random-symbols-and-restarts) (only needed once). The steps are:

```
cd <your-favourite-sample>
export SPI_SIZE=4M  # make sure to put the right flash size of your ESP device
esptool.py -p /dev/ttyUSB0 -b 115200 erase_flash # run it only once 
make flashinit  # again run it only once
```

And then you can upload the sample application
```
make # compile the sample
make flash # ... and flash it... This time things should work as expected
```

igormorse, 2017-08-06:
Thanks that worked just fine except for the command for exit the script that it seems to not work inside the c9 terminal.

Anyway to quit this script without the Ctrl + ] input ?

I did read the Wiki but sorry for not paying attention on the troubleshooting section.

igormorse, 2017-08-23:
You need to rebind the Ctrl-] from the Cloud9 Key Biddings to be able to stop the Python Application. That solved it.

closed 2017-08-23 by igormorse