SmingHub/Sming #1219: Possible TLS regression: out of memory on TLS handshake

MayaPosch, 2017-08-09:
Here at the office we have a number of nodes running with older Sming-based firmware images which connect using TLS to a HAProxy with the TLS fragment size set to 2 kB. This has so far allowed us to use TLS and MQTT without any issues.

Since building new firmware images, and updating to the current Sming develop branch, the TLS handshake is failing again with an out of memory error from axTLS:

TcpConnection::connect result:, 0
OnConnected
staticOnConnected: useSSL: 1, Error: 0
SSL: Starting connection...
SSL: Switching to 160 MHz
SSL: handshake start (1474 ms)
SSL: Session Id Length: 0
please start sntp first !
SSL: handshake is in progress...
TCP sent: 111
please start sntp first !
E:M 536

After this a software watchdog reset occurs, with the last lines of the stacktrace being:

0x40232ec4: alloc at /opt/Sming/Sming/third-party/axtls-8266/crypto/bigint.c:1106
0x40100de4: pvPortRealloc at ??:?
0x40100dfd: pvPortRealloc at ??:?
0x40232ec4: alloc at /opt/Sming/Sming/third-party/axtls-8266/crypto/bigint.c:1106
0x4025723c: ax_port_realloc at /opt/Sming/Sming/third-party/axtls-8266/replacements/mem.c:69
0x40232ec4: alloc at /opt/Sming/Sming/third-party/axtls-8266/crypto/bigint.c:1106
0x40232cc7: more_comps at /opt/Sming/Sming/third-party/axtls-8266/crypto/bigint.c:1072

This looks virtually identical to the previous issue we were having before reducing the TLS fragment size using HAProxy, namely of axTLS running out of heap space (nearly 30 kB before the handshake starts).

I looked around, but couldn't find anyone else with a similar issue. Is something wonky about my setup, or is this is a known issue?

ADiea, 2017-08-10:
Hi here is a internal picture of the workings
in __third-party/axtls-8266/ssl/tls1.c__ there is this function 
```int increase_bm_data_size(SSL *ssl, size_t size)```
Initially the ssl->bm_all_data is malloc'ed with a minimum required size
When a handshake is done the ssl->bm_all_data is increased using this function realloc'ed to the required size mak 16kB
You need to make sure this has not changed somehow and is still working as expected
1. that initially ssl->bm_all_data is malloc to a small size - how much?
2. that the increase_bm_data is called with a valid required size max 2kB in your case
Please report back with your findings on your application. It could be that the behavior is changed and it is now always malloc'ing 16kB regardless of fragment size.

MayaPosch, 2017-09-02:
Thank you for your answer. Unfortunately modifying the axTLS code or to remotely attach gdb isn't as easy as thought, so I am having trouble obtaining the requested information.

MayaPosch, 2017-12-04:
Seems like it's a known issue by now? https://github.com/esp8266/Arduino/issues/3490

Will there be a fix, or do I have to wait until axTLS is replaced/find an alternative myself?

slaff, 2020-01-30:
@MayaPosch  With the latest `develop` version we have support for both axTLS and BearSSL. You can read more from here: https://sming.readthedocs.io/en/latest/_inc/Sming/Components/ssl/index.html. If the problem is still there feel free to reopen the issue.

closed 2020-01-30 by slaff