SmingHub/Sming #1221: Third-party PWM library incompatible with HardwarePWM class

MayaPosch, 2017-08-13:
By default Sming is built with the third-party PWM library by Stefan Bruens. As the README for this library notes:

> By default there is one small difference to the SDK. The code uses a unit of 200ns for both period and duty. E.g. for 10% duty cycle at 1kHz you need to specify a period value of 5000 and a duty cycle value of 500, a duty cycle of 5000 or above switches the channel to full on.
> 
> To have full compatibility with the SDK, you have to set the SDK_PWM_PERIOD_COMPAT_MODE define to 1. 

As a result of this, HardwarePWM no longer works as expected, being initialised to 5 kHz by default instead of 1 kHz, and with the duty not working as normal either. It might be desirable for this third-party library to use the SDK behaviour.

For now I have added this parameter to the central Sming Makefile (custom PWM section, appended to CFLAGS), but I'm not sure what the preferred solution would be.

Thoughts?

slaff, 2017-08-14:
@MayaPosch Can you create a pull request with the suggested changes?

MayaPosch, 2017-09-01:
I have created a new pull request: #1237

slaff, 2017-09-07:
Fixed with the merge of #1237.

closed 2017-09-07 by slaff