SmingHub/Sming #1223: MacOS Quickstart fails

gjt211, 2017-08-18:
Hi, I have followed the MacOS quickstart guide to the letter and am unable to get the environment working.

I get to the **Get & Build Sming Core** section and enter `sudo make` and it ends with some errors.
```
make[2]: /opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc: Malformed Mach-o file
make[2]: *** [lwip/core/def.o] Error 1
make[1] *** [compiler/lib/liblwip_open.a] Error 2
make: *** [reload] Error 2
```

further up there is also an error after
```
Submodule path 'third-party/spiffs' : checked out
error: unrecognised input
make: [third-party/spiffs/makefile] Error 128 (ignored)
```

Any help or guidance is appreciated.

gjt211, 2017-08-18:
I wanted to provide an update to this - still unsuccessful!
On a windows 7 machine, I installed VirtualBox and Lubuntu.
I then followed the Linux Quickstart guide, and got as far as
```
cd Sming/Sming
make
```
Pretty much the same as on MacOS. Pretty much identical errors as above.
This is doing my head in. It should not be this difficult.
Any help anyone can provide is welcome.

IanAber, 2017-10-08:
I have exactly the same problem. Has anybody been able to get past this yet?

gjt211, 2017-10-08:
Hi @IanAber 
I still have not been successful. I have given up on this and moved to other platforms. It is obvious that nobody is able (or wants) to help newbies. I wanted to try this to see if it could be a new platform for development, and I had high hopes after looking at what it 'could' do. But it seems I won't ever know.
Sorry for the negative comments.

kwis2, 2017-10-09:
From my experience without advanced knowledge there is no way to install Sming on windows 32 bit, I failed to install on windows 10 - neither direct install nor docker. on windows 7 with direct install I was succesfull on only one machine, and failed with docker. Idea to install on linux is not for newbies because of errors in manual - for instance command `cd $SMING_HOME/../..` while there is no `$SMING_HOME` yet
Idea to install VirtualBox VM failed because esptool hangs while programming on my computers.
Even if you success with install try to compile not Basic_blink as in manual, but Basic_WebClient - this is real test if Sming works.
In my opinion Sming it could be very good but now it is dying software.

gjt211, 2017-12-20:
@kwis2, I revisited this just now and still no change.
It seems there is work happening on the Sming project, but it still seems like nobody cares about getting any new users. It has now been 4 months and still no reply to my initial problem, which others are also having.

Maybe your last statement about dying software is true. With the kind of support we are receiving it will certainly die.

I dont expect something for nothing, I am happy to click the "Paypal/Donate" button for software that I use and is supported.
At least the Arduino/ESP stuff actually works, and they respond quickly to bug submissions. Not always providing a solution, but at least there is visible progress and help.
Just my 2c worth.
I agree that Sming 'could' be very good but I can't even test it to find out.

kwis2, 2017-12-20:
@gjt211 I am only windows user with silmilar problems as your. I think @slaff is proper person to ask questions. 
In fact there was a lot of work made to prepare Sming, there are a lot of bugs solved, but it fails with just begining - instalation. On windows I found that solution could be if you install very old version of Sming and then update to newest one as in #1297

slaff, 2018-02-05:
@gjt211 @IanAber We have updated the Mac OS X getting started guide and the precompiled Mac OS X toolchain.  Can you try to follow the steps in the [document](https://github.com/SmingHub/Sming/wiki/MacOS-Quickstart) and report your progress?

gjt211, 2018-02-06:
@slaff , Thanks for the updated getting started guide.
I have started using the guide on a different MAC that has not had the install performed before.
1. Installed xcode tools successfully.
2. Installed homebrew successfully.
3. I did not install Eclipse.
4. Install brew/dupes failed!
```
Warning: homebrew/dupes was deprecated. This tap is now empty as all its formulae were migrated
```
I have not tried going any further.
Your advice is appreciated.
Thanks

slaff, 2018-02-06:
> 4. Install brew/dupes failed!

Thanks for the feedback so far. This is a warning only and we will see how to fix it. But that should not stop you from trying further. So, please, go on and report your progress. And thanks in advance.

slaff, 2018-02-20:
@gjt211 Any progress on this?

gjt211, 2018-06-28:
@slaff sorry for not getting back to this sooner. My paying job keeps me away for extremely extended periods of time.

I have just been through this again and I have been able to complete the install (with some errors in there that I don't understand) but have just now been able to compile the Basic_Blink example.
The new install guide was still not as simple as follow the steps but I got there.

Thanks for this and hopefully I can net get started testing and playing as time permits.

gjt211, 2018-06-28:
@slaff. I have been able to compile some examples but my any serial output is just garbage. I am using a ESP8266 NodeMCU board that works fine in the Arduino IDE.
I was wondering if there is a crystal frequency setting but was unable to find one (I think the ESP12 module uses a 26Mhz Xtal)
I have tried changing the baud rates without success. Are you able to point me in the right direction?

gjt211, 2018-06-28:
After hunting and searching I think my problem has been solved now.
I done the following;
```
export SPI_SIZE=4M
cd <path to my sming code>
make flashinit
```
I then done a `make` followed by a `make flash` and my code is running!
I have taken some notes of my experience and will be happy to make a guide when time permits which may help others using MacOS.

closed 2018-12-18 by gjt211