SmingHub/Sming #1227: decode-stacktrace.py doesn't work on Arch Linux

max-weller, 2017-08-20:
...because on Arch the default python version is 3.x, and decode-stacktrace.py requires 2.x, however the shebang line in the script references the unversioned python binary.

This can be fixed by PR #1225.

slaff, 2017-08-21:
@max-weller decode-stacktrace should work also with python3. Please, paste here the issues that you have. A proper fix IMHO would be making the script work also with python3 and not force it to use always python2.

max-weller, 2017-08-21:
That'd be the clean solution I guess :D

max-weller, 2017-08-21:
I get the following error after pasting a stack trace:
```
Traceback (most recent call last):
  File "tools/decode-stacktrace.py", line 50, in <module>
    pipe.stdin.write(line)
TypeError: a bytes-like object is required, not 'str'
```

max-weller, 2017-08-21:
OK it works now with Python3, I made a new PR  #1228

slaff, 2017-08-21:
Should be fixed with PR rebase #1229.

closed 2017-08-21 by slaff