SmingHub/Sming #1232: Bridge UART <-> TCP

fvpalha, 2017-08-25:
Hi, someone made a bridge (TCP <-> Serial) with Sming 3.3.0 ?

ADiea, 2017-08-28:
Hi @fvpalha  how is this supposed to work like SLIP? - this is only ip encapsulated over serial not tcp 
Do you have an example project? Are you referring to the AT commands firmware from Espressif?

fvpalha, 2017-08-28:
Hi @ADiea . I need, over TCP, send bytes to UART and receive bytes from UART. 
I am looking for a little sample.

ADiea, 2017-08-29:
This could help https://github.com/pvvx/esp8266web Forum thread is here https://esp8266.ru/forum/threads/razrabotka-biblioteki-malogo-webservera-na-esp8266.56/ in russian use google translate. Maybe you can extract the tcp2uart part from that project if possible needs investigation

fvpalha, 2017-08-29:
Thank you @ADiea