SmingHub/Sming #1233: SED not working on Mac OS

johndoe8967, 2017-08-26:
Hello,

since commit "Added dynamic recalculation of image offsets to prevent rom overlapping. #1208"
there is a "sed -r" in the makefile-project.

This is not supported on mac os.
I tried to fix it by exchanging it to "sed -E" but then I get segfaults because of wrong calculation of the ROM addresses

slaff, 2017-10-11:
> This is not supported on mac os.

Any idea what is the alternative for Mac OS?

slaff, 2017-12-09:
@johndoe8967 Can you check PR #1298 also on Mac OS?

johndoe8967, 2017-12-18:
@slaff sure, id did check but it doesn't address the problem that -r option is not allowed on Mac OS

slaff, 2018-01-26:
@johndoe8967 Did you install `gnu sed` with a command similar to the one below ?

```
brew install gnu-sed --with-default-names
```

This will make sure that the `sed` on your OS is the `GNU sed` application which the Makefile is expecting to have.

kiralikbeyin, 2019-08-07:
i uninstalled and 
brew install gnu-sed --with-default-names

Error: invalid option: --with-default-names