SmingHub/Sming #1234: Basic_WebClient with ssl crashes

icodk, 2017-08-26:
Basic_WebClient from samples, sming compiled with ENABLE_SSL=1

Compile OK 
Download  OK
When run get the following output:

ip:192.168.1.19,mask:255.255.255.0,gw:192.168.1.1
ip:192.168.1.19,mask:255.255.255.0,gw:192.168.1.1

Connected. Got IP: 192.168.1.19
Creating new httpConnection
HttpConnection::connect: TCP state: -1, isStarted: 0, isActive: 0
HttpConnection::connecting ...

+TCP connection
connect to: www.attachix.com
HttpConnection::connect: TCP state: 0, isStarted: 1, isActive: 0
HttpConnection::connect: TCP state: 0, isStarted: 1, isActive: 0
HttpConnection::connect: TCP state: 0, isStarted: 1, isActive: 0
HttpConnection::connect: TCP state: 0, isStarted: 1, isActive: 0
DNS record found: www.attachix.com = 87.106.90.173
TcpConnection::connect result:, 0
OnConnected
staticOnConnected: useSSL: 1, Error: 0
SSL: Starting connection...
SSL: Switching to 160 MHz
SSL: handshake start (1409 ms)
SSL: Session Id Length: 0
please start sntp first !
SSL: handshake is in progress...
TCP sent: 103
please start sntp first !

***** Fatal exception 3
pc=0x402302e7 sp=0x3ffff960 excvaddr=0x4023afc1
ps=0x00000030 sar=0x00000019 vpri=0x40235f49
r00: 0x402302d0=1076036304 r01: 0x3ffff960=1073740128 r02: 0x4023afc1=1076080577 
r03: 0x000000ca=       202 r04: 0x000000d1=       209 r05: 0x3fff4da4=1073696164 
r06: 0x00000004=         4 r07: 0x00000004=         4 r08: 0x4023af16=1076080406 
r09: 0x3fff4da4=1073696164 r10: 0x3ffffb1d=1073740573 r11: 0x8837d5db=-2009606693 
r12: 0x00000004=         4 r13: 0x4023af00=1076080384 r14: 0x4023af00=1076080384 
r15: 0x3fff4db0=1073696176 

build on  Windows
Sming 3.3.0 
SDK ESP8266_NONOS_SDK_V1.5.3_16_04_16

ADiea, 2017-08-28:
@icodk please attach stack trace by running the python script with the stack dump from console.

icodk, 2017-08-28:
@ADiea 
Tried to do a stack trace but failed:
I entered:
`C:\tools\Sming_3_3_0\tools>python decode-stacktrace.py C:\tools\Sming_3_3_0\samples\Basic_WebClient\out\build\app_0.out`
Then I paste the output and get:

``3fffff10:  3ffed162 3fff0f08 3fff1e20 40223e3d
Traceback (most recent call last):
  File "decode-stacktrace.py", line 40, in <module>
    data = input()
  File "<string>", line 1
    3fffff10:  3ffed162 3fff0f08 3fff1e20 40223e3d
           ^
SyntaxError: invalid syntax
``
C:\tools\Sming_3_3_0\tools>3fffff20:  003c4288 3fff1b20 3fff1b20 3ffefe88
'3fffff20:' is not recognized as an internal or external command,
operable program or batch file.
``
C:\tools\Sming_3_3_0\tools>3fffff30:  00000000 3fff1e20 0000001c 3ffed154
'3fffff30:' is not recognized as an internal or external command,
operable program or batch file.``
``
And lots of lines with the
...is not recognized as an internal or external command,
operable program or batch file.

I also gets two stack dumps
one on Fatal exception 3
and one on Software Watchdog Reset
(Attached)
[stackdump.txt](https://github.com/SmingHub/Sming/files/1256610/stackdump.txt)

icodk, 2017-08-28:
Basic_ssl gives similar results (**** Fatal exception 3)

icodk, 2017-08-28:
Found out that if I compile in docker it runs (both Basic_webClient and Basic_ssl) without isues
I can also see that there is a different between the two compilations results in regards to sizes.
Here are the two results:
Docker compilation:

bash-4.3# make ENABLE_SSL=1
OC out/build/libmain2.a
make -C /workspace/Sming/Sming/third-party/rboot RBOOT_GPIO_ENABLED=0
make[1]: Entering directory '/workspace/Sming/Sming/third-party/rboot'
CC rboot-stage2a.c
LD /host/Basic_WebClient/out/build/rboot-stage2a.elf
E2 /host/Basic_WebClient/out/build/rboot-hex2a.h
CC rboot.c
LD /host/Basic_WebClient/out/build/rboot.elf
E2 /host/Basic_WebClient/out/firmware/rboot.bin
make[1]: Leaving directory '/workspace/Sming/Sming/third-party/rboot'
C+ app/application.cpp
CC /workspace/Sming/Sming/third-party/rboot/appcode/rboot-api.c
CC /workspace/Sming/Sming/third-party/rboot/appcode/rboot-bigflash.c
AR out/build/app_app.a
LD out/build/app_0.out
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8388|     904
    rodata|           ReadOnly Data (RAM)|    3FFE8390|    3FFEA984|    9716
       bss|      Uninitialized Data (RAM)|    3FFEA988|    3FFF0F60|   26072
      text|            Cached Code (IRAM)|    40100000|    40107168|   29032
irom0_text|           Uncached Code (SPI)|    40202010|    40249CF8|  294120
Total Used RAM : 36692
Free RAM : 45228
Free IRam : 3754
E2 out/firmware/rom0.bin
(!) Spiffs support disabled. Remove 'DISABLE_SPIFFS' make argument to enable spiffs.
bash-4.3#

Windows compilation:

C:\Dev\shared_with_docker\Basic_WebClient>make ENABLE_SSL=1
OC out/build/libmain2.a
make -C /C/tools/Sming/Sming/third-party/rboot RBOOT_GPIO_ENABLED=0
make[1]: Entering directory `/C/tools/Sming/Sming/third-party/rboot'
CC rboot-stage2a.c
LD /c/Dev/shared_with_docker/Basic_WebClient/out/build/rboot-stage2a.elf
E2 /c/Dev/shared_with_docker/Basic_WebClient/out/build/rboot-hex2a.h
CC rboot.c
LD /c/Dev/shared_with_docker/Basic_WebClient/out/build/rboot.elf
E2 /c/Dev/shared_with_docker/Basic_WebClient/out/firmware/rboot.bin
make[1]: Leaving directory `/C/tools/Sming/Sming/third-party/rboot'
C+ app/application.cpp
CC /C/tools/Sming/Sming/third-party/rboot/appcode/rboot-api.c
CC /C/tools/Sming/Sming/third-party/rboot/appcode/rboot-bigflash.c
AR out/build/app_app.a
LD out/build/app_0.out
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8388|     904
    rodata|           ReadOnly Data (RAM)|    3FFE8390|    3FFEAA88|    9976
       bss|      Uninitialized Data (RAM)|    3FFEAA88|    3FFF0FF0|   25960
      text|            Cached Code (IRAM)|    40100000|    4010653B|   25915
irom0_text|           Uncached Code (SPI)|    40202010|    40245494|  275588
Total Used RAM : 36840
Free RAM : 45080
Free IRam : 6871
E2 out/firmware/rom0.bin
(!) Spiffs support disabled. Remove 'DISABLE_SPIFFS' make argument to enable spiffs.

C:\Dev\shared_with_docker\Basic_WebClient>

WHATS GOING ON ?

mtzfactory, 2017-09-12:
Hi,

I have the same problem with **MqttClient_Hello**, with _SSL activated_, using the `develop` branch.

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 1572, room 16 
tail 4
chksum 0xfb
load 0x3ffe8000, len 776, room 4 
tail 4
chksum 0xa9
csum 0xa9

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: GPIO skip mode (16)

Booting rom 0.
å„.Ï.ÇgÏísÉÛg|Ï.d.ddd`.å„{ìlå˛Station configuration is: Orange-B853
mode : sta(5c:cf:7f:06:f1:de)
add if0
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with Orange-B853, channel 2
dhcp client start...
connect to ssid Orange-B853, channel 2

ip:192.168.1.111,mask:255.255.255.0,gw:192.168.1.1
ip:192.168.1.111,mask:255.255.255.0,gw:192.168.1.1

MQTT start connection
+TCP connection
connect to: test.mosquitto.org
Storing 77 bytes in stream
timeout updating: 70 -> 65535
subscription 'main/status/#' registered
realloc 77 -> 226
Storing 20 bytes in stream
DNS record found: test.mosquitto.org = 37.187.106.16
TcpConnection::connect result:, 0
OnConnected
staticOnConnected: useSSL: 1, Error: 0
SSL: Starting connection...
SSL: Switching to 160 MHz
SSL: handshake start (1203 ms)
SSL: Session Id Length: 0
please start sntp first !
SSL: handshake is in progress...
TCP sent: 105
please start sntp first !

***** Fatal exception 3
pc=0x4000df1b sp=0x3ffff5c0 excvaddr=0x4023e1b0
ps=0x00000030 sar=0x00000018 vpri=0x00002440
r00: 0x402392e5=1076073189 r01: 0x3ffff5c0=1073739200 r02: 0x3ffff7c1=1073739713 
r03: 0x4023e1b0=1076093360 r04: 0x0000000f=        15 r05: 0x3ffff7c1=1073739713 
r06: 0x00001f88=      8072 r07: 0x00000000=         0 r08: 0x74746975=1953786229 
r09: 0x726f2e6f=1919889007 r10: 0x3fff3b58=1073691480 r11: 0x00000003=         3 
r12: 0x3ffff710=1073739536 r13: 0x0000000f=        15 r14: 0x0000007f=       127 
r15: 0x4023e1b0=1076093360 

Stack dump:
To decode the stack dump call from command line:
   python $SMING_HOME/../tools/decode-stacktrace.py out/build/app.out
and copy & paste the text enclosed in '===='.
================================================================
3ffff5c0:  4000df1b 00000030 00000018 00002440  
3ffff5d0:  402392e5 3ffff7c1 4023e1b0 0000000f  
3ffff5e0:  3ffff7c1 00001f88 00000000 74746975  
3ffff5f0:  726f2e6f 3fff3b58 00000003 3ffff710
```

Any solution?

slaff, 2017-10-13:
I can confirm the issue. That is again due to the SSL max fragment size OR SSL server certificate length on the server. If you change the domain to something like `attachix.com` the example above will work for you. Give it a try. 

> .. he Basic_Ssl example works as well... it connects to https://www.google.com:443/ with no problems.

Google is kind enough to send as much as needed for the SSL handshake and nothing more. Plus they have the SSL max fragment size, etc.

icodk, 2017-10-24:
@slaff Tried with attachix.com but the same results(see below)
If I used the espconn_secure set of functions against the same servers there are no such issues.
```Creating new httpConnection
HttpConnection::connect: TCP state: -1, isStarted: 0, isActive: 0
HttpConnection::connecting ...
+TCP connection
connect to: attachix.com
Connecting with DHCP
DNS record found: attachix.com = 87.106.90.173
TcpConnection::connect result:, 0
OnConnected
staticOnConnected: useSSL: 1, Error: 0
SSL: Starting connection...
SSL: Switching to 160 MHz
SSL: handshake start (53628 ms)
SSL: Session Id Length: 0
please start sntp first !
SSL: handshake is in progress...
TCP sent: 99
please start sntp first !

***** Fatal exception 3
pc=0x40236c03 sp=0x3ffff960 excvaddr=0x40242e03
ps=0x00000030 sar=0x00000019 vpri=0x4023c865
r00: 0x40236bec=1076063212 r01: 0x3ffff960=1073740128 r02: 0x40242e03=1076112899 
r03: 0x000000d7=       215 r04: 0x000000ae=       174 r05: 0x3fff86d4=1073710804 
r06: 0x00000004=         4 r07: 0x00000004=         4 r08: 0x40242d81=1076112769 
r09: 0x3fff86d4=1073710804 r10: 0x3ffffb1d=1073740573 r11: 0xa83ad279=-1472540039 
r12: 0x00000004=         4 r13: 0x40242d10=1076112656 r14: 0x40242d10=1076112656 
r15: 0x3fff86e0=1073710816 

Stack dump:
To decode the stack dump call from command line:
   python $SMING_HOME/../tools/decode-stacktrace.py out/build/app.out
and copy & paste the text enclosed in '===='.
================================================================
3ffff960:  40236c03 00000030 00000019 4023c865  
3ffff970:  40236bec 40242e03 000000d7 000000ae  
3ffff980:  3fff86d4 00000004 00000004 40242d81  
3ffff990:  3fff86d4 3ffffb1d a83ad279 00000004  
```

icodk, 2017-10-25:
I also tried to build Basic_Ssl with Sming built in Docker, it linked OK but the error persist. Differences in the build process (on Windows vs Docker) is the SDK and the compiler as far as I can see.
I also noticed that in the failing program it only sends the first 99 bytes  vs  99 and then 358 in the program that works. Here are the two outputs:
Failing program output (Windows): 
```
Connected. Got IP: 192.168.1.29
Free Heap: 42080
CPU Frequency: 160 MHz
Creating new httpConnection
HttpConnection::connect: TCP state: -1, isStarted: 0, isActive: 0
HttpConnection::connecting ...
+TCP connection
connect to: attachix.com
DNS record found: attachix.com = 87.106.90.173
TcpConnection::connect result:, 0
OnConnected
staticOnConnected: useSSL: 1, Error: 0
SSL: Starting connection...
SSL: Switching to 160 MHz
SSL: handshake start (1406 ms)
SSL: Session Id Length: 0
please start sntp first !
SSL: handshake is in progress...
TCP sent: 99
please start sntp first !

***** Fatal exception 3
pc=0x402302e7 sp=0x3ffff960 excvaddr=0x4023b085
ps=0x00000030 sar=0x00000019 vpri=0x40235f49
r00: 0x402302d0=1076036304 r01: 0x3ffff960=1073740128 r02: 0x4023b085=1076080773 
r03: 0x0000000a=        10 r04: 0x000000d0=       208 r05: 0x3fff424c=1073693260 
r06: 0x00000004=         4 r07: 0x00000004=         4 r08: 0x4023b073=1076080755 
r09: 0x3fff424c=1073693260 r10: 0x3ffffb1d=1073740573 r11: 0x6a476c00=1783065600 
r12: 0x00000004=         4 r13: 0x4023afb0=1076080560 r14: 0x4023afb0=1076080560 
r15: 0x3fff4258=1073693272 

Stack dump:
To decode the stack dump call from command line:
   python $SMING_HOME/../tools/decode-stacktrace.py out/build/app.out
and copy & paste the text enclosed in '===='.
================================================================
3ffff960:  402302e7 00000030 00000019 40235f49  
3ffff970:  402302d0 4023b085 0000000a 000000d0  
3ffff980:  3fff424c 00000004 00000004 4023b073  

```
Successful program output (from Docker)
```
Connected. Got IP: 192.168.1.29
Free Heap: 42176
CPU Frequency: 160 MHz
Creating new httpConnection
HttpConnection::connect: TCP state: -1, isStarted: 0, isActive: 0
HttpConnection::connecting ...
+TCP connection
connect to: attachix.com
DNS record found: attachix.com = 87.106.90.173
TcpConnection::connect result:, 0
OnConnected
staticOnConnected: useSSL: 1, Error: 0
SSL: Starting connection...
SSL: Switching to 160 MHz
SSL: handshake start (1210 ms)
SSL: Session Id Length: 0
please start sntp first !
SSL: handshake is in progress...
TCP sent: 99
please start sntp first !
TCP sent: 358
SSL: Handshake done (1405 ms).
SSL: Switching back to 80 MHz
TCP connected
HttpConnection::onReadyToSendData: waitingQueue.count: 1
Storing 36 bytes in stream
realloc 36 -> 197
Storing 32 bytes in stream
Storing 20 bytes in stream
Storing 2 bytes in stream
onReadyToSendData: 0
Written: 90, Available: 90, isFinished: 1, PushCount: 1 [TcpBuf: 2631]
TcpClient request completed
TCP sent: 149
HttpConnection::onReadyToSendData: waitingQueue.count: 0
onReadyToSendData: 2
Written: 24, Available: 24, isFinished: 1, PushCount: 1 [TcpBuf: 2695]
TcpClient request completed
SSL: Decrypted data len 4096
The headers are complete
TCP received: 4096 bytes
HttpConnection::onReadyToSendData: waitingQueue.count: 0
Nothing in the waiting queue
onReadyToSendData: 1
TCP sent: 85
HttpConnection::onReadyToSendData: waitingQueue.count: 0
Nothing in the waiting queue
onReadyToSendData: 2
SSL: Decrypted data len 4096
TCP received: 4096 bytes
HttpConnection::onReadyToSendData: waitingQueue.count: 0
Nothing in the waiting queue
onReadyToSendData: 1
SSL: Decrypted data len 2951
staticOnMessageComplete: Execution queue: 0, https://attachix.com:443/
Got response code: 200
Got content starting with: <!DOCTYPE HTML>
<html>
<head>
    <meta http-equiv
Success: 1
```

closed 2017-11-13 by slaff