SmingHub/Sming #1236: HttpRequest body not being populated in HttpServerConnection in v3.3.0

jcarterof, 2017-08-30:
When HttpServer receives POST request with JSON (or any bytes) in the body of the message, that data does not make it into the HttpRequest object.  HttpRequest has three setBody() methods and as far as I can tell none of them is ever called.  I modified the HttpServerConnection::staticOnBody() method, adding the line:

`connection->request.setBody(String(at, length));`

right below the null connection check.  This makes it work properly for my purposes (REST using JSON) but is not likely correct for other cases.

ericogr, 2017-09-08:
@jcarterof 

Same problem here...
To workaround this issue, I set a jsonparser and used a setBody.

```cpp
_HttpServer->setBodyParser("application/json", HttpBodyParserDelegate(&DeviceServerConfiguration::parseJson,this));

//...

void DeviceServerConfiguration::parseJson(HttpRequest& request, const char *at, int length) {
    if (length > 0) {
        request.setBody(String(at, length));
    }
}
```
**This problem didn't happen in version 3.2.0**

slaff, 2017-09-15:
HttpServer does not store the HTTP body of a request in a String for a reason. And the reason is limited RAM. If it is desired to store all body data in a String then we can do the following:

- Modify the `HttpServer::setBodyParser` command to understand wildcard mime types. For example `*/*` should be triggered always and 'text/*' should be triggered only when the mime type of the coming request is "text/<something>". 
- Add in the framework a reusable body parser that stores all body data in a string (similar  to the body parser from @ericogr that is given above)

For the mime type 'application/json' I would love to see a solution that uses a body for that type and makes the parser return a JSON object at the end of the request.

slaff, 2017-10-12:
@jcarterof @ericogr  Get the latest `develop` version and read here #1254 what can be done.
Feel free to reopen the issue, if needed.

closed 2017-10-12 by slaff