SmingHub/Sming #1241: Windows QuickStart fails

fredimartins, 2017-09-09:
> C:\WINDOWS\system32>choco install sming -y -source 'https://www.myget.org/f/sming/'
> Chocolatey v0.10.8
> Installing the following packages:
> sming
> By installing you accept licenses for the packages.
> sming not installed. An error occurred during installation:
>  Unable to resolve dependency 'jre8 '.
> sming package files install completed. Performing other installation steps.
> The install of sming was NOT successful.
> sming not installed. An error occurred during installation:
>  Unable to resolve dependency 'jre8 '.
> 
> Chocolatey installed 0/1 packages. 1 packages failed.
>  See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log).
> 
> Failures
>  - sming (exited 1) - sming not installed. An error occurred during installation:
>  Unable to resolve dependency 'jre8 '.

kwis2, 2017-09-16:
with choco install -y sming on fresh system I got

> Chocolatey v0.10.8
> Installing the following packages:
> sming
> By installing you accept licenses for the packages.
> sming not installed. An error occurred during installation:
>  Unable to resolve dependency 'spiffy '.
> sming package files install completed. Performing other installation steps.
> The install of sming was NOT successful.
> sming not installed. An error occurred during installation:
>  Unable to resolve dependency 'spiffy '.
> 
> Chocolatey installed 0/1 packages. 1 packages failed.
>  See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log).
> 
> Failures
>  - sming (exited 1) - sming not installed. An error occurred during installation
> :
>  Unable to resolve dependency 'spiffy '.
> 

and in log file 

> 2017-09-16 21:07:23,528 2976 [DEBUG] - _ Chocolatey:ChocolateyInstallCommand - Normal Run Mode _
> 2017-09-16 21:07:23,538 2976 [INFO ] - Installing the following packages:
> 2017-09-16 21:07:23,538 2976 [INFO ] - sming
> 2017-09-16 21:07:23,538 2976 [INFO ] - By installing you accept licenses for the packages.
> 2017-09-16 21:07:25,130 2976 [INFO ] - [NuGet] Attempting to resolve dependency 'eclipse-cpp '.
> 2017-09-16 21:07:25,300 2976 [INFO ] - [NuGet] Attempting to resolve dependency 'jre8 '.
> 2017-09-16 21:07:25,510 2976 [INFO ] - [NuGet] Attempting to resolve dependency 'esp8266-udk '.
> 2017-09-16 21:07:25,621 2976 [INFO ] - [NuGet] Attempting to resolve dependency 'mingw-get '.
> 2017-09-16 21:07:25,691 2976 [INFO ] - [NuGet] Attempting to resolve dependency '7zip.commandline'.
> 2017-09-16 21:07:25,821 2976 [INFO ] - [NuGet] Attempting to resolve dependency '7zip.portable'.
> 2017-09-16 21:07:26,452 2976 [INFO ] - [NuGet] Attempting to resolve dependency 'sming.core '.
> 2017-09-16 21:07:26,452 2976 [INFO ] - [NuGet] Attempting to resolve dependency 'spiffy '.
> 2017-09-16 21:07:26,562 2976 [ERROR] - sming not installed. An error occurred during installation:
>  Unable to resolve dependency 'spiffy '.
> 2017-09-16 21:07:26,582 2976 [INFO ] - sming package files install completed. Performing other installation steps.
> 2017-09-16 21:07:26,582 2976 [DEBUG] - No package information as package is null.
> 2017-09-16 21:07:26,602 2976 [DEBUG] - No package information to save as package is null.
> 2017-09-16 21:07:26,602 2976 [DEBUG] - Sending message 'HandlePackageResultCompletedMessage' out if there are subscribers...
> 2017-09-16 21:07:26,602 2976 [ERROR] - The install of sming was NOT successful.
> 2017-09-16 21:07:26,612 2976 [ERROR] - sming not installed. An error occurred during installation:
>  Unable to resolve dependency 'spiffy '.
> 2017-09-16 21:07:26,622 2976 [WARN ] - 
> Chocolatey installed 0/1 packages. 1 packages failed.
>  See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log).
> 2017-09-16 21:07:26,622 2976 [INFO ] - 
> 2017-09-16 21:07:26,622 2976 [ERROR] - Failures
> 2017-09-16 21:07:26,632 2976 [ERROR] -  - sming (exited 1) - sming not installed. An error occurred during installation:
>  Unable to resolve dependency 'spiffy '.
> 2017-09-16 21:07:26,632 2976 [DEBUG] - Sending message 'PostRunMessage' out if there are subscribers...
> 2017-09-16 21:07:26,722 2976 [DEBUG] - Exiting with 1
>

slaff, 2017-09-18:
Ok, may be I was not clear enough. But if you are following our [updated Wiki guide](https://github.com/SmingHub/Sming/wiki/Windows-Quickstart) you will see that FIRST you have to add sming to the list of repositories with

```
choco sources add -name smingrepo -source 'https://www.myget.org/F/sming/'
```

And THEN run

```
choco install -y sming
```

R3cycl0r, 2017-09-18:
I'm having the same issue about "spiffy" also adding first the list of repos and then running installer, what to do?

kwis2, 2017-09-18:
@slaff my procedure of windows install looks like below.
fresh VM with win 7 pro with jre8 and 7zip instaled

cmd as admin
`@powershell -NoProfile -ExecutionPolicy unrestricted -Command "iex ((new-object net.webclient).DownloadString('https://chocolatey.org/install.ps1'))" && SET PATH=%PATH%;%ALLUSERSPROFILE%\chocolatey\bin`

then 
`choco sources add -name smingrepo -source 'https://www.myget.org/F/sming/'`

finally in powershell as admin
`choco install -y sming`

it work fine untill mingw install - it hangs with window with `C:\ProgramData\cocolatey\bin\7zip.exe`

> MinGW
> mingw-get
> WARNING: Url has SSL/TLS available, switching to HTTPS for download
> Downloading mingw-get
>   from 'https://prdownloads.sourceforge.net/mingw/mingw-get-0.6.2-mingw32-beta-20131004-1-bin.tar.xz?download'
> Progress: 100% - Completed download of C:\Users\pomiar\AppData\Local\Temp\chocolatey\mingw-get\mingw-get-0.6.2-mingw32-b
> eta-20131004-1-bin.tar.xz (260.27 KB).
> Download of mingw-get-0.6.2-mingw32-beta-20131004-1-bin.tar.xz (260.27 KB) completed.
> C:\Users\pomiar\AppData\Local\Temp\chocolatey\mingw-get\mingw-get-0.6.2-mingw32-beta-20131004-1-bin.tar.xz
> Extracting 'C:\Users\pomiar\AppData\Local\Temp\chocolatey\mingw-get\mingw-get-0.6.2-mingw32-beta-20131004-1-bin.tar.xz'
> to 'C:\tools\MinGW'

For information below is full install log
cmd log:

> Microsoft Windows [Wersja 6.1.7601]
> Copyright (c) 2009 Microsoft Corporation. Wszelkie prawa zastrzeżone.
> 
> C:\Windows\system32>@powershell -NoProfile -ExecutionPolicy unrestricted -Comman
> d "iex ((new-object net.webclient).DownloadString('https://chocolatey.org/instal
> l.ps1'))" && SET PATH=%PATH%;%ALLUSERSPROFILE%\chocolatey\bin
> 
> Mode                LastWriteTime     Length Name
> ----                -------------     ------ ----
> d----        2017-09-18     20:14            chocInstall
> Unable to set PowerShell to use TLS 1.2 and TLS 1.1 due to old .NET Framework i
> nstalled. If you see underlying connection closed or trust errors, you may need
>  to do one or more of the following: (1) upgrade to .NET Framework 4.5+ and Pow
> erShell v3, (2) specify internal Chocolatey package location (set $env:chocolat
> eyDownloadUrl prior to install or host the package internally), (3) use the Dow
> nload + PowerShell method of install. See https://chocolatey.org/install for al
> l install options.
> Getting latest version of the Chocolatey package for download.
> Getting Chocolatey from https://chocolatey.org/api/v2/package/chocolatey/0.10.8
> .
> Downloading 7-Zip commandline tool prior to extraction.
> Extracting C:\Users\pomiar\AppData\Local\Temp\chocolatey\chocInstall\chocolatey
> .zip to C:\Users\pomiar\AppData\Local\Temp\chocolatey\chocInstall...
> Installing chocolatey on this machine
> Creating ChocolateyInstall as an environment variable (targeting 'Machine')
>   Setting ChocolateyInstall to 'C:\ProgramData\chocolatey'
> WARNING: It's very likely you will need to close and reopen your shell
>   before you can use choco.
> Restricting write permissions to Administrators
> We are setting up the Chocolatey package repository.
> The packages themselves go to 'C:\ProgramData\chocolatey\lib'
>   (i.e. C:\ProgramData\chocolatey\lib\yourPackageName).
> A shim file for the command line goes to 'C:\ProgramData\chocolatey\bin'
>   and points to an executable in 'C:\ProgramData\chocolatey\lib\yourPackageName
> '.
> 
> Creating Chocolatey folders if they do not already exist.
> 
> WARNING: You can safely ignore errors related to missing log files when
>   upgrading from a version of Chocolatey less than 0.9.9.
>   'Batch file could not be found' is also safe to ignore.
[... 166 lines cut ...]
>  The install of 7zip.portable was successful.
>   Software installed to 'C:\ProgramData\chocolatey\lib\7zip.portable\tools\7z-extra'
> 
> 7zip.commandline v16.02.0.20170209 [Approved]
> 7zip.commandline package files install completed. Performing other installation steps.
>  The install of 7zip.commandline was successful.
>   Software install location not explicitly set, could be in package or
>   default install location if installer.
> 
> mingw-get v1.0.2 - Possibly broken
> mingw-get package files install completed. Performing other installation steps.
> Get-BinRoot is going to be deprecated in v1 and removed in v2. It has been replaced with Get-ToolsLocation (starting wit
> h v0.9.10), however many packages no longer require a special separate directory since package folders no longer have ve
> rsions on them. Some do though and should continue to use Get-ToolsLocation.
> MinGW
> mingw-get
> WARNING: Url has SSL/TLS available, switching to HTTPS for download
> Downloading mingw-get
>   from 'https://prdownloads.sourceforge.net/mingw/mingw-get-0.6.2-mingw32-beta-20131004-1-bin.tar.xz?download'
> Progress: 100% - Completed download of C:\Users\pomiar\AppData\Local\Temp\chocolatey\mingw-get\mingw-get-0.6.2-mingw32-b
> eta-20131004-1-bin.tar.xz (260.27 KB).
> Download of mingw-get-0.6.2-mingw32-beta-20131004-1-bin.tar.xz (260.27 KB) completed.
> C:\Users\pomiar\AppData\Local\Temp\chocolatey\mingw-get\mingw-get-0.6.2-mingw32-beta-20131004-1-bin.tar.xz
> Extracting 'C:\Users\pomiar\AppData\Local\Temp\chocolatey\mingw-get\mingw-get-0.6.2-mingw32-beta-20131004-1-bin.tar.xz'
> to 'C:\tools\MinGW'

fredimartins, 2017-09-18:
exacly same here, it work fine untill mingw install - it hangs with window with C:\ProgramData\cocolatey\bin\7zip.exe

kwis2, 2017-09-21:
Witha another try to install sming 
VM, Win 7 pro 32 bit, procedure as above, but while install hangs on 7zip window I manually unpacked files

install hangs on `May not be able to find 'C:\tools\mingw64\bin\mingw-get.exe'` which file was in `c:\tools\MinGW `folder.

> Installing / updading required mingw packages
> WARNING: May not be able to find 'C:\tools\mingw64\bin\mingw-get.exe'. Please use full path for executables.
> ERROR: Exception calling "Start" with "0" argument(s): "Nie można odnaleźć określonego pliku"
> Environment Vars (like PATH) have changed. Close/reopen your shell to
>  see the changes (or in powershell/cmd.exe just type `refreshenv`).
> The install of sming was NOT successful.
> Error while running 'C:\ProgramData\chocolatey\lib\sming\tools\chocolateyInstall.ps1'.
>  See log for details.
> 
> Chocolatey installed 9/10 packages. 1 packages failed.
>  See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log).
> 
> Installed:
>  - 7zip.portable v16.04
>  - 7zip.commandline v16.02.0.20170209
>  - spiffy v1.0.4
>  - jre8 v8.0.144
>  - mingw-get v1.0.2
>  - sming.core v3.3.0
>  - eclipse-cpp v4.4.2.20150413
>  - esp8266-udk v2.1.0
>  - mingw v5.3.0
> 
> Failures
>  - sming (exited -1) - Error while running 'C:\ProgramData\chocolatey\lib\sming\tools\chocolateyInstall.ps1'.
>  See log for details.
> PS C:\Windows\system32>

So I decided to copy MinGW into mingw64, and installation was succesfull with a lot of errors like 

> mingw-get.exe: *** ERROR *** C:\tools\mingw64\/msys/1.0/share/man/man1/xargs.1: extraction failed
> mingw-get.exe: *** ERROR *** C:\tools\mingw64\/msys/1.0/share/man/man1/xargs.1: probable package conflict; existing file
>  not overwritten

and finally I decide to run choco install --force -y sming and got

>  The install of sming was successful.
>   Software installed to 'C:\tools\sming'
> 
> Chocolatey installed 1/1 packages.
>  See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log).

but `make` don't work at all

cocolatey log https://github.com/kwis2/Windows-QuickStart-fails/blob/master/chocolatey.txt

anthonyvallee, 2017-11-02:
I am having the same issue here, `choco install -y sming` hangs with 7zip not extracting the files.

bryanlikes, 2017-11-17:
I also have the exact same symptoms as kwis2 post on sept 21.

kwis2, 2017-11-20:
choco still not work on 32 bit windows, and it try to install old, 3.3.0 version of sming

kwis2, 2017-11-25:
Now there is no problem with 7zip, but now there is another problem.

> Chocolatey installed 6/8 packages. 2 packages failed.
>  See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log).
> 
> Installed:
>  - 7zip.portable v16.04
>  - 7zip.commandline v16.02.0.20170209
>  - spiffy v1.0.4
>  - sming.core v3.3.0
>  - esp8266-udk v2.1.0
>  - mingw v5.3.0
> 
> Failures
>  - mingw-get (exited 404) - Error while running 'C:\ProgramData\chocolatey\lib\mingw-get\tools\chocolateyInstall.ps1'.
>  See log for details.
> 
>  - sming (exited -1) - Error while running 'C:\ProgramData\chocolatey\lib\sming\tools\chocolateyInstall.ps1'.
>  See log for details.

below is full log [chocolatey.txt](https://github.com/kwis2/Windows-QuickStart-fails/blob/master/chocolatey.txt)

dbelz, 2017-12-08:
What was the solution for this problem? Have the same behavior with hanging on 7zip extraction.

System is Windows 10 64Bit