SmingHub/Sming #1242: Issue when publishing a message via MQTT

jns2011, 2017-09-10:
According to the on screen debugging information,it seems that the underlying TcpConnection is leaking memory on every publsh invocation. 
I get messsages reporting the data stored on stream that is increasing on every call. Frequent memory reallocations, and very often shows up of the "WAITING FOR FREE SPACE" message.
When the firmware keeps running for some time .. the WTD ends up doing its job. 
Not sure if due to this, but , it is my first suspicion.
Have someone noticed this issue.?

zgoda, 2017-09-13:
Yes but it does not "leak". Memory is allocated on heap and reclaimed after some time (when connection closes for good). Dynamic memory allocation creates "holes" in otherwise continuous space and after some time WDT has to kick in with restart when there's not enough continuous space on the heap to allocate requested amount.

See https://stackoverflow.com/a/80113/12138

jns2011, 2017-09-13:
Thanks for you answer.
To be honest I didn't watched in deep inside the code.. just in order to
notice the source of the "wait .." message.
Of course I understand the dynamic memory allocation process, but what
caught my eye was the fact that
sending packets of fixed size (150 bytes each) there are several allocation
steps that doubles buffer size.

And.. of course after some time (because I send packeds every 5 seconds)  I
notice that the WDT fires.. this is what is strange.
Anyways I'll keep paying attention ant trust your knowledge about the
source code.

Thanks again for your support.

zgoda, 2017-09-15:
Simple MQTT communication should not induce such behaviour, not that often anyway. ESP has plenty of RAM so if you do not waste it elsewhere (eg. in HTTP server) MQTT should not be a problem.

closed 2018-11-19 by slaff