SmingHub/Sming #1244: FileStream uplaod from SPIFF to server : Exception20

bbourdel, 2017-09-15:
Hello, 
On version 3.3.0, with an ESP-12 / 32Mbit, when you try to do a http request with a FileStream as data, it raise an Exception 20.
See my code above :
`FileStream fileS = FileStream("data.raw");
	uploadClient.send(
		uploadClient.request("http://posttestserver.com/")
		->setHeaders(requestHeaders)
		->setBody(&fileS)
		->onRequestComplete(onDownload)
	);`

It raise an Fatal Exception 20 like this :
`***** Fatal exception 20
pc=0x00000000 sp=0x3ffffba0 excvaddr=0x00000000
ps=0x00000030 sar=0x00000008 vpri=0x4023496c
r00: 0x40235f8e=1076060046 r01: 0x3ffffba0=1073740704 r02: 0x3ffffd88=1073741192 
r03: 0x00000000=         0 r04: 0x3ffffd88=1073741192 r05: 0x0000007e=       126 
r06: 0x6000001c=1610612764 r07: 0xff000000= -16777216 r08: 0x002e6d61=   3042657 
r09: 0x00000120=       288 r10: 0x3ffffab2=1073740466 r11: 0x00000001=         1 
r12: 0x3fff1a48=1073683016 r13: 0x3fff1818=1073682456 r14: 0x3fff0770=1073678192 
r15: 0x00000000=         0 `

After few research, the error take place in HttpConnection.cpp, line 405 when it try to call the function length() of the FileStream (IDataSourceStream).
https://github.com/SmingHub/Sming/blob/d408e7de1382c22c2ac0289a1a074821eb504186/Sming/SmingCore/Network/Http/HttpConnection.cpp#L405

I've check the pointer value, it's the same as when "fileS" is created.
Any idea on this ?

bbourdel, 2017-09-16:
OK, i've found a way, adding this in HttpRequest.cpp :
```
	HttpRequest* HttpRequest::setBodyFile(const String& body) {
		if(stream != NULL) {
			debugf("HttpRequest::setBody: Discarding already set stream!");
			delete stream;
			stream = NULL;
		}

		FileStream *memory = new FileStream();
		memory->attach(body, eFO_ReadOnly);
		stream = (IDataSourceStream*)memory;
		return this;
	}
```
And this in the "main" loop :

```	uploadClient.send(
		uploadClient.request("http://posttestserver.com/post.php?dir=aver")
		->setHeaders(requestHeaders)
		->setBodyFile("data.raw")
		->onRequestComplete(onDownload)
	);
```

Now, the FileStream object is created in the HttpRequest class and it's still accessible in HttpClient, Http Connection, TCPClient and TCP connection classes.

closed 2017-09-18 by bbourdel

slaff, 2017-09-22:
> FileStream fileS = FileStream("data.raw");

That will not work because you are passing a reference to a local variable. What can work is this

`FileStream* fileS = new FileStream("data.txt");`

thus leading to the following modified example

```
HttpHeaders headers;
headers["Content-Type"]  = "application/octet-stream";

FileStream* fileS = new FileStream("data.txt");

client->send(
        client->request("http://domain.com/data.txt")
        ->setMethod(HTTP_PUT)
        ->setHeaders(headers)
        ->setBody(fileS)
        ->onRequestComplete(onDownload)
);
```