SmingHub/Sming #1250: Example with HttpRequestAuth

icodk, 2017-10-04:
While upgrading to Sming 3.3.0 my old basic authentication is not supported (ex. using response.authorizationRequired() )
Where can I find en example of how to use it in sming 3.3.0 ?

slaff, 2017-10-11:
>  my old basic authentication is not supported

@icodk Are you using basic authentication with HttpClient (to view protected pages) or with HttpServer ( to protect pages ) ?

icodk, 2017-10-11:
Yes exactly. As I described in #585

slaff, 2017-10-11:
> Yes exactly

Which one - Server or Client?

icodk, 2017-10-11:
No,I want to protect a server page.

slaff, 2017-10-11:
Once  version 3.4.0 is released I can think about refactoring the auth adapters to work for both http clients and servers.

jochenjagers, 2017-10-21:
@icodk: i had the same problem with response.badRequest()
I solved it by setting response.code = HTTP_STATUS_BAD_REQUEST as described for forbidden and notFound in HttpResponse.h
You sould be fine by replacing "response.authorizationRequired()" by response.code = HTTP_STATUS_UNAUTHORIZED

icodk, 2017-10-22:
@jochenjagers  Thanks
Will try it out