SmingHub/Sming #1256: UART0 used  for communication with another controller

aparaske, 2017-10-12:
Hi,

We're using UART0/Serial for communication with another controller. How can we avoid any logging (from Sming, rboot, etc) from being written to UART0, or redirect it to UART1?

Regards,
Alex

slaff, 2017-10-12:
@aparaske Just check the code in the `init` function from the Basic_Serial example: https://github.com/SmingHub/Sming/blob/develop/samples/Basic_Serial/app/application.cpp

aparaske, 2017-10-12:
@slaff I am aware of the '.systemDebugOutput(false);' function. I just want to confirm that there's nothing else that could affect UART0, as I am getting some bizarre behaviour from the external controller, especially during WiFi disconnects or ESP restarts etc.

Informatic, 2017-10-12:
ESP restarts always send "random" data (mask rom bootloader logs at ~70kbps baudrate) on UART0, but WiFi disconnects shouldn't. Keep in mind you can use `Serial.swap()` to swap UART0 pins over to GPIO13 & GPIO15. (source: http://bbs.espressif.com/viewtopic.php?f=7&t=22&p=54&hilit=system_uart_swap#p54)

aparaske, 2017-10-12:
Wouldn't swapping UART0 pins over to GPIO13 & GPIO15 render HardwareSerial on GPIO1 & GPIO3 useless? When defining a HardwareSerial object, we need to define the port, which is either UART0 or UART1 (used for logging using GPIO2). But if UART0 is mapped to GPIO13 & GPIO15 due to swapping and the external controller is hooked up to GPIO1 & GPIO3, how could the ESP communicate with the external controller?

Thanks,
Alex

aparaske, 2017-10-18:
I noticed that rBoot and crash_handler use ets_printf. Could that be the reason for our issues? Is there a way to disable ets_printf or direct it only to UART1?

Thanks,
Alex

slaff, 2018-11-19:
@aparaske I will close the ticket due to inactivity. If you still need help feel free to reopen it.

Just to summarize:
- There are initial messages coming from the ROM bootloader that cannot be silenced. BUT you can instruct the external microcontroller to wait for X milliseconds before starting to read data from UART0.
- There are messages coming from rBoot that can be silenced with RBOOT_SILENT make directive.
- All further debug messages from the SDK or your program can be redirected to UART1 so that UART0 is free to communicate with the external microcontroller.

We use Sming Framework in our [U:Kit project](https://github.com/attachix/ukit) and in there our ESP microcontroller talks with an Attiny microcontroller.

closed 2018-11-19 by slaff