SmingHub/Sming #1263: ESP8266 security vulnerability! Krack attack.

anakod, 2017-10-17:
New vulnerability was found in the WPA2 wireless protocol called Krack that could allow attackers to eavesdrop on wireless connections and inject data into the wireless stream.

Espressif already published patch:
https://github.com/espressif/ESP8266_NONOS_SDK/commit/b762ea222ee94b9ffc5e040f4bf78dd8ba4db596

More information\related links:

- https://www.bleepingcomputer.com/news/security/list-of-firmware-and-driver-updates-for-krack-wpa2-vulnerability/
- https://github.com/arendst/Sonoff-Tasmota/issues/1024
- https://github.com/esp8266/Arduino/issues/3725

anakod, 2017-10-17:
@slaff Hello. I think we should publish hotfix for this issue.

slaff, 2017-10-17:
I have something half-baked ( https://github.com/slaff/Sming/tree/update/sdk-2.1 ) BUT it means that we should switch to the latest SDK 2.1.0++ and stop supporting the other SDKs. The patch is based on that PR #1153.

@anakod If you want this fixed soon I will need your help to test it and make it rock-solid.

anakod, 2017-10-17:
May be you will prepare P-R and we will start testing?

slaff, 2017-10-17:
@anakod Check this PR #1264

SvenLuebke, 2017-10-27:
Did anyone get the latest SDK from https://github.com/espressif/ESP8266_NONOS_SDK completely running? I've fixed all compiler errors and warnings, but afterwards the linker complains about:

xtensa-lx106-elf/bin/ld: out/build/app.out.tmp section `.text' will not fit in region `iram1_0_seg'
xtensa-lx106-elf/bin/ld: region `iram1_0_seg' overflowed by 56808 bytes

for Basic_WiFi. Save/move nearly 57kB seems to be impossible.

slaff, 2017-11-23:
With commit c2cc852e40b1971a1987aceeedab5b28b8c7297c we have experimental support for SDK 2.1.0. Please, test and report any issues that you have noticed.

closed 2017-11-23 by slaff