SmingHub/Sming #1272: [feature request] Stop a HTTPServer

bbourdel, 2017-10-26:
There is any fail-safe shutdown sequence to HttpServer class.

I'd tried something like this : 
```
HttpServer* server = new HttpServer();
server->listen(80);
server->addPath("/", onServer);

//Bla bla bla

delete server;
``` 
It's work but only if `delete server;` is called without any TCPConnection opened. If not, a fatal exception will be raised when the remaining TCPConnection will de closed.

bbourdel, 2017-10-26:
Here my modifications for HTTPServer, it's nearly good.
```
diff --git a/Sming/SmingCore/Network/HttpServer.cpp b/Sming/SmingCore/Network/HttpServer.cpp
--- a/Sming/SmingCore/Network/HttpServer.cpp
+++ b/Sming/SmingCore/Network/HttpServer.cpp
@@ -17,12 +17,14 @@

 HttpServer::HttpServer()
 {
+    active = true;
     settings.keepAliveSeconds = 0;
     configure(settings);
 }

 HttpServer::HttpServer(HttpServerSettings settings)
 {
+    active = true;
     configure(settings);
 }

@@ -44,6 +46,7 @@ void HttpServer::configure(HttpServerSettings settings) {

 HttpServer::~HttpServer()
 {
+    active = true;
     for(int i=0; i< resourceTree.count(); i++) {
         if(resourceTree.valueAt(i) != NULL) {
             delete resourceTree.valueAt(i);
@@ -99,7 +102,23 @@ void HttpServer::setDefaultResource(HttpResource* resource) {
     addPath("*", resource);
 }

+void HttpServer::close() {
+    active = false;
+    if(totalConnections==0){
+        debugf("Closing server");
+        delete this;
+    }
+    else{
+        debugf("Wait end connection before closing server");
+    }
+}
+
+
 void HttpServer::onConnectionClose(TcpClient& connection, bool success) {
     totalConnections--;
+    if(totalConnections==0 && !active){
+        debugf("Closing server");
+        delete this;
+    }
     debugf("Closing connection. Total connections: %d", totalConnections);
 }

diff --git a/Sming/SmingCore/Network/HttpServer.h b/Sming/SmingCore/Network/HttpServer.h
--- a/Sming/SmingCore/Network/HttpServer.h
+++ b/Sming/SmingCore/Network/HttpServer.h
@@ -77,6 +77,8 @@ public:
     void setDefaultHandler(const HttpPathDelegate& callback);
     void setDefaultResource(HttpResource* resource);

+    void close();
+

 protected:
     virtual TcpConnection* createClient(tcp_pcb *clientTcp);
@@ -91,6 +93,7 @@ private:
     HttpServerSettings settings;
     ResourceTree resourceTree;
     BodyParsers bodyParsers;
+    bool active = true;
 };

 /** @} */
```

slaff, 2017-11-11:
Implemented in PR #1284. @bbourdel Thanks for the help and initial code.

closed 2017-11-11 by slaff