SmingHub/Sming #1274: Software watchdog bites due to PR#1097 - m_printf: stacksize reduced

frankdownunder, 2017-11-01:
If I use the old version of Sming/system/m_printf.cpp my code works fine.  With the new version introduced on Sep 30, the dog bites me.
Two conclusions are possible:
1.  My code is making unreasonable demands
2.  The new version of m_printf.cpp is slower then the old version, possibly even a forever loop.

I suspect the second reason, and will do some timing tests and so on to investigate further.

slaff, 2017-11-08:
There is definitely something fishy with the latest changes to m_printf.cpp ed94fb77d7e8a948e7fc6837844b48d1a94f933c. Reverting those changes made the Basic_Serial example work again with SDK 2.1+.

@frankdownunder Did you have time to analyze the changes?

frankdownunder, 2017-11-08:
@slaff Hi,  I found nothing conclusive.  I was hoping for a minimal "this demonstrates the bug" but couldn't. The new function didn't seem to be significantly slower either.  I am fighting other fires now,  sorry..

slaff, 2017-11-09:
@frankdownunder Feel free to reopen the ticket when you have more information about it.

closed 2017-11-09 by slaff