SmingHub/Sming #1275: Samples rBoot Fatal on 4M flash

lrodorigo, 2017-11-02:
Hi. I'm getting the Fatal exception 28 when trying the rBoot sample on an Ai-Thinker ESP-12-F (no SSL download, I don't know if can be related to https://github.com/SmingHub/Sming/issues/1167).
I noticed that the firmware seems to be flashed, but the rom switch is not working.
The next reboot always happens from rom 0.
I didn't copy any linked file  into the sample folder, because I am using  the example as-is on a 4M flash.

I also get two distinct fatal exceptions, here's the debug output and the two stack traces:

**Debug Output:**
```
staticOnMessageComplete: Execution queue: 0, http://xxxxxx/ver0.bin

Firmware download finished!
 - item: 0, addr: 202000, len: 287360 bytes
In callback...
Firmware updated, rebooting to rom 1...
state: 5 -> 0 (0)
rm 0
TCP connection error: -8
del if0
usl
Firmware updated.

***** Fatal exception 28
pc=0x401063f7 sp=0x3ffff9f0 excvaddr=0x0000007c
ps=0x00000030 sar=0x00000010 vpri=0x40100de4
r00: 0x401063f2=1074815986 r01: 0x3ffff9f0=1073740272 r02: 0x00000000=         0
r03: 0x00000000=         0 r04: 0x60000000=1610612736 r05: 0x0000007e=       126
r06: 0x6000001c=1610612764 r07: 0x00000004=         4 r08: 0x3fff05b0=1073677744
r09: 0x00000130=       304 r10: 0x00000000=         0 r11: 0x3ffef790=1073674128
r12: 0x3fff1110=1073680656 r13: 0x00000000=         0 r14: 0x3fff11b0=1073680816
r15: 0x3ffffaf0=1073740528

```
**first stack trace:**
```
0x401063f7: HttpConnection::staticOnMessageComplete(http_parser*) at /opt/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:125
0x40100de4: pvPortRealloc at ??:?
0x401063f2: Delegate<int (HttpConnection&, bool)>::operator()(HttpConnection&, bool) const at /opt/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:125
 (inlined by) HttpConnection::staticOnMessageComplete(http_parser*) at /opt/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:209
0x40105b50: String::changeBuffer(unsigned int) at /opt/Sming/Sming/Wiring/WString.cpp:729
0x4023019a: String::~String() at /opt/Sming/Sming/Wiring/WString.cpp:729
0x402301e2: String::reserve(unsigned int) at /opt/Sming/Sming/Wiring/WString.cpp:729
0x40206e94: free at C:\DEV\ESP8266\esp_microc/malloc/malloc.c:15
0x40105c40: String::operator=(String const&) at /opt/Sming/Sming/Wiring/WString.cpp:729
0x40230160: _GLOBAL__sub_I__ZN9IPAddressC2Ev at /opt/Sming/Sming/Wiring/IPAddress.cpp:122
0x402380af: URL::toString() at /opt/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:125
 (inlined by) URL::toString() at /opt/Sming/Sming/SmingCore/Network/Http/../URL.h:30
0x4022ffe0: m_vprintf at /opt/Sming/Sming/system/m_printf.cpp:86
0x40206e94: free at C:\DEV\ESP8266\esp_microc/malloc/malloc.c:15
0x4023f719: Print::write(char const*, unsigned int) at /opt/Sming/Sming/Wiring/Print.h:54
 (inlined by) Print::print(String const&) at /opt/Sming/Sming/Wiring/Print.cpp:115
0x401063f2: Delegate<int (HttpConnection&, bool)>::operator()(HttpConnection&, bool) const at /opt/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:125
 (inlined by) HttpConnection::staticOnMessageComplete(http_parser*) at /opt/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:209
0x4010000a: call_user_start at ??:?
0x40104d97: wDev_ProcessFiq at ??:?
0x4000050c: ?? ??:0
0x4023f36b: TcpClient::onConnected(signed char) at /opt/Sming/Sming/SmingCore/Network/TcpClient.cpp:98
0x40206ea8: realloc at C:\DEV\ESP8266\esp_microc/malloc/malloc.c:20
0x40105b50: String::changeBuffer(unsigned int) at /opt/Sming/Sming/Wiring/WString.cpp:729
0x40238ab4: HttpConnection::onConnected(signed char) at /opt/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:378
0x4023eb98: TcpConnection::staticOnReceive(void*, tcp_pcb*, pbuf*, signed char) at /opt/Sming/Sming/SmingCore/Network/TcpConnection.cpp:503
0x40238a90: HttpConnection::onConnected(signed char) at /opt/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:372
0x40103a46: pp_post at ??:?
0x401036f5: ppEnqueueRxq at ??:?
0x401035aa: ppEnqueueRxq at ??:?
0x401062d0: HttpConnection::staticOnBody(http_parser*, char const*, unsigned int) at /opt/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:125
0x40235936: http_parser_execute at /opt/Sming/Sming/third-party/http-parser/http_parser.c:2169
0x4000050c: ?? ??:0
0x401023ea: lmacProcessTxSuccess at ??:?
0x40238ad0: FIFO<HttpRequest*, 20>::peek() const at /opt/Sming/Sming/SmingCore/Network/../../Wiring/FIFO.h:95
[... 96 lines cut ...]
0x40102219: lmacProcessTXStartData at ??:?
0x4023eb98: TcpConnection::staticOnReceive(void*, tcp_pcb*, pbuf*, signed char) at /opt/Sming/Sming/SmingCore/Network/TcpConnection.cpp:503
0x4000050c: ?? ??:0
0x40104a94: wDev_ProcessFiq at ??:?
0x40223f0c: ip_output_if_opt at ??:?
0x4022684b: tcp_input at ??:?
0x40226695: tcp_input at ??:?
0x402207a0: pbuf_alloc at ??:?
0x40223f80: ip_output at ??:?
0x4000050c: ?? ??:0
0x40221e29: tcp_send_empty_ack at ??:?
0x40202b4a: ets_timer_handler_isr at ??:?
0x40221eda: tcp_output at ??:?
0x402268a4: tcp_input at ??:?
0x402207d1: pbuf_alloc at ??:?
0x40223c89: ip_input at ??:?
0x4022464d: ethernet_input at ??:?
0x401031fe: lmacTxFrame at ??:?
0x40103619: ppEnqueueRxq at ??:?
0x401033af: ppProcessTxQ at ??:?
0x401033ea: ppProcessTxQ at ??:?
0x4020687b: ets_snprintf at ??:?
0x40000f49: ?? ??:0
0x40000f49: ?? ??:0
```

bbourdel, 2017-11-03:
Did you try to reset/restart the chip after the inital flashing before updating the chip ?

lrodorigo, 2017-11-03:
You mean to flash (by uart) -> power cycle -> OTA?

Anyway, yesterday evening I noticed a strange behaviour.
The booted ROM seems to change (randomly?) at each power cycle.

lrodorigo, 2017-11-06:
I can confirm that the rom in use switches at each power cycle, even if no rom switch is made

MayaPosch, 2017-12-21:
Getting the same issue with rBoot OTA updates on the develop branch:

`0x40247e26: HttpConnection::staticOnMessageComplete(http_parser*) at /opt/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:129
0x40100de4: pvPortRealloc at ??:?
0x40247e21: Delegate<int (HttpConnection&, bool)>::operator()(HttpConnection&, bool) const at /opt/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:129
 (inlined by) HttpConnection::staticOnMessageComplete(http_parser*) at /opt/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:213
0x401059f8: String::changeBuffer(unsigned int) at /opt/Sming/Sming/Wiring/WString.cpp:729
0x4023516e: String::reserve(unsigned int) at /opt/Sming/Sming/Wiring/WString.cpp:729
0x402351b6: String::copy(char const*, unsigned int) at /opt/Sming/Sming/Wiring/WString.cpp:729
0x40207204: free at C:\DEV\ESP8266\esp_microc/malloc/malloc.c:15
0x40105ae8: String::operator=(String const&) at /opt/Sming/Sming/Wiring/WString.cpp:729
0x40235134: String::~String() at /opt/Sming/Sming/Wiring/WString.cpp:729
0x40247963: URL::toString() at /opt/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:129
0x40234d44: m_printf at /opt/Sming/Sming/system/m_printf.cpp:110
0x40207204: free at C:\DEV\ESP8266\esp_microc/malloc/malloc.c:15
0x40249725: MethodCaller<int (rBootHttpUpdate::*)(HttpConnection&, bool)>::invoke(HttpConnection&, bool) at /opt/Sming/Sming/SmingCore/Network/../Delegate.h:80 (discriminator 3)
0x40247e21: Delegate<int (HttpConnection&, bool)>::operator()(HttpConnection&, bool) const at /opt/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:129
 (inlined by) HttpConnection::staticOnMessageComplete(http_parser*) at /opt/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:213
0x4024869c: HttpConnection::onReceive(pbuf*) at /opt/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:569
0x40234cd8: m_vprintf at /opt/Sming/Sming/system/m_printf.cpp:75
0x40239bfb: TcpClient::onReceive(pbuf*) at /opt/Sming/Sming/SmingCore/Network/TcpClient.cpp:138
0x4024865c: HttpConnection::onReceive(pbuf*) at /opt/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:559
0x4010473d: trc_NeedRTS at ??:?
0x401031fe: lmacTxFrame at ??:?
0x401048cc: trc_NeedRTS at ??:?
0x4024780c: HttpConnection::staticOnBody(http_parser*, char const*, unsigned int) at /opt/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:129
0x40104cfa: wDev_ProcessFiq at ??:?
0x402442be: http_parser_execute at /opt/Sming/Sming/third-party/http-parser/http_parser.c:2169
0x4024869c: HttpConnection::onReceive(pbuf*) at /opt/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:569
0x4022ff3d: espconn_init at ??:?
0x4022ff3d: espconn_init at ??:?
0x40239c88: TcpClient::onReadyToSendData(TcpConnectionEvent) at /opt/Sming/Sming/SmingCore/Network/TcpClient.cpp:148
0x40248639: HttpConnection::onReadyToSendData(TcpConnectionEvent) at /opt/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:437
0x40213bd3: pp_attach at ??:?
0x40213c22: pp_attach at ??:?
0x40213d2e: pp_attach at ??:?
0x40103a46: pp_post at ??:?
0x40213093: ppTxPkt at ??:?
0x40218fa4: ieee80211_output_pbuf at ??:?
0x40224a18: etharp_send_ip at etharp.c:?
0x40220f89: pbuf_free at ??:?
0x40224c45: etharp_output_to_arp_index at etharp.c:?
0x40247188: TcpConnection::staticOnReceive(void*, tcp_pcb*, pbuf*, signed char) at /opt/Sming/Sming/SmingCore/Network/TcpConnection.cpp:707
0x40224eac: etharp_output at ??:?
0x4022eeb8: inet_chksum_pseudo at ??:?
0x40224808: ip_output_if_opt at ??:?
0x4021de7c: cnx_update_bss_more at ??:?
0x4022483c: ip_output_if at ??:?
0x402210cc: pbuf_alloc at ??:?
0x4022487c: ip_output at ??:?
0x40222735: tcp_send_empty_ack at ??:?
0x402227e6: tcp_output at ??:?
0x402271a0: tcp_input at ??:?
0x40224585: ip_input at ??:?
0x40221c9c: tcp_tmr at ??:?
0x40224f49: ethernet_input at ??:?
0x401008fa: ets_timer_setfn at ??:?
0x4010083e: wdt_feed at ??:?
0x40202dc8: ets_timer_handler_isr at ??:?
0x40202dee: ets_timer_handler_isr at ??:?
0x40206acf: ets_snprintf at ??:?
0x40000f49: ?? ??:0`

The next boot the correct ROM appears to be selected, however.

lrodorigo, 2017-12-21:
Hi! Thanks for reporting.
Can you try to power cycle the device? In my case I see a different the
boot partition at each boot

vijaygill, 2018-07-10:
I got this error too when I tried a small sample of my own (adapted from sample in Sming) but a "make flashinit" before "make flash" fixed it.

kmihaylov, 2018-08-15:
1st I warn that I do not use the example directly. Despite that I must say I only copy-pasted the code into a new OTA class. Probably it is about the same thing.

My ESP restarts (Software Watchdog Reset) after trying to perform OTA update.

Attached is the dump:

> 0x40249d08: http_parser_execute at /opt/Sming/Sming/third-party/http-parser/http_parser.c:2107
> 0x4024f345: HttpConnection::onReceive(pbuf*) at /opt/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:558
> 0x40103d8e: ppEnqueueRxq at ??:?
> 0x402084ac: free at C:\DEV\ESP8266\esp_microc/malloc/malloc.c:15
> 0x402083b6: rboot_write_end at /opt/Sming/Sming/third-party/rboot/appcode/rboot-api.c:114
> 0x402084ac: free at C:\DEV\ESP8266\esp_microc/malloc/malloc.c:15
> 0x4023f820: rBootItemOutputStream::close() at /opt/Sming/Sming/Wiring/Print.h:35
> 0x402084ac: free at C:\DEV\ESP8266\esp_microc/malloc/malloc.c:15
> 0x4023b442: String::copy(char const*, unsigned int) at /opt/Sming/Sming/Wiring/WString.cpp:729
> 0x401061ed: String::operator=(char const*) at /opt/Sming/Sming/Wiring/WString.cpp:729
> 0x4024f450: HttpResponse::reset() at /opt/Sming/Sming/SmingCore/Network/Http/HttpResponse.cpp:116 (discriminator 1)
> 0x4024e0f1: HttpConnection::reset() at /opt/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:558
> 0x4024e226: HttpConnection::cleanup() at /opt/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:558
> 0x4024f37c: HttpConnection::onReceive(pbuf*) at /opt/Sming/Sming/SmingCore/Network/Http/HttpConnection.cpp:558
> 0x40228937: tcp_receive at tcp_in.c:?
> 0x40229431: tcp_input at ??:?
> 0x40218593: ieee80211_ht_updateparams at ??:?
> 0x4021f1f3: cnx_update_bss_more at ??:?
> 0x4021cd69: scan_parse_beacon at ??:?
> 0x4021cdc9: scan_parse_beacon at ??:?
> 0x4021ccb1: scan_parse_beacon at ??:?
> 0x40213e88: ppRecycleRxPkt at ??:?
> 0x40218fb5: ieee80211_parse_beacon at ??:?
> 0x40100b50: ets_timer_arm_new at ??:?
> 0x4023f43b: TcpConnection::staticOnReceive(void*, tcp_pcb*, pbuf*, signed char) at /opt/Sming/Sming/SmingCore/Network/TcpConnection.cpp:584
> 0x40000f58: ?? ??:0
> 0x40229640: tcp_input at ??:?
> 0x40222571: pbuf_alloc at ??:?
> 0x402259f9: ip_input at ??:?
> 0x402263bd: ethernet_input at ??:?
> 0x40100976: ets_timer_setfn at ??:?
> 0x401008ba: wdt_feed at ??:?
> 0x40203994: ets_timer_handler_isr at ??:?
> 0x402039ba: ets_timer_handler_isr at ??:?
> 0x40207e83: ets_snprintf at ??:?
> 0x40000f49: ?? ??:0
> 0x40000f49: ?? ??:0

What I was able to find is that the situation happens immediately after `otaUpdater->start();` has been executed.

Even if I put `return;` at the first row of rBootHttpUpdate::start(), it crashes.
If I comment `otaUpdater->start();` it doesn't perform the update, but it doesn't crash.

slaff, 2018-08-15:
@kmihaylov Would it be possible to paste the code that you use, or at least to provide a sample code where the issue is reproducible. 

> Even if I put return; at the first row of rBootHttpUpdate::start(), it crashes.
> If I comment otaUpdater->start(); it doesn't perform the update, but it doesn't crash.

Hm.. My blind guess is that otaUpdate is either not initialized or points to a memory address that was freed earlies and thus is invalid.

kmihaylov, 2018-08-15:
> #include "ota.h"
> #include "application.h"
> #include <SmingCore/SmingCore.h>
> 
> rBootHttpUpdate* otaUpdater = 0;
> 
> OTA::OTA() { debugf("OTA init"); }
> OTA::~OTA() {}
> 
> void OTA::initCommand() {
> 	commandHandler.registerCommand(
> 		CommandDelegate(
> 			"updateFW",
> 			"update the firmware",
> 			"OTA",
> 			commandFunctionDelegate(&OTA::processCmd,this)
> 		)
> 	);
> }
> 
> void OTA::processCmd(String commandLine, CommandOutput* commandOutput) {
> 	Vector<String> commandToken;
> 	int numToken = splitString(commandLine, ' ' , commandToken);
> 
> 	if (numToken == 1) {
> 		if( commandToken[0] == "updateFW" ) {
> 			this->update();
> 		}
> 	} 
> 	/*
> 	else {
> 		if (commandToken[1] == "on") {
> 			status = true;
> 			commandOutput->printf("Status ON\r\n");
> 		} else if (commandToken[1] == "off") {
> 			status = false;
> 			commandOutput->printf("Status OFF\r\n");
> 		} else if (commandToken[1] == "status") {
> 			String tempString = status ? "ON" : "OFF";
> 			commandOutput->printf("Example Status is %s\r\n",tempString.c_str());
> 		};
> 	}
> 	*/
> }
> 
> void OTA::updateCallBack(rBootHttpUpdate& client, bool result) {
> 	
> 	Serial.println("In callback...");
> 	if(result == true) {
> 		// success
> 		uint8 slot;
> 		slot = rboot_get_current_rom();
> 		if (slot == 0) slot = 1; else slot = 0;
> 		// set to boot new rom and then reboot
> 		Serial.printf("Firmware updated, rebooting to rom %d...\r\n", slot);
> 		rboot_set_current_rom(slot);
> 		System.restart();
> 	} else {
> 		// fail
> 		Serial.println("Firmware update failed!");
> 	}
> }
> 
> void OTA::update() {
> 	
> 	uint8 slot;
> 	rboot_config bootconf;
> 	
> 	Serial.println("Updating...");
> 	
> 	// need a clean object, otherwise if run before and failed will not run again
> 	if (otaUpdater) delete otaUpdater;
> 	otaUpdater = new rBootHttpUpdate();
> 
> 	// select rom slot to flash
> 	bootconf = rboot_get_config();
> 	slot = bootconf.current_rom;
> 	if (slot == 0) slot = 1; else slot = 0;
> 
> #ifndef RBOOT_TWO_ROMS
> 	// flash rom to position indicated in the rBoot config rom table
> 	otaUpdater->addItem(bootconf.roms[slot], ROM_0_URL);
> #else
> 	// flash appropriate rom
> 	if (slot == 0) {
> 		otaUpdater->addItem(bootconf.roms[slot], ROM_0_URL);
> 	} else {
> 		otaUpdater->addItem(bootconf.roms[slot], ROM_1_URL);
> 	}
> #endif
> 	
> #ifndef DISABLE_SPIFFS
> 	// use user supplied values (defaults for 4mb flash in makefile)
> 	if (slot == 0) {
> 		otaUpdater->addItem(RBOOT_SPIFFS_0, SPIFFS_URL);
> 	} else {
> 		otaUpdater->addItem(RBOOT_SPIFFS_1, SPIFFS_URL);
> 	}
> #endif
> 
> 	// request switch and reboot on success
> 	otaUpdater->switchToRom(slot);
> 	// and/or set a callback (called on failure or success without switching requested)
> 	//otaUpdater->setCallback(OTA::updateCallBack);
> 	Serial.printf("rBoot_config, romSlot: %d , addr1: %#010x , addr2: %#010x\r\n" , bootconf.current_rom , bootconf.roms[0] , bootconf.roms[1]);
> 	system_soft_wdt_stop();
> 	// start update
> 	otaUpdater->start();
> }
> 
> void OTA::Switch() {
> 	uint8 before, after;
> 	before = rboot_get_current_rom();
> 	if (before == 0) after = 1; else after = 0;
> 	Serial.printf("Swapping from rom %d to rom %d.\r\n", before, after);
> 	rboot_set_current_rom(after);
> 	Serial.println("Restarting...\r\n");
> 	System.restart();
> }

I can send an archive of the whole project for you to test if you would like to do so?

slaff, 2018-08-16:
> I can send an archive of the whole project for you to test if you would like to do so?

That is something that can definitely help but I will be able to review it after a month on a real device. Before we go that further why don't you try to flash the original unmodified rBoot sample and check if that one is working. If it also does not work then take care to `flashinit` the device as [described in our WIKI](https://github.com/SmingHub/Sming/wiki/Application-prints-random-symbols-and-restarts).

kmihaylov, 2018-08-16:
Hello slaff,

Since I'm building the project on another machine (I'm flashing it on my windows machine), I had to track exactly what flashinit does to perform it. Actually I'm using similar command. 

Here is what I run manually, but should completely correspond to flashinit:

> esptool.py --port COM3 --baud 230400 erase_flash
> esptool.py --port COM3 --baud 230400 write_flash --flash_freq 40m --flash_mode qio --flash_size 4MB 0x3fc000 esp_init_data_default.bin 0x3fe000 blank.bin 0x100000 blankfs.bin

For some readon the ESP doesn't respond to serial commands (from the example), I added the following lines to the bottom of init();

> WifiStation.config(WIFI_SSID, WIFI_PWD);
> WifiStation.enable(true);
> WifiStation.connect();
> WifiEvents.onStationGotIP(gotIP);

gotIP:
> void gotIP(IPAddress ip, IPAddress netmask, IPAddress gateway)
> {
> 	OtaUpdate();
> }

And the result is here. Actually now it works!

> rll▒▒|▒l▒|
> ▒
> l▒
> b|▒▒▒▒r▒b▒
> b▒▒nn▒lnn▒▒▒
> bp▒▒lrlrlp▒n▒
> ▒
> l
> b
> n▒|l▒
> ▒▒b▒▒nn▒l▒▒l`▒nn
> l`nr▒▒▒n
> ▒`p▒n▒
> r▒▒▒▒▒
> b
> n▒|▒▒▒
> b▒▒nn▒
> l`▒nnl▒l`nr▒▒▒n
> b▒▒`r▒▒n
> b▒▒`rl▒b▒nn▒▒▒p|
> ▒blb▒np`b▒b▒b▒▒nn▒n▒b▒nbp~▒nl▒n▒n▒n~b
> ▒
> llpb▒▒`▒n▒nlln▒▒▒
> l▒lll▒n▒n▒▒▒▒b
> ▒
> lll~rl▒b▒nn▒▒l▒nnn▒n▒▒▒n▒nlrl▒bn▒▒|▒
> ▒▒▒▒n▒▒▒nn▒▒▒nn▒▒▒bl
> ▒nn▒▒|▒▒▒n▒
> ▒bl▒▒▒▒n▒r▒▒n|▒
> l
> lll`▒▒r▒l▒l▒
> l`▒▒r▒l▒l▒
> l`▒▒r▒l
> ▒▒▒
> ll`▒▒▒n▒nlrlStrying to mount spiffs at 0x00100000, length 65536
> fs.start: size:64 Kb, offset:0x100000
> 
> mount res: 0
> 
> 
> Currently running rom 0.
> Type 'help' and press enter for instructions.
> 
> Station configuration was updated to: vila-net-3
> mode : sta(5c:cf:7f:13:99:a6)
> add if0
[... 185 lines cut ...]
> rBoot Option: Big flash
> 
> Booting rom 1.
> ▒▒▒▒n▒r▒▒n|▒
> l
> lll`▒▒r▒l▒▒trying to mount spiffs at 0x00300000, length 65536
> fs.start: size:64 Kb, offset:0x300000
> 
> mount res: 0
> 
> 
> Currently running rom 1.
> Type 'help' and press enter for instructions.
> 
> This is the rom from the OTA update!
> mode : null
> 

To be honest I had a mistake with my esptool script, instead of 4M I was using 4M-c1 (my version doesn't support -fs 32m), also I was flashing the default_data to few other addresses...

Also Fatal esception 28 is different from my previous fatal exception 0.

So now it downloads the fw and reflashes it. I'm using fomain name, not IP (FYI) so I'm happy. You can give us more light on what the error is, but otherwise the OTA works...

BTW: I don't know why the Basic_rBoot doesn't accept any commands. Any ideas? For my project I use the CommandProcessing class which works good.

juppin, 2018-08-26:
I´ve the same problems and can´t get the ota update working...
I´ve tried also the Basic_rBoot example but always with the same error.
For the Basic_rBoot example i´ve changed only the url´s and my wifi credentials.

I´ve also tried latest Sming 3.5 and 3.1 and SDK 1.5.4 and 2.0.0 and have always the same error.

My development board is a NodeMCU v0.9 and my devel system is debian stretch.

Here is my serial log:
```
/home/johann/dev/esp8266/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 write_flash -ff 40m -fm qio -fs 32m 0x00000 out/firmware/rboot.bin 0x01000 /home/johann/dev/esp8266/esp-open-sdk/sdk/bin/blank.bin 0x02000 out/firmware/rom0.bin 0x100000 out/firmware/spiff_rom.bin
esptool.py v1.2
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0040
Writing 4096 @ 0x0... 4096 (100 %)
Wrote 4096 bytes at 0x0 in 0.4 seconds (89.8 kbit/s)...
Writing 4096 @ 0x1000... 4096 (100 %)
Wrote 4096 bytes at 0x1000 in 0.4 seconds (89.8 kbit/s)...
Writing 303104 @ 0x2000... 303104 (100 %)
Wrote 303104 bytes at 0x2000 in 26.3 seconds (92.3 kbit/s)...
Writing 65536 @ 0x100000... 65536 (100 %)
Wrote 65536 bytes at 0x100000 in 5.7 seconds (92.1 kbit/s)...
Leaving...
python -m serial.tools.miniterm /dev/ttyUSB0 115200 
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Booting rom 0.
��s�l�$␓�␌␌�l ␃��{�l�l␛�␌␄␌d`␃��;�$␌��␀�␄␌d$`␃sd�trying to mount spiffs at 0x00100000, length 65536
fs.start: size:64 Kb, offset:0x100000

mount res: 0

Currently running rom 0.
Type 'help' and press enter for instructions.

mode : null
mode : sta(5c:cf:7f:81:3c:b5)
add if0
Station configuration was updated to: home-A
del if0
usl
mode : null
mode : sta(5c:cf:7f:81:3c:b5)
add if0
f r0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt 

connected with home-A, channel 1
dhcp client start...
ip:192.168.0.111,mask:255.255.255.0,gw:192.168.0.254
Updating...
pm open,type:2 0
In callback...
Firmware updated, rebooting to rom 1...
state: 5 -> 0 (0)
rm 0
pm close 7
del if0
usl

***** Fatal exception 28
pc=0x4023efde sp=0x3ffffc10 excvaddr=0x00000068
ps=0x00000030 sar=0x00000000 vpri=0x3ff1fe00
r00: 0x4023efd4=1076096980 r01: 0x3ffffc10=1073740816 r02: 0x00000000=         0 
[... 219 lines cut ...]
0x40104c3e: wDev_ProcessFiq at ??:?
0x40224d78: etharp_output at ??:?
0x40104a94: wDev_ProcessFiq at ??:?
0x402246d4: ip_output_if_opt at ??:?
0x40227069: tcp_input at ??:?
0x40224708: ip_output_if at ??:?
0x40220f98: pbuf_alloc at ??:?
0x40224748: ip_output at ??:?
0x40222601: tcp_send_empty_ack at ??:?
0x40241a67: TcpConnection::staticOnReceive(void*, tcp_pcb*, pbuf*, signed char) at /home/johann/dev/esp8266/Sming/Sming/SmingCore/Network/TcpConnection.cpp:649
0x4022706c: tcp_input at ??:?
0x40221981: tcp_slowtmr at ??:?
0x40224451: ip_input at ??:?
0x40221b7c: tcp_tmr at ??:?
0x40224e15: ethernet_input at ??:?
0x401031fe: lmacTxFrame at ??:?
0x40103619: ppEnqueueRxq at ??:?
0x401033af: ppProcessTxQ at ??:?
0x401033ea: ppProcessTxQ at ??:?
0x4020706f: ets_snprintf at ??:?
0x40000f49: ?? ??:0
0x40000f49: ?? ??:0
```

Due to that also the old Sming versions throws the same errors i think this is related to the sdk or esptool?