SmingHub/Sming #1293: Upgrade 3.2.0 to 3.4.0 problem sendHttp - request queue is full at the moment

fernandomorse, 2017-12-02:
Case:

I am upgrading from my Sming version 3.2.0 to version 3.4.0, I ran the test below, and I have some problems and doubts.

Problem:

Only the first request gets the callback, after a few seconds waitingQueue.count reaches its maximum value, and the message "The request queue is full at the moment" is displayed.

How can I read and execute this queue of requests (HttpConnection :: onReadyToSendData: waitingQueue.count: 20)

Thanks for the help.

Code:

```
HttpClient httpClient;
HttpHeaders headers;
Timer pushTimer01,pushTimer02,pushTimer03;

int callBackHttp(HttpConnection& connection, bool success)
{
    Serial.println(connection.getResponseString());
}

void ApiDateTime()
{
    httpClient.send(
        httpClient.request("10.1.1.22/api/test/time")        
        ->setMethod(HTTP_GET)
        ->onRequestComplete(callBackHttp) 
    );
}

//Station connected
void connected(IPAddress ip, IPAddress mask, IPAddress gateway)
{	
	debugf("Connected");	
	pushTimer01.initializeMs(1000,ApiDateTime ).start();
	pushTimer02.initializeMs(1000, ApiDateTime).start();
	pushTimer03.initializeMs(1000, ApiDateTime).start();
}

// Connection failed
void connectionFailed(String ssid, uint8_t ssidLength, uint8_t *bssid, uint8_t reason)
{
	debugf("Not Connected");
	system_restart();
}

void init()
{
    Serial.begin(SERIAL_BAUD_RATE); // 115200 by default
    Serial.systemDebugOutput(true); // Debug output to serial

    WifiStation.config(WIFI_SSID, WIFI_PWD);
    WifiStation.enable(true);
    WifiAccessPoint.enable(false);

    WifiEvents.onStationGotIP(connected);
    WifiEvents.onStationDisconnect(connectionFailed);        

}  
```

Here is my flash/terminal log:

Connected
Creating new httpConnection
HttpConnection::connect: TCP state: -1, isStarted: 0, isActive: 0
HttpConnection::connecting ...
[... 912 lines cut ...]
TCP received: (null)
TCP sent: 67
HttpConnection::onReadyToSendData: waitingQueue.count: 20
Storing 46 bytes in stream
realloc 46 -> 194
Storing 19 bytes in stream
Storing 2 bytes in stream
onReadyToSendData: 2
Written: 67, Available: 67, isFinished: 1, PushCount: 1 [TcpBuf: 2713]
TcpClient stream finished
HTTP parser error: HPE_CB_message_complete
TCP received: (null)
TcpConnection::staticOnReceive: pbuf is NULL
TCP received: (null)
TCP connection closing
-TCP connection
The request queue is full at the moment
The request queue is full at the moment
The request queue is full at the moment
The request queue is full at the moment
The request queue is full at the moment
The request queue is full at the moment
The request queue is full at the moment
The request queue is full at the moment
The request queue is full at the moment

slaff, 2017-12-03:
```
TcpClient stream finished
HTTP parser error: HPE_CB_message_complete
```

The error message here means that the callback for message_complete didn't not `return 0` to signal success. But your code is saying something else. 

```
int callBackHttp(HttpConnection& connection, bool success)
{
    Serial.println(connection.getResponseString());
    return 0;
}
```

Can it be that you are pasting here one code and actually using another code?

fernandomorse, 2017-12-03:
Thanks @slaff  
I checked that in my code, I had no return 0. Problem solved.

fernandomorse, 2017-12-07:
Testing the code below, verified that after some runtime, the queue is full at the moment

Code:

```
Timer pushTimer1,pushTimer2,pushTimer3,pushTimer4,pushTimer5,pushTimer6;
HttpClient httpPusher;

int onDataSent(HttpConnection& client, bool successful)
{
	if(successful)
	{
		Serial.println("###### Success sent");
		Serial.println("###### Success sent");
		Serial.println("###### Success sent");
	} else {
		Serial.println("###### Failed");
		Serial.println("###### Failed");
		Serial.println("###### Failed");
	}
	return 0;
}

void pushItTime1()
{
	Serial.println("###### downloadString 1");
	Serial.println("###### downloadString 1");
	Serial.println("###### downloadString 1");
	httpPusher.downloadString("http://pudim.com.br/", onDataSent);
	
	Serial.print("Free memory: ");
	Serial.println(system_get_free_heap_size());// SEE, FREE MEMORY ALWAYS DECREASING!!
}

void pushItTime2()
{
	Serial.println("###### downloadString 2");
	Serial.println("###### downloadString 2");
	Serial.println("###### downloadString 2");
	httpPusher.downloadString("http://pudim.com.br/", onDataSent);
	
	Serial.print("Free memory: ");
	Serial.println(system_get_free_heap_size());// SEE, FREE MEMORY ALWAYS DECREASING!!
}

void pushItTime3()
{
	Serial.println("###### downloadString 3");
	Serial.println("###### downloadString 3");
	Serial.println("###### downloadString 3");
	httpPusher.downloadString("http://pudim.com.br/", onDataSent);
	
	Serial.print("Free memory: ");
	Serial.println(system_get_free_heap_size());// SEE, FREE MEMORY ALWAYS DECREASING!!
}

void pushItTime4()
{
	Serial.println("###### downloadString 4");
	Serial.println("###### downloadString 4");
	Serial.println("###### downloadString 4");
	httpPusher.downloadString("http://pudim.com.br/", onDataSent);
	
	Serial.print("Free memory: ");
	Serial.println(system_get_free_heap_size());// SEE, FREE MEMORY ALWAYS DECREASING!!
}

void pushItTime5()
{
	Serial.println("###### downloadString 5");
	Serial.println("###### downloadString 5");
	Serial.println("###### downloadString 5");
	httpPusher.downloadString("http://pudim.com.br/", onDataSent);
	
	Serial.print("Free memory: ");
	Serial.println(system_get_free_heap_size());// SEE, FREE MEMORY ALWAYS DECREASING!!
}

void pushItTime6()
{
	Serial.println("###### downloadString 6");
	Serial.println("###### downloadString 6");
	Serial.println("###### downloadString 6");
	httpPusher.downloadString("http://pudim.com.br/", onDataSent);
	
	Serial.print("Free memory: ");
	Serial.println(system_get_free_heap_size());// SEE, FREE MEMORY ALWAYS DECREASING!!
}

void STAGotIP(IPAddress ip, IPAddress mask, IPAddress gateway)
{
	Serial.printf("GOTIP - IP: %s, MASK: %s, GW: %s\n", ip.toString().c_str(),
																mask.toString().c_str(),
																gateway.toString().c_str());
    
	pushTimer1.initializeMs(1000, pushItTime1).start();
	pushTimer2.initializeMs(1000, pushItTime2).start();
	pushTimer3.initializeMs(1000, pushItTime3).start();
	pushTimer4.initializeMs(1000, pushItTime4).start();
	pushTimer5.initializeMs(1000, pushItTime5).start();
	pushTimer6.initializeMs(1000, pushItTime6).start();
}
```
Code Log:
https://pastebin.com/fJMHe4Bw

Similar error when using:

```
  HttpClient httpClient;

     httpClient.send (
         httpClient.request (url)
         -> setMethod (HTTP_GET)
         -> onRequestComplete (callBackHttp)
     );

```
For six instances

fernandomorse, 2017-12-07:
In this version it is not possible to use isProcessing in HttpClient, as it would be a palliative way to suppress this error.

void pushItTime1()
{
   if (pushItTime1.isProcessing()) return; else {      
	Serial.println("###### downloadString 1");
	Serial.println("###### downloadString 1");
	Serial.println("###### downloadString 1");
	httpPusher.downloadString("http://pudim.com.br/", onDataSent);
	
	Serial.print("Free memory: ");
	Serial.println(system_get_free_heap_size());
    }
}

dpechman, 2018-06-04:
for replacement of the isProcessing in HttpClient its ok to use cleanup()?
it seems to fix the "request queue is full at the moment" if called before downloadString

closed 2018-11-07 by slaff