SmingHub/Sming #1294: Memory Heap

fernandomorse, 2017-12-03:
**Code:**

```
Timer pushTimer01,pushTimer02;

void executaAPI01();
void executaAPI02();

void sendHTTP(String url,String secretkey);
int callBackHttp(HttpConnection& connection, bool success);

int callBackHttp(HttpConnection& connection, bool success)
{
    if(success) {
        Serial.println(connection.getResponseString());
        Serial.println(connection.getResponseCode());
    }
    return 0;
}

void sendHTTP(String url,String secretkey)
{
    HttpClient httpClient;
    HttpHeaders headers;
    
    headers["Authorization"] = "Bearer "+secretkey;
    headers["Connection"]="Close";
    headers["ContentType"] = "application/json";

    httpClient.send(
        httpClient.request(url)        
        ->setMethod(HTTP_GET)
        ->setHeaders(headers)
        ->onRequestComplete(callBackHttp) 
    );

    Serial.println("############################################");
    String heap = String(system_get_free_heap_size());
    Serial.println("Memory HEAP "+heap);
    Serial.println("############################################");
}

void executaAPI01()
{
    sendHTTP("http://10.1.1.22/api/device/7B16E2CF93B8E6A573A58536/info","12A2A34DC412379C3D8C4B8F");
}

void executaAPI02()
{
    sendHTTP("http://10.1.1.22/api/device/DA4EE7A60810523513FDA50C/info","BD22CDB878618CA8D39EC937");
}

//Station connected
void connected(IPAddress ip, IPAddress mask, IPAddress gateway)
{
	debugf("Connected");	
	pushTimer01.initializeMs(3000, executaAPI01).start();
	pushTimer02.initializeMs(3000, executaAPI02).start();
}

```

**Problem:**

Memory heap decreases until reset esp8266.

**Here is my flash/terminal log:**

Connected
pm open,type:2 0
[... 3694 lines cut ...]
3fffff20:  3ffe899a 3ffea764 00000000 60000600  
3fffff30:  0a24ead5 3ffea030 3ffeffa8 402376c0  
3fffff40:  3fff0c40 00180018 3fffaf48 00390039  
3fffff50:  40219b64 3ffea764 3ffea030 40239408  
3fffff60:  40209f1c 3ffeffac 3ffea030 4010656a  

3fffff70:  60000600 3ffeffac 3ffeffac 40209f29  
3fffff80:  40209f6e 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffea050 40000f65 3fffdab0 00000000  
3fffffa0:  40000f49 3ffea9e0 ffffff01 55aa55aa  
================================================================

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 26860, room 16 
tail 12
chksum 0x65
ho 0 tail 12 room 4
load 0x3ffe8000, len 900, room 12 
tail 8
chksum 0xad
load 0x3ffe8390, len 6796, room 0 
tail 12
chksum 0xd1
csum 0xd1

slaff, 2017-12-04:
@fernandomorse Is the code above the one that you are actually testing or it again a different one?

fernandomorse, 2017-12-04:
@slaf  Você anda problema its diferente.

fernandomorse, 2017-12-05:
Issue # 1138 , verified that it works perfectly with 2 timer calls. To match the tests, I put the code below:

```
Timer pushTimer01,pushTimer02;

void executaAPI01();
void executaAPI02();

void sendHTTP(String url);
int callBackHttp(HttpConnection& connection, bool successful);

int callBackHttp(HttpConnection& connection, bool successful)
{
	if(successful)
	{
		Serial.println("###### Success sent");
		Serial.println("###### Success sent");
		Serial.println("###### Success sent");
	} else {
		Serial.println("###### Failed");
		Serial.println("###### Failed");
		Serial.println("###### Failed");
	}
	return 0;
}

void sendHTTP(String url)
{
    HttpClient httpClient;
    HttpHeaders headers;
    
    headers["Connection"]="Close";
    headers["ContentType"] = "application/json";

    httpClient.send(
        httpClient.request(url)        
        ->setMethod(HTTP_GET)
        ->setHeaders(headers)
        ->onRequestComplete(callBackHttp) 
    );

    String heap = String(system_get_free_heap_size());
    Serial.println("Memory HEAP "+heap);   
}

void executaAPI01()
{
    sendHTTP("http:/pudim.com.br/");
}

void executaAPI02()
{
    sendHTTP("http://pudim.com.br/");
}

//Station connected
void connected(IPAddress ip, IPAddress mask, IPAddress gateway)
{
	debugf("Connected");	
	pushTimer01.initializeMs(1000, executaAPI01).start();
	pushTimer02.initializeMs(1000, executaAPI02).start();
}

```

Memory decreases until crash.

fernandomorse, 2017-12-05:
@slaff the codes are exactly what I'm testing. Sorry the previous answer i was wrong

fernandomorse, 2017-12-06:
I noticed that not using the header, or just a header does not happen the problem. example

No problem:

 HttpHeaders headers;   
 headers["Connection"]="Close";

Problem:

 HttpHeaders headers;    
 headers["Connection"]="Close";
 headers["ContentType"] = "application/json";

slaff, 2017-12-06:
> I noticed that not using the header, or just a header does not happen the problem

That is a very useful information! And thank you for it. I will revise the HttpRequest class once I have more time. If you don't want to wait for me take a look at it yourself and see if you can find something that is related to the issue and should be fixed.

fernandomorse, 2017-12-06:
Okay, I'll try. thank you

slaff, 2018-06-15:
@fernandomorse See the discussion here #1394.

closed 2018-06-15 by slaff