SmingHub/Sming #1303: PWM stopped working

MayaPosch, 2017-12-21:
With a pre-3.4.0-release build of Sming, the PWM functionality (using the new PWM lib) was working properly. With current develop builds this functionality seems hopelessly broken. After trying with current develop branch on both Windows (WSL, Ubuntu 16.04) and Linux (Mint 18.x), the same result:

PWM frequency set to 1 kHz, duty set to 0%: 0V on output pin.
PWM frequency set to 1 kHz, duty set to >0%: 3.3V on output pin, no PWM.

Compiling the same application code using the aforementioned Sming build results in the pin output being as expected.

I first looked at the PWM library itself, trying to use the NMI timer instead of TIMER1: https://github.com/StefanBruens/ESP8266_new_pwm/issues/18

Did something change in Sming with regard to timer usage that would prevent the PWM library from using either timer? The behaviour which I am observing is apparently due to the PWM library being unable to use the selected timer, causing digital output instead of PWM.

slaff, 2018-01-12:
That is probably related to this change: https://github.com/SmingHub/Sming/commit/c2cc852e40b1971a1987aceeedab5b28b8c7297c#diff-95f5a43cb31a804075bd87e65b4be18aR249 

which finally activates your change: 

https://github.com/SmingHub/Sming/commit/d408e7de1382c22c2ac0289a1a074821eb504186#diff-95f5a43cb31a804075bd87e65b4be18a . 

Before that the compiler directive `-DSDK_PWM_PERIOD_COMPAT_MODE=1` was not really added to the compiler arguments.

MayaPosch, 2018-01-12:
I was pretty sure that I had recompiled Sming without the parameter to the PWM library. Maybe I made a mistake there, though.

Regardless, the PWM library is supposed to still work no matter its configuration. The behaviour which I'm seeing is that the PWM library fails to acquire any timer. Something which might also be due to the new SDK compatibility?

perpernorbi, 2018-08-24:
This will fix it: https://github.com/SmingHub/Sming/pull/1433

MayaPosch, 2018-09-15:
Confirmed fixed with #1433.

closed 2018-09-15 by MayaPosch