SmingHub/Sming #1304: Makefile - can not have same filenames in different locations

cr1st1p, 2017-12-22:
I have one library (added via MODULES, in my Makefile) that has a base Controller.cpp file - in one namespace.
Then, in my application, I have app/Controller.cpp , in a different namespace (a class  derived from the library one)
The problem is that current Makefile will compile the library one, but not also the app one.
This is because of the vpath setting and implicit .cpp rules, which makes Make find that 
*build/app/Controller.o dependency is on my module/Controller.cpp instead of app/Controller.cpp

There is a mention about this type of issue also in http://www.conifersystems.com/whitepapers/gnu-make/ , chapter "Pattern Rules and Search Paths"

cr1st1p, 2017-12-22:
I'm on my way to fix this. Bumped also on a problem with the archive program ('ar') which also has problem with same filenames.

cr1st1p, 2018-01-04:
Forgot to mention - Pull request is #1305

slaff, 2018-02-28:
Fixed with that rebase #1335

closed 2018-02-28 by slaff