SmingHub/Sming #1310: Please add 'const' to class member functions that are not modifying the object

cr1st1p, 2018-01-04:
Example I bumped into, just now:
DateTime::toShortDateString() 
Code does not change internal state of the object, so it should be declared as 'const', so that I'm allowed to use it on a 'const DateObject&'.

I think most of DateTime functions should be const.

Thank you.

slaff, 2018-01-08:
@cr1st1p Can you make a PR  with the suggested changes?

frankdownunder, 2018-05-21:
There are static code analysis tools that would tell us which member functions can be marked const, Ill see if i can find one.

slaff, 2018-05-21:
For the use of "const" we should make the changes across the whole Sming Framework classes and not just here and there. A "const" change should be merged with that PR #1384.

closed 2024-07-04 by mikee47