SmingHub/Sming #1311: Memory Leak in IR_lib sample

horde3d, 2018-01-07:
Maybe I have missed something, but when I look into the IR_Lib sample I can't see a reason for allocating the sendbuff variable. It looks like a memory leak to me.
```
if(irrecv.decode(&dresults)==DECODED){
        irTimer.stop();
        unsigned int * sendbuff = new unsigned int[dresults.rawlen-1];
        for(int i=0; i<dresults.rawlen-1; i++){
            sendbuff[i]=dresults.rawbuf[i+1]*50;
        }
        irsend.sendNEC(dresults.value, dresults.bits);
        Serial.println("Sent IR Code");
        irrecv.enableIRIn();
        irTimer.start();
}
```
Maybe it would also make sense to add a check for dresults.decode_type == NEC before sending the dresults value via sendNEC. 
I understand that the sample is for demonstration purpose only but IMHO it should at least does not do nonsense.

slaff, 2018-01-08:
@horde3d You are absolutely right! Can you make a PR request with the suggested changes?

horde3d, 2018-01-08:
I've played around with the IR lib implementation of Sming a bit. I had some troubles when trying to decode a JVC remote control. After some maybe unrelated changes I ended up having all codes being decoded as Sanyo remote codes. That brought me to the solution that the Sanyo decoding is buggy in general. Same conclusion seems to be made by David Conran at https://github.com/markszabo/IRremoteESP8266 (see the comment on the Sanyo S866500B decoder).
So to cut a long story short, I will try to create a pull request with an updated example as soon as I updated the IR lib within Sming to the version of https://github.com/markszabo/IRremoteESP8266.
BTW. Was there a special reason why the implementation within Sming currently uses a timer instead of an interrupt for measuring the IR codes? I just started to play around with the ESP8266 but to me measuring the IR signals seem to be a time critical thing that should be done by IRQs and not with timers or did I miss something here?

slaff, 2018-01-12:
> Was there a special reason why the implementation within Sming currently uses a timer instead.

Looking at the code I would say 'no'. Please, in the PR give your best to update the example and we will review it.

horde3d, 2018-01-28:
I finally integrated the IRRemoteESP8266 library. I changed the implementation a bit. Based on my observations it wasn't possible to decode two commands in a row, because until the first one is decoded no other signals were recorded. So I integrated the RingBufCPP module from D. Aaron Wisner, to have the possibility to record more than one command in case they follow in a very short period.
The updated sample does reply what is received. I tested it with my JVC remote. Although it seems like the code send is received again correctly (after all the sample is an endless loop when you send what you receive) I wasn't able to trigger my JVC device with it. But with a Yamaha device it did what it should (pressing once the loudness up button, the ESP repeats that command and the device's volume goes up). 
Next to the JVC code not received by my JVC device problem, I also had various hardware watchdog resets in another test sample I did for me (based on the RBoot sample with IRRecv code integrated). As I didn't see those WDT problems with the simple IR_Lib sample, it might be something else, but I couldn't find out what might be the problem.
Maybe someone else want to have a look at the code https://github.com/horde3d/Sming

slaff, 2018-01-29:
@horde3d I created a PR from out fork #11318 where we can discuss the code and possible improvements, issues, etc.

closed 2018-05-10 by slaff