SmingHub/Sming #1317: Understanding Timer in Sming

igormorse, 2018-01-28:
When you call a Timer on Sming, each call is a new Thread executing ? 

**For Example:**

```
Timer refreshTimer;

refreshTimer.initializeMs(1000, TimerDelegate(&function, this)).startOnce();

refreshTimer.initializeMs(100, TimerDelegate(&function, this)).startOnce();
```

I know that ESP8266 have only one core and its not possible to run things in parallel in the same time. Are Timers put in some Linear Schedule to be executed after every call of it ? ( Feels like things running in parallel but they are not )

I would be really appreaciated if I was given a quick explanation of how Timers works in background. I really need to understand their behaviour for my project.

Thanks,
Igor Morse.

Vinimuller, 2018-10-09:
I was wondering the same thing yesterday, does anyone has an answer?

mikee47, 2018-10-09:
There may be a better explanation elsewhere, but here's my take on things.

It's probably easiest to look at an example, so let's go with Samples/Basic_Blink. If you take  a look in application.cpp, this is all there is:

```
#include <user_config.h>
#include <SmingCore/SmingCore.h>

#define LED_PIN 2 // GPIO2

Timer procTimer;
bool state = true;

void blink()
{
	digitalWrite(LED_PIN, state);
	state = !state;
}

void init()
{
	pinMode(LED_PIN, OUTPUT);
	procTimer.initializeMs(1000, blink).start();
}
```

So what's happening? Well, the whole show is actually run by the _ESP8266 Non-OS SDK_, provided by _Espressif_ (Sming is a C++ framework built on top of that). The main 'loop' lives inside their code, so it's hidden.
At startup, `init()` gets called by the SDK; we set the LED output pin mode and start a timer, to repeat every 1s. We also tell the timer to call our `blink()` function when it expires. Control then returns to the SDK.
When `procTimer` expires, `blink()` gets called by the SDK and we do our thing and return control back to the SDK.

@igormores, the answer to your question is yes, Timers are put in a schedule, which is managed by the SDK. In your example, you initialise two timers and both of them will call `this->function`, but only once each.

Sming is event-driven, so things happen and we get called to do work. The SDK has the WiFi to manage, amongst other things, so we have to co-operate with it and avoid spending too much time in our callbacks. This is why Sming is so fast, because it doesn't hang around sitting in loops.

closed 2018-11-07 by slaff