SmingHub/Sming #1323: LMIC port to SmingHub

3nd0y, 2018-02-09:
Hi all,

I love sming because i could compile source code faster than using arduino/platformio, but unfortunately not all library from arduino can be used in sming. One of them is LMIC, some define can't recognize after compile, and a few of them there's conflict name used.

Is someone has working for this library?

Regards,
Fadli

slaff, 2018-02-09:
@3nd0y Can you give us the git repository URL of the library and we can try to compile it with Sming?

3nd0y, 2018-02-16:
@slaff hi sorryy late to reply. Here the library i try use arduino and platform IO are working well
https://github.com/things4u/LoRa-LMIC-1.51

3nd0y, 2018-02-17:
@slaff the official repository was from https://github.com/matthijskooijman/arduino-lmic but seems it's only for AVR, so i use link above (https://github.com/things4u/LoRa-LMIC-1.51) since it working for ESP8266 at arduino/platformio enviroment