SmingHub/Sming #1328: Better out-of-the box experience

slaff, 2018-02-19:
Sming is fantastic and very performance framework but the initial experience is still a bit "rough" for non-advanced developers. This task is created to collect the things that can be improved:

- [ ] Less debug messages in the TCP stack. This way all network related pieces will work much faster out-of-the box and the console output will contain only important warning or error messages. (See #1327).
- [ ] Easy-to-use installer for Windows, MacOS X and Linux. (probably based on @riban-bw's [SmingRelease](https://github.com/riban-bw/Sming-Release) )
- [ ] Preconfigured and pre-compiled toolchains for Windows, Mac OS X and Linux. Not all users interesting in the framework are skilled enough to compile the toolchain from source.

riban-bw, 2018-02-19:
I completely agree. Last two points are the aspiration of my installer. It may be advantageous to automate the build of the precompiled toolchain to reduce dependencies on individual having to manually maintain it but that may be a later stage, after getting something we are happy with. I would also add to this list, better introductory documentation and examples. I feel it can be challenging to get started even after installing everything.

slaff, 2018-02-19:
@riban-bw The Mac OS X toolchain is updated and a pre-compiled version can be downloaded from [here](https://dl.bintray.com/sminghub/toolchain/esp-open-sdk-2.0-macosx.tgz). You can put also the other pre-compiled toolchains in `bintray.com` so that you don't have to worry about traffic limits on personal dropbox accounts.

riban-bw, 2018-04-11:
I am updating Sming-Release to 3.5.1 and moving packages to bintray as suggested. Also adding the OS X toolchain (thanks @slaff). I will try to get this done this week. Sming-Release was going to be something a bit different but is now an installer so maybe I should change the name of the github project to Sming-Installer?

slaff, 2018-04-11:
> ... should change the name of the github project to Sming-Installer?

Sming-Installer sounds great and once ready it can become part of the main distribution.  You can check also those projects that are dealing with easier installation, or better getting started: https://github.com/zgoda/sming-qstart, https://github.com/ajrkerr/sming-osx,

riban-bw, 2018-04-16:
Sming-Release renamed Sming-Installer (https://github.com/riban-bw/Sming-Installer)..
Moved packages to bintray.
Used OS X toolchain supplied by slaff.
Still working on Sming 3.5.1 update.

closed 2021-11-16 by slaff