SmingHub/Sming #1331: DHT library doesnt work

frankdownunder, 2018-02-25:
The sming sample Humidity_DHT22 doesnt work. I am using a Wemos D1 mini and the Wemos DHT22 shield using pin 2.  SDK:  2.0.0(5a875ba)

To investigate this I ported the Arduino code DHTesp library [ Arduino ESP library for DHT11, DHT22, etc Temp & Humidity Sensors Optimized libray to match ESP32 requirements  ]
and that did work OK under sming. 

My recommendation is to ditch the existing code and replace it with the DTHesp library 

Observations:
The existing library does not have a default constructor.  The problem with this is that there are occasions when you want to defer the decision as to which pin till run-time.

The DHTesp library also supports the ESP32.

Poorly named variables eg "humid" and "temp"instead of "humidity" and "temperature"

I feel the definition of ONE_DURATION_THRESH_US is suspicious.
#define ONE_DURATION_THRESH_US 30
According to the data sheet,  a '1' is if HIGH lasts 70us not 30us

In the DHTesp library, I found that getTempAndHumidity
didn't work on its own, but after replacing it with the calls
	float humidity = dht.getHumidity();
	float temperature = dht.getTemperature();
it worked.  I couldnt see a reason for that behaviour in the code however

Should I make a PR with the DHTesp as a replacement? Thoughts?

ADiea, 2018-02-26:
Hi @frankdownunder I am the original author of the current DHT lib
Since I don't have time currently to keep it up to date and fix your observations I think it would be wiser to replace it completely with the up to date version from arduino
Best regards

slaff, 2018-02-26:
> My recommendation is to ditch the existing code and replace it with the DTHesp library

If DHTesp is supported then it would be better to replace the current unsupported DHT library with it. @frankdownunder Please, submit a PR and I will try to revise it later this week.

frankdownunder, 2018-02-27:
I will do this.