SmingHub/Sming #1337: Make Delegators Compatible with Lambdas and std::function

zigarrre, 2018-03-02:
## Problem

Currently one can only create `Delegate` instances from raw function pointers or from a pointer to a member function and a pointer to a corresponding instance.

This means that `Delegate` does not work with modern C++ features such as lambda functions or `std::function`. (non-capturing lambdas already work as they can be casted to function pointers but captures are a integral part of lambdas that makes them much more useful)

This would be a minimum code example of using lambdas with delegates wich currently does NOT work:

```c++
Timer t;
int foo = 123;
t.initializeMs(1000,
    [foo] {
        // do something with foo
    }).start();
```

This trivial example might not fully show the usefulness of captures but it becomes obvious with more complex code. For example you can also capture `this` which means you could write callbacks using lambdas where you can access other members from within the callback as if it where part of the enclosing class. So without a doubt this would be an extremely useful feature to have.

## Solution

The `Delegate` class should accept ether a `std::function` or - probably even better - use a template to accept anything that is callable.

I don't fully understand all the stuff that is going on in `Delegate.h` so I don't feel comfortable making those changes myself in fear of unknown consequences.

slaff, 2018-03-05:
It would be probably better to replace completely Delegate with std::function as the latter is coming directly since C++ 11, as you correctly noted above. And [std::function](http://en.cppreference.com/w/cpp/utility/functional/function) should cover all the use cases that we have with the Delegates in Sming.

frankdownunder, 2018-03-18:
I had a quick try of std::function, with code like:
`typedef std::function<void(bool)> PumpDisabledDelegate;`
` class Pump `
`{ `
`    public: `
`        Pump(PumpDisabledDelegate pumpDisabledCallback); `
but got linker errors:
out/build/app_app.a(pump.o): In function std::function<void (bool)>::operator()(bool) const:
pump.cpp:390: undefined reference to std::__throw_bad_function_call()

Is there something I need to do to dismiss the exceptions?

slaff, 2018-03-19:
> Is there something I need to do to dismiss the exceptions?

Due to size considerations we are using microc / microgcc and that is most probably causing the problem. You can try to add the following code:

```
namespace std {void __throw_bad_function_call() { while(1); }; }
```

and check again.

frankdownunder, 2018-03-20:
Thanks Slaff that did the trick!

Sming currently provides Delegates to handle callbacks to member functions, eg
```
    ledTimer.initializeMs(500, TimerDelegate(&LedBlinker::ledBlink,this)).start();
```
Sming also uses classic c-style callback functions,
eg InterruptCallback, defined as typedef void (*InterruptCallback)(void);

As I understand it the proposal is to use std::function rather than the Delegate class 

```
typedef Delegate<void()> TimerDelegate;
    becomes  >>>>
typedef std::function<void(void)> TimerDelegate;
```

The header `#include <functional>` must be added,
and this next line added somewhere in the source (to satisfy the linker):
```
namespace std { void __throw_bad_function_call() { while (1); }; }
```

Consider this example, using a function with one bool parameter, 
```
#include <user_config.h>
#include <SmingCore/SmingCore.h>
#include <functional>

typedef std::function<void(bool)> TimingDelegate;

// Needed to satisfy linker
namespace std { void __throw_bad_function_call() { while (1); }; }

class Timing
{
public:
	Timing&		initializeMs(uint32_t milliseconds, TimingDelegate delegateFunction = NULL);
	void		start(bool repeating = true);
private:
	TimingDelegate delegate_func = nullptr;
};

void classicCstyleCallback(bool f)
{
	debugf("TimingDelegate classicCstyleCallback called \n");
	if (f)debugf("T\n"); else debugf("F\n");
}

class Fred
{
public:
	void fredCallback(bool f)
	{
		debugf("fredCallback ");
		if (f)debugf("T\n"); else debugf("F\n");
	}
};

void fnction()
{
	Timing t;
	Fred fred;
	// - - Show use of classic c-style callback
	TimingDelegate a = classicCstyleCallback;
	t.initializeMs(100, a).start(true);
	
	// - - Show use of member function callback
	TimingDelegate b = std::bind(&Fred::fredCallback, fred, std::placeholders::_1);
	t.initializeMs(100, b).start(true);

	// - - Show use of lamda callback
	int foo = 123;
	TimingDelegate c = [foo](bool f)
	{
		if (f && foo == 123)
			debugf("lamda Callback ");
	};
	t.initializeMs(1000, c).start(true);

	// - - Again good measureShow 
	t.initializeMs(1000,
		[foo] (bool f)
		{
			if ( f && foo == 123)
				debugf("lamda Callback ");
		}
	).start(true);
}

Timing& Timing::initializeMs(uint32_t milliseconds, TimingDelegate delegateFunction)
{
	delegate_func = delegateFunction;
	return *this;
}

void Timing::start(bool f)
{
	delegate_func(f);
}

```

Note it handles
- member functions
- classic c-style callback functions
- lamdas

What next?  A PR?

slaff, 2018-03-20:
> What next? A PR?

Yes, of course :). A PR should include major overhaul of all Sming components using delegators + all samples where the delegators are used. The delegator class itself has to be marked as deprecated. Since this change will be a backwards incompatible change we should merge it in a major version.

frankdownunder, 2018-03-20:
Yes a fair bit of work. When are you thinking of doing the next major version?

slaff, 2018-03-21:
> When are you thinking of doing the next major version?

Whenever we have something useful to offer to our users. BUT before that it would be great if we can do that migration in multiple steps:
* add std::function alongside delegators for one commonly used component/class (for example for the `Timer` only) 
* mark the methods that use old Delegators as @deprecated and specify the replacement method.
* make sure that all samples that depend on `Timer` use  `std::function` instead of Delegators.
* merge it in a minor version and allow also others to test and report feedback

And those things can be merged also in the next minor version. Further on:

* If the feedback is positive then we can propagate those changes also to other components/classes.
...

avr39-ripe, 2018-03-21:
Thumbs-up for std::function as replacement of Delegates. I like Sming mostly for its async nature and this was based also on Delegates, so if we can replace it to standard C++ std::function with superior functionality it well be great! We need to make good documentation and example for this feature and also migration guide in form "was such -> become such" :)

frankdownunder, 2018-04-03:
@avr39-ripe @slaff I suspect one or both of you have started on this already.  If not I will.

slaff, 2018-04-11:
> If not I will.

@frankdownunder The initial work is in the `develop` branch. Feel free to add on top of it.

frankdownunder, 2018-04-12:
PR is on its way.

slaff, 2018-04-12:
@zigarrre Can you please check PR #1361?

slaff, 2019-07-04:
Should be fixed with PR #1734 . Try it in the latest `develop` code. 
Feel free to reopen it if there are still pending issues.

closed 2019-07-04 by slaff