SmingHub/Sming #1338: ENABLE_CUSTOM_PWM is Ignored and Custom PWM is Never Used

zigarrre, 2018-03-03:
According to `README.md` the "Custom PWM" implementation is enabled by default but even when `ENABLE_CUSTOM_PWM=1` is added to `Makefile-user.mk` it is NOT used.

To test this I added `debugf("pwm_init");` at the start of the `pwm_init` function in `third-party/pwm/pwm.c` and recompiled Sming with `make clean; make`. Then I compiled and flashed the `Basic_HwPWM` example and confirmed that it did not print `pwm_init` and therefore it does not use the custom pwm implementation.

By looking at the output when building Sming I also noticed that `third-party/pwm/pwm.c` is not getting compiled.

slaff, 2018-03-05:
Can you please try to compile a Sming sample. Any sample. And make sure to compile the sample with V=1

```
make V=1
```

Do you see in the lines at the end related to the linking phase something like `-lpwm_open` ?

zigarrre, 2018-03-05:
Yes I see `-lpwm_open` there.

slaff, 2018-03-05:
Ok - then the samples should be linked against libpwm_open.a. And if the linked does not complain then it means that `libpwm_open.a` is already compiled from the `third-party/pwm/` code. So you are actually using that custom PWM if I am not terribly wrong :)

zigarrre, 2018-03-05:
But where/when is that `libpwm_open.a` compiled? And why do changes to `third-party/pwm/pwm.c` not have any effect?

slaff, 2018-03-05:
The `libpwm_open.a` library is located under `Sming/compiler/lib/`. The compilation of Sming compiles also `libpwm_open.a`. 

> And why do changes to third-party/pwm/pwm.c not have any effect?

Try the following:
```
cd $SMING_HOME
make compiler/lib/libpwm_open.a
# then make a change to third-party/pwm/pwm.c
# run
make compiler/lib/libpwm_open.a
# ^- that should compile the modified library
make compiler/lib/libpwm_open.a
# ^- that should report something like: `make: `compiler/lib/libpwm_open.a' is up to date.`

```

zigarrre, 2018-03-05:
Compilation of Sming does NOT compile libpwm_open.a:

```
cd $SMING_HOME
make clean
# compiler/lib/libpwm_open.a does still exist
rm compiler/lib/libpwm_open.a
make
# compiler/lib/libpwm_open.a does not exists
```

But after running `make compiler/lib/libpwm_open.a` it was compiled and now the changes I made to it's soure code have taken effect. Thank you.

It would probably be a good idea to build this library when Sming is built. I don't even know how it was generated in the first place because it doesn't come with Sming and I never compiled it manually before.

closed 2018-05-10 by slaff