SmingHub/Sming #1339: Memory Leak in member function downloadFile

frankdownunder, 2018-03-04:
When Wifi is unavailable,  the  downloadFile function of class HttpClient leaks
The following code has been lifted out of a normally working program.  On one occasion however, the Wifi was unavailable, and I noticed a memory leak as a result.

`
#include <user_config.h>
#include <SmingCore/SmingCore.h>
#ifndef WIFI_SSID
#define WIFI_SSID	"XXXXX" 
#define WIFI_PWD	"YYYYY"
#endif
HttpClient downloadClient;

void onTimer_checkDownloadNewFirmware();
Timer procTimerCheckDownloadNewFirmware;
void onTimer_printMem();
Timer procTimerPrintMem;
void init()
{
	Serial.begin(921600);
	procTimerPrintMem.initializeMs(1 * 1 * 1000, onTimer_printMem).start();
	procTimerCheckDownloadNewFirmware.initializeMs(1 * 2 * 1000, 
             onTimer_checkDownloadNewFirmware).start();
}

void onTimer_checkDownloadNewFirmware()
{
	downloadClient.downloadFile("http://www.XXXXXX.com/buildVersion.json",
		(RequestCompletedDelegate)([](HttpConnection& connection, bool success) -> int
			{
				if (success) {
					debugf("buildVersion.json success\n");
				}
			}
		)
	);
}
void onTimer_printMem()
{
	Serial.println(String("FreeHeap: ") + system_get_free_heap_size());
}
`

slaff, 2018-03-05:
> On one occasion however, the Wifi was unavailable, and I noticed a memory leak as a result.

@frankdownunder I am not sure if that is a leak or just the fact that the client fills its internal queue because it cannot fulfil the requests. This is expected behaviour and if you want to limit the queueing you can decrease in your own project the length of the queue.

What would be better IMHO is not to use a timer but to queue the next request once the previous one has finished.

slaff, 2018-03-05:
@frankdownunder I will check it myself but I guess you are right.  Probably the fileStream pointer is not freed if `send` returns just `false`. 

```
bool HttpClient::downloadFile(const String& url, const String& saveFileName, RequestCompletedDelegate requestComplete /* = NULL */)
{

       // ...   
	FileOutputStream* fileStream = new FileOutputStream(file);

	return send(request(url)
				   ->setResponseStream(fileStream)
				   ->setMethod(HTTP_GET)
				   ->onRequestComplete(requestComplete)
			  );
}
```

slaff, 2018-03-05:
@frankdownunder Take a look at PR #1342 and tell me if it fixes the issue that you have discovered.

frankdownunder, 2018-03-05:
@slaff  All good now.  It was a minor leak of little consequence, with an easy workaround, but good that it is fixed.  May I ask do you have tools to find leaks? Do you know of any that work on ESP?

PR #1342 (2018-03-06): Make sure to free also the responseStream in the desctructor.
Closes #1339.
closed 2018-03-06 by slaff