SmingHub/Sming #1340: *** Preparation for Release 3.5.2 ***

slaff, 2018-03-05:
This issue is created to collect all the tasks we need to do before the 3.5.2.

### Source Code
- [ ] Update Sming/SmingCore/SmingCore.h 
- [ ] Update Readme.md
- [ ] Merge develop into master
- [ ] Tag the source code in the master branch

### New Features and Improvements
All PRs scheduled for this release can be seen from [here](https://github.com/SmingHub/Sming/milestone/15)

kwis2, 2018-03-19:
Maybe it will be possible to add ability to work with windows ??? #1297, #1241

frankdownunder, 2018-04-21:
Any chance of getting SMTP Sample #1312 into this release?

slaff, 2018-04-21:
The SmtpClient in its current state is not ready for a release. Some days ago I started to write a text with the issues that are there nd how to program in async manner and suddenly ended up with a version that, hopefully soon, will be a full-featured performant SMTP client for embedded devices. 
You can follow the progress from here: https://github.com/slaff/Sming/tree/feature/smtp-client  and that client will replace your PR.

frankdownunder, 2018-04-21:
Excellent.   I will study your code, hopefully will learn some tricks.

From: slaff
Sent: Saturday, 21 April 2018 5:50 PM
To: SmingHub/Sming
Cc: frankdownunder; Comment
Subject: Re: [SmingHub/Sming] *** Preparation for Release 3.5.2 *** (#1340)

The SmtpClient in its current state is not ready for a release. Some days ago I started to write a text with the issues that are there nd how to program in async manner and suddenly ended up with a version that, hopefully soon, will be a full-featured performant SMTP client for embedded devices.
You can follow the progress from here: https://github.com/slaff/Sming/tree/feature/smtp-client and that client will replace your PR.
—

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

kwis2, 2018-04-26:
According to #1354 in my opinion it is high time to say that Sming is not supported on Windows

slaff, 2018-05-02:
v3.5.2 is officially released: https://github.com/SmingHub/Sming/releases/tag/3.5.2.

closed 2018-05-02 by slaff