SmingHub/Sming #1345: Bad define for digitalPinToInterrupt

frankdownunder, 2018-03-11:
----Sming/SmingCore/Interrupts.h ----
The define for digitalPinToInterrupt looks like this:
-#define digitalPinToInterrupt(pin)  ( (p) < ESP_MAX_INTERRUPTS ? (p) : -1 )

It should be like this:
+#define digitalPinToInterrupt(pin)  ( (pin) < ESP_MAX_INTERRUPTS ? (pin) : -1 )

slaff, 2018-03-11:
Yeah, I also noticed this while testing one of the IR libraries. Can you send a PR?

frankdownunder, 2018-03-11:
OK.  

I was hoping to avoid fighting with git again !  I have a subversion background and git gives me a headache.

Frank

From: slaff
Sent: Monday, 12 March 2018 4:14 AM
To: SmingHub/Sming
Cc: frankdownunder; Author
Subject: Re: [SmingHub/Sming] Bad define for digitalPinToInterrupt (#1345)

Yeah, I also noticed this while testing one of the IR libraries. Can you send a PR?
—

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

PR #1346 (2018-03-12): Fix Issue #1345
Fix a spelling error in the define

----
Closes  #1345.
closed 2018-03-12 by slaff