SmingHub/Sming #1351: Error Compiling Sming in Ubuntu Linux 16.04

sanyaade-buildtools, 2018-03-23:
Hello Everybody,

I am having type declaration conflicts in some files in Sming.
Please as I don't know which of those to comment out or leave. Many thanks!
Error Screenshots shown below:
`v4/ip_frag.o lwip/netif/etharp.o lwip/app/dhcpserver.o espconn_dummy.o
make[2]: Leaving directory '/opt/Sming/Sming/third-party/esp-open-lwip'
AS system/irq_check.s
AS gdb/gdbstub-entry.s
CC system/flashmem.c
In file included from include/user_config.h:33:0,
                 from system/flashmem.h:11,
                 from system/flashmem.c:1:
system/include/esp_systemapi.h:48:13: error: conflicting types for 'ets_timer_arm_new'
 extern void ets_timer_arm_new(ETSTimer *ptimer, uint32_t milliseconds, bool repeat_flag, int isMstimer);
             ^
In file included from system/include/esp_systemapi.h:7:0,
                 from include/user_config.h:33,
                 from system/flashmem.h:11,
                 from system/flashmem.c:1:
/opt/esp-open-sdk/sdk/include/osapi.h:65:6: note: previous declaration of 'ets_timer_arm_new' was here
 void ets_timer_arm_new(os_timer_t *ptimer, uint32_t time, bool repeat_flag, bool ms_flag);
      ^
In file included from include/user_config.h:33:0,
                 from system/flashmem.h:11,
                 from system/flashmem.c:1:
system/include/esp_systemapi.h:57:13: error: conflicting types for 'ets_delay_us'
 extern void ets_delay_us(uint32_t us);
             ^
In file included from system/include/esp_systemapi.h:7:0,
                 from include/user_config.h:33,
                 from system/flashmem.h:11,
                 from system/flashmem.c:1:
/opt/esp-open-sdk/sdk/include/osapi.h:33:6: note: previous declaration of 'ets_delay_us' was here
 void ets_delay_us(uint16_t us);
      ^
Makefile:557: recipe for target 'out/build/system/flashmem.o' failed
make[1]: *** [out/build/system/flashmem.o] Error 1
make[1]: Leaving directory '/opt/Sming/Sming'
Makefile:512: recipe for target 'reload' failed
make: *** [reload] Error 2
sanyaade@ns1:/opt/Sming/Sming$ 
`
Also my first time of using Sming but I am learning the framework along as I work with it. Many thanks for it!
Please help me!
Hear from you soon!

God blesses!!!

Best regards,
Sanyaade

avr39-ripe, 2018-03-23:
Try to 

export SDK_BASE="$SMING_HOME/third-party/ESP8266_NONOS_SDK"

and make dist-clean && make

sanyaade-buildtools, 2018-03-24:
@avr39-ripe: 

Thank you so much for your support and quick response to my problems.
Yes! I did what you outlined above and it works like charm :1st_place_medal:  Kudos!!! Many thanks

**So in a nut shell:**
export ESP_HOME=/opt/esp-open-sdk
export SMING_HOME=/opt/Sming/Sming
export SDK_BASE="$SMING_HOME/third-party/ESP8266_NONOS_SDK"
make dist-clean && make

I have test the Basic_Blink sample, I was able to compile and flash my ESP8266Ex. Many thanks!
Once again, many thanks!

God blesses!!!

Best regards,
Sanyaade

slaff, 2018-03-24:
> God blesses!!!

@sanyaade-buildtools Can you please close the ticket ?

closed 2018-03-24 by sanyaade-buildtools